ros2/rmw_zenoh #39: rmw_zenoh fails to find typesupport for a type

esteve, 2020-10-05:
Tried rmw_zenoh with both talker and listener, but they fail with the same error:

```
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run demo_nodes_cpp talker
[INFO] [1601909676.433125064] []: /rosout
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl node: type support not from this implementation, at /home/esteve/Projects/adlink/rmw_zenoh_ws/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_publisher.cpp:109, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/publisher.c:172
```

methylDragon, 2020-10-05:
Hi there! Did you source your workspace and rebuild it, as per the installation instructions?

esteve, 2020-10-05:
Yes, I did, but I'll try again.

methylDragon, 2020-10-05:
Hmm, ok, an additional caveat is that any message packages you use must also be rebuilt with the rmw_zenoh package built and sourced in the workspace.

There might be issues with nested messages though.

esteve, 2020-10-08:
@methylDragon I've added the messages to the workspace and rebuilt it, but I still have the same issue, any other suggestion?

I submitted https://github.com/osrf/rmw_zenoh/pull/42 for review to add the repos for the messages, let me know if the changes look fine.

Thanks.

methylDragon, 2020-10-08:
Hi esteve, confirmed that you rebuilt it with the workspace sourced?

Could I trouble you to go into the build directory of your workspace, enter the interface/message package and show me the contents of that folder?

esteve, 2020-10-08:
I just added a few more message repos and rebuilt the workspace, it now works. Thanks!

I'll look into why I need the second `colcon build`, it shouldn't be necessary. As for the messages in the workspace, yes, it's a known issue in ROS 2 because if you combine prebuilt packages with either new typesupports or language bindings, colcon/ament won't know that they need to be regenerated. One solution would be to check the ament index for a given marker and if it can't be found, regenerate the message packages.

methylDragon, 2020-10-08:
It's great to hear that it works!

The second colcon build issue appears to be an issue with the dependency resolution. Colcon seems to be ignoring the "member of group" tag in the package.xml, which causes the typesupport package for zenoh to not build itself before any message or interface packages are built.

So the first build essentially allows for the typesupport package to be built, and the second build allows for the typesupport code to be generated since the typesupport package will have been built in the first build.

If you do manage to resolve the issue that would be super helpful though! I'm very interested to find out how the problem could be solved.

Cheers!

clalancette, 2023-11-09:
I'm going to close this old issue out, so we can focus on newer issues with new development.

closed 2023-11-09 by clalancette