ros2/rmw_zenoh #45: Spin off zenoh_vendor

esteve, 2020-10-19:
I'm working on adding support for `zenoh-pico` (https://github.com/eclipse-zenoh/zenoh-pico) to MicroROS and currently `zenoh_vendor` being in the same repo as `rmw_zenoh` causes `colcon` to build it. Could you spin off `zenoh_vendor` to a separate repository? And if it's not too much asking, naming it `zenoh-c_vendor`? It'd make it clear that it's about the C bindings for `zenoh` and not the other client implementations (e.g. `zenoh-pico`).

I'm working on a `zenoh-pico_vendor` package.

gbiggs, 2020-10-20:
Renaming the package is fine by me. If we're going to have multiple FFI packages then it makes sense.

I don't see a reason to move it to a separate repository, though. It's easy enough to tell colcon not to build the package, and `rmw_zenoh_cpp` depends on it. If we were to put several vendor packages for different zenoh FFIs in a single repository, that would make more sense and I'd be happy to do that.

esteve, 2020-10-20:
The reason for moving `zenoh_vendor` to a separate repo is to keep the micro-ROS infrastructure as simple as possible, currently it's just a bunch of shell scripts and I'd rather to have tweak them when `zenoh-pico` is integrated as well.

clalancette, 2023-11-09:
I'm going to close this old issue out, so we can focus on newer issues with new development.

closed 2023-11-09 by clalancette