ros2/rmw_zenoh #53: Messages are not received

tobiasblass, 2021-01-17:
Hi,
I tried to install rmw_zenoh as described in the readme. I then tried to test the installation as described using
```
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
ros2 run demo_nodes_cpp talker
```
and
``` 
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
ros2 run demo_nodes_cpp listener
```

I can see the publisher submitting the data (`[INFO] [1610904238.851981101] [talker]: Publishing: 'Hello World: 64'`) but nothing shows up in the receiver.

Things I've tried:
- Run zenohd by hand in a third terminal window (I'm not 100% sure whether that's required or not, so I tried both)
- Force the session to the local interface (I don't trust my Wifi to properly handle multicast messages) using `RMW_ZENOH_SESSION_LOCATOR=tcp/127.0.0.1:7447`

I've compiled rmw_zenoh and zenohd from source on a ROS 2 Foxy system.

Please let me know if you need more information or if there is some kind of debug option that I can enable to pin down the issue.

Regards
Tobias

esteve, 2021-01-18:
@tobiasblass thanks for trying out `rmw_zenoh`. On both terminals, did you source your workspace (i.e. `source install/setup.bash`) before running the exaples? `zenohd` can be installed via Debian packages from http://zenoh.io/docs/getting-started/installation/, but you only need it if you're using zenoh in client mode, but in peer mode listener and talker should connect to each other automatically.

tobiasblass, 2021-01-18:
> On both terminals, did you source your workspace (i.e. source install/setup.bash) before running the examples

Yes. Sorry, I forgot to mention that. To be precise, I use the `local_setup.bash` file, but that shouldn't make a difference here, does it?

> but you only need it if you're using zenoh in client mode, but in peer mode listener and talker should connect to each other automatically.

Okay, thanks for the clarification. I was slightly confused what the server was good for, but this makes a lot of sense.
So this sounds like the listener and the talker cannot peer up for some reason, or they do peer up but don't exchange messages properly. Is there a way to get some debug information from the zenoh implementation itself on that matter?

I also have attached the ROS debug output for both components (using `ros2 run demo_nodes_cpp talker --ros-args --log-level debug > /tmp/talker.log 2>&1`) in case that helps. The two components use the topic `/chatter`. I haven't been able to extract any useful information from it, but maybe it makes more sense to you.
[listener.log](https://github.com/esteve/rmw_zenoh/files/5830964/listener.log)
[talker.log](https://github.com/esteve/rmw_zenoh/files/5830965/talker.log)

clalancette, 2023-11-09:
I'm going to close this old issue out, so we can focus on newer issues with new development.

closed 2023-11-09 by clalancette