ros2/rmw_zenoh #117: Crash when deserializing message of matching topic but different type.

Yadunund, 2024-02-22:
Presently the key expressions for pub/subs do not include the topic type. Therefore with pub/subs created with same topic name and different type, the sub will still receive serialized payload form the pub but an error may be thrown when the sub fails to deserialize the payload into the sub type.
We can either include the type in the key expressions or just handle the error on the sub side.

```bash
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Fast CDR exception deserializing message of type std_msgs::msg::dds_::String_., at /home/yadunund/ws_rmw/src/rmw_alternative/rmw_zenoh_cpp/src/detail/type_support.cpp:133'

with this new error message:

  'could not deserialize ROS message, at /home/yadunund/ws_rmw/src/rmw_alternative/rmw_zenoh_cpp/src/rmw_zenoh.cpp:1567'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
failed to take message from subscription: could not deserialize ROS message, at /home/yadunund/ws_rmw/src/rmw_alternative/rmw_zenoh_cpp/src/rmw_zenoh.cpp:1567, at /home/yadunund/ros2_rolling/src/ros2/rcl/rcl/src/rcl/subscription.c:562
```

CannonFodder511, 2024-03-11:
Did you solve this?
Having the same issue.

Yadunund, 2024-03-12:
Hi @CannonFodder511 

The issue still needs to be solved.

> We can either include the type in the key expressions or just handle the error on the sub side.

Personally I think we should just handle the error in the subscription side. I wanted to first see how other DDS based middlewares handle this scenario.

Let me know if you're keen to contribute the fix and I can assign the ticket to you. If not, I will probably look at fixing the issue sometime next week or so.

closed 2024-06-24 by Yadunund