ros2/rmw_zenoh #126: Revisit `zenoh_router_check`

clalancette, 2024-03-07:
During `rmw_init`, we end up making a call to see if the Zenoh router is available: https://github.com/ros2/rmw_zenoh/blob/ae142228c946147bd44ecb8696f46a7eb7208746/rmw_zenoh_cpp/src/detail/zenoh_router_check.cpp#L51-L81 .

However, in a very busy system that check might timeout, even though it would (eventually) find the router.

There are several ways we could handle this:
1.  Increase the timeout for `z_info_routers_zid`.  This has the problems of a) not knowing how to do that, and b) choosing a timeout value that is short enough not to hang if the router isn't available, but long enough to deal with busy systems.
2.  Wait forever for a router to appear.  This is very ROS 1-like behavior.
3.  Do router detection asynchronously.  During initialization we would query for the router, but we wouldn't wait around for a response.  Assuming that the router "eventually" responded, we would have a callback that marked that fact.  If the router didn't respond in some timeout (this could be many seconds), then we would print a warning message to the user.

There may be additional ways to deal with this.

fujitatomoya, 2024-03-21:
@Yadunund /assign

Yadunund, 2024-04-04:
Closed by #135

closed 2024-04-04 by Yadunund