ros2/rmw_zenoh #130: Unable to obtain publisher GID and sequence number errors

AlexDayCRL, 2024-03-20:
Hi,

Thanks for all your work on the Zenoh RMW! We are very excited to try it out. 

I'm getting the following errors when using data from a rosbag through `ros2 bag play`. These errors show up in any node that subscribes to topics published over Zenoh. It seems like these errors go away when subscribing to "live" topics rather than those played back through `ros2 bag`. Do you have any idea what the root cause could be? 

```
[ERROR] [1710953784.261021973] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[ERROR] [1710953784.261040865] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
```

AlexDayCRL, 2024-03-20:
@clalancette sure! I just made a PR (#131).

closed 2024-03-20 by clalancette

ciandonovan, 2024-06-27:
This is from a subscriber using the latest rolling branch of rmw_zenoh on a Jazzy distro subscribing live to a camera topic. It works, but deluges the stdout with these messages. Using the shared-memory transport.

```
[mockup_qml_gui-1] [ERROR] [1719497157.000221366] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.000314464] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.000341074] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.032723472] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.032816011] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.032842760] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.049600756] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.049701398] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.049727169] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.082071154] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.082165510] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.082191840] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.099065772] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.099161315] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.099209017] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.131853878] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.131955986] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.131984621] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.148039035] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.148139258] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.148165378] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.180826513] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.180918982] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.180944405] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.197441264] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.197542394] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.197568585] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.230215471] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.230317649] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.230342862] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.246597162] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.246691099] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.246716311] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.279746836] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.279852297] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.279877509] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.296512026] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.296601632] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.296626356] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.329035363] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.329169458] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.329204728] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.345503378] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[... cut ...]
[mockup_qml_gui-1] [ERROR] [1719497157.970319451] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.970345921] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.986660494] [rmw_zenoh_cpp]: Unable to obtain publisher GID from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.986756386] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
[mockup_qml_gui-1] [ERROR] [1719497157.986782158] [rmw_zenoh_cpp]: Unable to obtain sequence number from the attachment.
```

clalancette, 2024-06-27:
> This is from a subscriber using the latest rolling branch of rmw_zenoh on a Jazzy distro subscribing live to a camera topic. It works, but deluges the stdout with these messages. Using the shared-memory transport.

This looks like it is using an older version of `rmw_zenoh_cpp` along with a newer version.  Are you sure you are running the same version of `rmw_zenoh_cpp` everywhere?

ciandonovan, 2024-06-27:
The publisher also spits out the same messages, maybe 50 of them, but not continuously like the subscriber

clalancette, 2024-06-27:
> Also should mention they're running in separate rootless Podman containers, with /tmp and /dev/shm bind-mounted into each and both using the host's network namespace so the shared-memory transport can work

It might be the case that shared-memory transport doesn't support attachments.  If that is the case, then this can't possibly work; we rely on the attachments to send metadata along with the messages.  That said, I really don't know what's happening; I don't see any of this locally.

ciandonovan, 2024-06-27:
Functionally it does work though, I'm seeing the camera stream in the application, and there's no significant loopback traffic, so it must be using the shared-memory transport. The RUST_LOG=trace also indicates this with mentions of `zenoh_shm: SharedMemoryBuf::as_slice()` as each message is sent

clalancette, 2024-06-27:
> Functionally it does work though, I'm seeing the camera stream in the application, and there's no significant loopback traffic, so it must be using the shared-memory transport. The RUST_LOG=trace also indicates this with mentions of `zenoh_shm: SharedMemoryBuf::as_slice()` as each message is sent

Yeah, it will deliver data, but none of the metadata will be there.  So things like timestamps and the such will all be missing and empty.  Many applications in ROS don't actually depend on any of that, but some do (like rosbag).

ciandonovan, 2024-06-27:
Is this on the roadmap to be addressed? You mentioned it might not be possible, why is that? Would it require reworking how shared-memory is done in Zenoh itself upstream or is it more fundamental?

clalancette, 2024-06-27:
> Is this on the roadmap to be addressed? You mentioned it might not be possible, why is that? Would it require reworking how shared-memory is done in Zenoh itself upstream or is it more fundamental?

No, not on the roadmap at the moment; this is the first I've heard about it.  We'll have to do some deeper digging to figure out what is happening.  I'm going to suggest you open a new issue, with a reproducer that we can use to do that digging.

ciandonovan, 2024-06-30:
On further investigation it's not an issue exclusive to the shared-memory transport, as the issue also occurs with it disabled and nodes communicating on the loopback interface.

Reproduced with image pub/sub, not sure if it's an issue with that implementation or the fact it's higher-bandwidth.