ros2/rmw_zenoh #137: Service call with ROS2 CLI results in `add_new_query` called repeatedly filling queue

jackg0, 2024-03-25:
Thanks for this great open source project! We're really excited to use this internally and help out where we can.

Note this issue is was spotted while using the `yadu/events` branch. I am using it because it added support for the events needed for RVIZ. I built ros2 rolling from source inside a docker container with ubuntu 24.04 as a base image. rmw_zenoh was built from source with the `yadu/events` branch on commit `d93ad02`. I don't know if this issue is related to this branch or just that the logging messages that were added in the branch made the issue clearer.

The issue occurs when making service calls using `ros2 service call <service> <type>`. I am using a `std_srvs/srv/Trigger` for simplicity. I'll see the following errors at a high frequency and the service call will time out. Note that the executor will also spin and presumably call the callback, but it does not result in a response to the service call. The info messages before and after spinning are provided in the output below as an example.

```
[ERROR] [1711319879.861642564] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
[ERROR] [1711319879.861683387] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
[ERROR] [1711319879.861818046] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
[ERROR] [1711319879.861846561] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
[INFO] [1711319879.861854427] [simple_server]: before spin
[ERROR] [1711319879.861936467] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
[INFO] [1711319879.862053562] [simple_server]: after spin
[ERROR] [1711319879.862092667] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
[ERROR] [1711319879.862160480] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
[ERROR] [1711319879.862237294] [rmw_zenoh_cpp]: Query queue depth of 1 reached, discarding oldest Query for service for 0/simple_service/trigger
```

To cause the issue, the following seems to be required (with `ros2 run rmw_zenoh_cpp rmw_zenohd` running):
* At least 2 separate nodes running on system in separate processes with 1 node having the service you're making the service call to.
* I must tab complete the service name using `ros2 service call <partial-service-name>` and all nodes will print the following warnings:
```
[WARN] [1711303086.719254316] [rmw_zenoh_cpp]: Received liveliness token to remove node /_ros2cli_8088 from the graph before all pub/subs/clients/services for
 this node have been removed. Removing all entities first...
[WARN] [1711303086.727274245] [rmw_zenoh_cpp]: Received liveliness token to remove unknown node /_ros2cli_8088 from the graph. Ignoring...
```
* I can then run the service call a few times and cause the issue.

Let me know if any other information would be helpful!

Yadunund, 2024-03-25:
Hello,

Thanks for the feedback!

The `Received liveliness token to remove node...` is expected at the moment as maybe reflective of a deeper issue within `rcl`.

However I was unable to reproduce the high frequency errors highlighted with `add_two_ints_server`.

```bash
# terminal 1
ros2 run demo_nodes_cpp add_two_ints_server
```

```bash
# terminal 2 
ros2 service call  # Press tab here a few times -> The "Received liveliness token to remove node..." warnings show up in terminal 1 but no other errors.
## Successfully populates list of active services
```

Is the `simple_server` a node that you wrote? If so, it would be very helpful if you could share the code or a way for me to run the node to try to reproduce the behavior.

jackg0, 2024-03-26:
Thank you for the reply!

Yes, I'll make sure to share the source code when I get a chance.

An important note to help reproduce the issue is I have to have other nodes running. I'm not sure why this is. With other nodes running I do see the `Received liveliness token to remove node...` in each terminal. Afterwards, I can trigger the high frequency error messages after several service calls. It's also common to see one-off error messages saying `Query queue depth of 1 reached...`.

Yadunund, 2024-03-26:
I had a whole bunch of other nodes running when I tried that test with `add_two_ints_server`. Repeatedly sending the service request also does not trigger the error.
I'll wait for you to share instructions on reproducing your setup.

I'll also add that if you had `rolling` branch built and you later switched to `yadu/events`, then you'll need to ensure you stop all ROS 2 processes and especially the ros2 daemon before running any nodes with that branch. This is because the liveliness token keyexpressions have changes between these versions. `pkill -9 -f ros && ros2 daemon stop`

jackg0, 2024-03-26:
This issue was possibly caused by building on `noble` with `rustc` version 1.76. I am unable to reproduce using `yadu/noble` rebased on `rolling` (with `yadu/events` merged) which builds with `rustc` version 1.75.

Feel free to close this issue unless you see any reason to keep it open.

Thanks for the help!

closed 2024-03-27 by Yadunund