ros2/rmw_zenoh #152: [test_rclcpp] test_services_cpp (test_rclcpp.TestTwoExecutables.test_services_cpp)

Yadunund, 2024-04-05:
Failure

```bash
[test_services_server_cpp-1]   what():  failed to send response: Unable to find taken request. Report this bug., at /home/yadunund/ros2_rolling/src/rmw_alternative/rmw_zenoh_cpp/src/rmw_zenoh.cpp:2890, at /home/yadunund/ros2_rolling/src/ros2/rcl/rcl/src/rcl/service.c:400
```

Yadunund, 2024-04-05:
With some debug statements it looks like the the `sequence_number` generated for each client is always `1` (See lines "Generated sequence_id 1 for client ..." below). Therefore the service cannot find the same `sequence_number` after it returns a response for the first client.

```bash
[test_services_client_cpp-2] [ RUN      ] test_services_client.test_add_noreqid
[test_services_client_cpp-2] Generated sequence_id 1 for client over /add_two_ints_noreqid
[test_services_client_cpp-2] [       OK ] test_services_client.test_add_noreqid (109 ms)
[test_services_client_cpp-2] [ RUN      ] test_services_client.test_add_reqid
[test_services_client_cpp-2] Generated sequence_id 1 for client over /add_two_ints_reqid
[test_services_client_cpp-2] [       OK ] test_services_client.test_add_reqid (2 ms)
[test_services_client_cpp-2] [ RUN      ] test_services_client.test_return_request
[test_services_client_cpp-2] Generated sequence_id 1 for client over /add_two_ints_reqid_return_request
[test_services_client_cpp-2] [       OK ] test_services_client.test_return_request (2 ms)
[test_services_client_cpp-2] [ RUN      ] test_services_client.test_add_two_ints_defered_cb
[test_services_client_cpp-2] Generated sequence_id 1 for client over /add_two_ints_defered_cb
[test_services_client_cpp-2] [       OK ] test_services_client.test_add_two_ints_defered_cb (3 ms)
[test_services_client_cpp-2] [ RUN      ] test_services_client.test_add_two_ints_defcb_with_handle
[test_services_server_cpp-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[test_services_server_cpp-1]   what():  failed to send response: Unable to find taken request for sequence_number 1. Report this bug., at /home/yadunund/ros2_rolling/src/rmw_alternative/rmw_zenoh_cpp/src/rmw_zenoh.cpp:2897, at /home/yadunund/ros2_rolling/src/ros2/rcl/rcl/src/rcl/service.c:400
```

Yadunund, 2024-04-05:
Okay so this is caused by the  `sequence_to_query_map_` data structure used to map `sequence_number` of client request and the `ZenohQuery`, ie [this map](https://github.com/ros2/rmw_zenoh/blob/58404f7339ad56ec3c219be645aead71ae56034e/rmw_zenoh_cpp/src/detail/rmw_data_types.hpp#L268-L269).
We're not distinguishing between which client the query came from.

Two solutions come to mind:
1. Make sequence_number generated by the client globally unique. 
2. Update `sequence_to_query_map_` such that it maps `std::unordered_map<gid, std::unordered_map<sequence_number, ZenohQuery>>`

@clalancette what do you think? Personally I think we 2 is easier.

clalancette, 2024-04-05:
> We're not distinguishing between which client the query came from.

Ah, of course.

> 2\. Update `sequence_to_query_map_` such that it maps `std::unordered_map<gid, std::unordered_map<sequence_number, ZenohQuery>>`

Just to be clear; the gid as the kye here would be the `gid` that the client attaches, correct?  If so, then yes, I think this is the way to go; it uses information we already have.

Yadunund, 2024-04-05:
Yupp the `gid` the client already attaches.

closed 2024-04-13 by Yadunund