ros2/rmw_zenoh #156: [test_communication] failures

Yadunund, 2024-04-13:
Requires `test_communication` package to be built with `colcon build --mixin release --packages-select  test_communication --cmake-args -DSKIP_MULTI_RMW_TESTS=On` to disable tests between rmw_zenoh and DDS based middlewares.

```bash

98% tests passed, 3 tests failed out of 128

Label Time Summary:
copyright      =   2.45 sec*proc (1 test)
cppcheck       =   0.38 sec*proc (1 test)
cpplint        =   0.95 sec*proc (1 test)
flake8         =   0.56 sec*proc (1 test)
gtest          = 102.12 sec*proc (12 tests)
launch_test    = 558.91 sec*proc (48 tests)
lint_cmake     =   0.36 sec*proc (1 test)
linter         =   6.95 sec*proc (8 tests)
pep257         =   0.45 sec*proc (1 test)
uncrustify     =   0.49 sec*proc (1 test)
xmllint        =   1.31 sec*proc (1 test)

Total Test time (real) = 718.31 sec

The following tests FAILED:
	 85 - test_messages_c__rmw_zenoh_cpp (Timeout)
	101 - test_publisher_subscriber__rclpy__rmw_zenoh_cpp (Subprocess killed)
	124 - flake8 (Failed)
```

MichaelOrlov, 2024-04-25:
@Yadunund On the latest waffle triage meeting we decided to move this issue to the backlog for now.

imstevenpmwork, 2024-05-28:
Hello @Yadunund, here's the backtrace from gdb:

```log
(gdb) info threads
  Id   Target Id                                          Frame 
* 1    Thread 0x7173c022d8c0 (LWP 9979) "test_messages_c" 0x00007173c0498d61 in __futex_abstimed_wait_common64 (
    private=-90121620, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x568e99b9a298)
    at ./nptl/futex-internal.c:57
  2    Thread 0x7173b5e006c0 (LWP 10013) "RX-3"           syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  3    Thread 0x7173b62006c0 (LWP 10012) "Net-3"          syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  4    Thread 0x7173b6a006c0 (LWP 9992) "RX-1"            0x00007173c052a042 in epoll_wait (epfd=35, 
    events=0x7173a0131610, maxevents=1024, timeout=8831) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  5    Thread 0x7173b6e006c0 (LWP 9991) "RX-0"            syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  6    Thread 0x7173b72006c0 (LWP 9990) "TX-1"            syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  7    Thread 0x7173b76006c0 (LWP 9989) "TX-0"            0x00007173c052a042 in epoll_wait (epfd=31, 
    events=0x7173a00099e0, maxevents=1024, timeout=1855) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7173b7e006c0 (LWP 9987) "Acceptor-1"      0x00007173c052a042 in epoll_wait (epfd=27, 
    events=0x568e99b83810, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  9    Thread 0x7173bc6006c0 (LWP 9986) "Acceptor-0"      syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  10   Thread 0x7173bca006c0 (LWP 9985) "Net-1"           syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  11   Thread 0x7173bce006c0 (LWP 9984) "Net-0"           0x00007173c052a042 in epoll_wait (epfd=23, 
    events=0x568e99b7b3b0, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  12   Thread 0x7173bd2006c0 (LWP 9983) "Application-1"   syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  13   Thread 0x7173bd6006c0 (LWP 9982) "Application-0"   0x00007173c052a042 in epoll_wait (epfd=18, 
    events=0x568e99b71dc0, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  14   Thread 0x7173bf8006c0 (LWP 9981) "test_messag-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  15   Thread 0x7173c02006c0 (LWP 9980) "test_messag-ust" 0x00007173c052be3b in __recvmsg_syscall (flags=0, 
    msg=0x7173c01fe4d0, fd=3) at ../sysdeps/unix/sysv/linux/recvmsg.c:27
(gdb) backtrace
#0  0x00007173c0498d61 in __futex_abstimed_wait_common64 (private=-90121620, cancel=true, abstime=0x0, op=393, 
    expected=0, futex_word=0x568e99b9a298) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=-90121620, abstime=0x0, clockid=0, expected=0, 
    futex_word=0x568e99b9a298) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x568e99b9a298, expected=expected@entry=0, 
    clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0) at ./nptl/futex-internal.c:139
#3  0x00007173c049b7dd in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x568e99b9a2a0, cond=0x568e99b9a270)
    at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=0x568e99b9a270, mutex=0x568e99b9a2a0) at ./nptl/pthread_cond_wait.c:627
#5  0x00007173bf9b9aef in rmw_wait (subscriptions=0x568e99b832c8, guard_conditions=0x568e99b832e0, 
    services=0x568e99b83310, clients=0x568e99b832f8, events=0x568e99b83328, wait_set=0x568e99b9a250, wait_timeout=0x0)
    at /home/zenoh/Documents/rmw_ws/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:3288
#6  0x00007173c0b9d1fc in rcl_wait () from /opt/ros/rolling/lib/librcl.so
#7  0x0000568e993df0d8 in TestMessagesFixture::test_message_type<test_msgs__msg__BasicTypes> (this=0x568e99b70da0, 
    topic_name=0x568e994596ac "test_basic_types", ts=0x7173c0b6ad80, context=0x568e99b70380)
    at /home/zenoh/Documents/st_ws/src/system_tests/test_communication/test/test_messages_c.cpp:187
#8  0x0000568e993cd893 in TestMessagesFixture_test_basic_types_Test::TestBody (this=0x568e99b70da0)
    at /home/zenoh/Documents/st_ws/src/system_tests/test_communication/test/test_messages_c.cpp:390
#9  0x0000568e99442fd9 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (
    object=0x568e99b70da0, method=&virtual testing::Test::TestBody(), location=0x568e9945d09b "the test body")
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2612
#10 0x0000568e9943a94f in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (
    object=0x568e99b70da0, method=&virtual testing::Test::TestBody(), location=0x568e9945d09b "the test body")
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2648
#11 0x0000568e99411706 in testing::Test::Run (this=0x568e99b70da0)
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2687
--Type <RET> for more, q to quit, c to continue without paging--
#12 0x0000568e994122d7 in testing::TestInfo::Run (this=0x568e99b6eb40)
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2836
#13 0x0000568e99412d50 in testing::TestSuite::Run (this=0x568e99b6f0f0)
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:3015
#14 0x0000568e99423ed4 in testing::internal::UnitTestImpl::RunAllTests (this=0x568e99b6eda0)
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:5920
[... cut ...]
    location=0x568e9945dae8 "auxiliary test code (environments or event listeners)")
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2648
#17 0x0000568e99422307 in testing::UnitTest::Run (this=0x568e9948f3c0 <testing::UnitTest::GetInstance()::instance>)
    at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:5484
#18 0x0000568e993fcb01 in RUN_ALL_TESTS () at /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:2317
#19 0x0000568e993fcae9 in main (argc=1, argv=0x7ffedb474508) at /opt/ros/rolling/src/gtest_vendor/src/gtest_main.cc:64
(gdb) 
```

clalancette, 2024-08-21:
I just retested this, and it seems to work with the latest version of `rmw_zenoh_cpp`.  I believe that the reworks we did for `rmw_wait` fixed it.  I'm going to close this out, but feel free to reopen if you see it again.

closed 2024-08-21 by clalancette