ros2/rmw_zenoh #158: Timer callback stops firing when no data is moving through the system

AlexDayCRL, 2024-04-17:
Hi,

I am having troubles with my timer callback calling after a few seconds when using the zenoh middleware with my application. I am not experiencing this issue with cyclonedds. I have tried and failed to reproduce it with a small example. Is there any way to introspect the timer queue within Zenoh?

clalancette, 2024-04-17:
So it is tricky because `rmw_zenoh` doesn't directly deal with the timer callbacks.  Those are handled at the `rcl`/`rclcpp` layer.  However, if the `rmw_wait` that is implemented in Zenoh gets hung up for some reason, it can cause the timer callbacks to be delayed.  They should eventually fire (after `rmw_wait` times out), but they may be delayed.

It's hard to say here without an example, but there have been some other issues we've been trying to track down with `rmw_wait`.  #153 should have solved some of it, but seems to have caused a regression.  If you apply #157, does the problem go away?

AlexDayCRL, 2024-04-17:
@clalancette I still have the same issues on #157. From your comments it seems that I would want to instrument the `rmw_wait` function if I wanted to drill down into what was happening though.

MichaelOrlov, 2024-04-25:
Moving to the backlog for now since other relevant issues #153 and #157 haven't been resolved yet.

AlexDayCRL, 2024-04-30:
This issue also only happens when my node has multiple timers.

Yadunund, 2024-05-10:
Hey @AlexDayCRL could you try doing a clean build with the latest `rolling` branch and confirm if the problem still exists?

clalancette, 2024-08-21:
There's been no response here in a long time, so I'm going to close this out.  If you are still having problems with the latest, please feel free to reopen.

closed 2024-08-21 by clalancette