ros2/rmw_zenoh #161: [test_quality_of_service] Failures

Yadunund, 2024-04-18:
`colcon test --packages-select test_quality_of_service`

```bash
86% tests passed, 3 tests failed out of 22

Label Time Summary:
copyright     =   1.68 sec*proc (1 test)
cppcheck      =   0.39 sec*proc (1 test)
cpplint       =   0.62 sec*proc (1 test)
gtest         = 109.42 sec*proc (16 tests)
lint_cmake    =   0.38 sec*proc (1 test)
linter        =   4.23 sec*proc (6 tests)
uncrustify    =   0.43 sec*proc (1 test)
xmllint       =   0.72 sec*proc (1 test)

Total Test time (real) = 113.66 sec

The following tests FAILED:
	 13 - test_deadline__rmw_zenoh_cpp (Failed)
	 15 - test_liveliness__rmw_zenoh_cpp (Failed)
	 16 - test_best_available__rmw_zenoh_cpp (Failed)

```

Yadunund, 2024-04-18:
Tests fail as these event types are not yet supported in `rmw_zenoh`

### 13 - test_deadline__rmw_zenoh_cpp

```bash
[INFO] [1713435048.611577533] [subscriber]: created subscriber subscriber test_deadline
unknown file: Failure
C++ exception with description "Failed to initialize event: provided event_type 7 is not supported by rmw_zenoh_cpp, at /home/yadunund/ros2_rolling/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_event.cpp:52" thrown in the test body.
```

### 15 - test_liveliness__rmw_zenoh_cpp

```bash
[INFO] [1713435203.641990760] [subscriber]: created subscriber subscriber test_automatic_liveliness_changed
unknown file: Failure
C++ exception with description "Failed to initialize event: provided event_type 0 is not supported by rmw_zenoh_cpp, at /home/yadunund/ros2_rolling/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_event.cpp:105" thrown in the test body.
```

###  16 - test_best_available__rmw_zenoh_cpp
Failures are related to deadline and liveliness_lease_duration

```bash
[INFO] [1713435303.382419270] [subscriber]: created subscriber subscriber /test_best_available_subscription
/home/yadunund/ros2_rolling/src/ros2/system_tests/test_quality_of_service/test/test_best_available.cpp:87: Failure
Expected equality of these values:
  actual_qos.deadline()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 00-00 00-00 00-00 00-00>
  publisher_qos_profile.deadline()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 80-84 1E-00 00-00 00-00>

/home/yadunund/ros2_rolling/src/ros2/system_tests/test_quality_of_service/test/test_best_available.cpp:89: Failure
Expected equality of these values:
  actual_qos.liveliness_lease_duration()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 00-00 00-00 00-00 00-00>
  publisher_qos_profile.liveliness_lease_duration()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 C0-C6 2D-00 00-00 00-00>

[  FAILED  ] QosRclcppTestFixture__rmw_zenoh_cpp.test_best_available_policies_subscription (1533 ms)
[ RUN      ] QosRclcppTestFixture__rmw_zenoh_cpp.test_best_available_policies_publisher
[INFO] [1713435303.901693591] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id 6678b139223765f71bf21f1c8967c71.
[INFO] [1713435304.903443449] [publisher]: created publisher publisher /test_best_available_publisher

[INFO] [1713435304.904354009] [subscriber]: created subscriber subscriber /test_best_available_publisher
/home/yadunund/ros2_rolling/src/ros2/system_tests/test_quality_of_service/test/test_best_available.cpp:147: Failure
Expected equality of these values:
  actual_qos.deadline()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 00-00 00-00 00-00 00-00>
  subscription_qos_profile.deadline()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 80-84 1E-00 00-00 00-00>

/home/yadunund/ros2_rolling/src/ros2/system_tests/test_quality_of_service/test/test_best_available.cpp:148: Failure
Expected equality of these values:
  actual_qos.liveliness()
    Which is: 4-byte object <01-00 00-00>
  subscription_qos_profile.liveliness()
    Which is: 4-byte object <03-00 00-00>

/home/yadunund/ros2_rolling/src/ros2/system_tests/test_quality_of_service/test/test_best_available.cpp:149: Failure
Expected equality of these values:
  actual_qos.liveliness_lease_duration()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 00-00 00-00 00-00 00-00>
  subscription_qos_profile.liveliness_lease_duration()
    Which is: 16-byte object <78-AB B4-AC 1B-7F 00-00 C0-C6 2D-00 00-00 00-00>

```

MichaelOrlov, 2024-04-25:
@Yadunund A friendly ping to handle this issue.  
At the waffle meeting, we decided to assign it to you, but I don't have permission to do so.

clalancette, 2024-08-21:
This is still a problem on the latest as of today.

Yadunund, 2024-12-27:
Tests skipped for now https://github.com/ros2/system_tests/pull/551

closed 2024-12-27 by Yadunund