ros2/rmw_zenoh #162: [test_rclcpp] #104 test_n_nodes__rmw_zenoh_cpp

Yadunund, 2024-04-18:
Nodes do not shutdown properly

```bash
104: ======================================================================
104: FAIL: test_10_nodes (test_rclcpp.TestNNodesAfterShutdown.test_10_nodes)
104: ----------------------------------------------------------------------
104: Traceback (most recent call last):
104:   File "/home/yadunund/ros2_rolling/build/test_rclcpp/test_n_nodes__rmw_zenoh_cpp_Release.py", line 61, in test_10_nodes
104:     launch_testing.asserts.assertExitCodes(
104:   File "/home/yadunund/ros2_rolling/install/launch_testing/lib/python3.12/site-packages/launch_testing/asserts/assert_exit_codes.py", line 62, in assertExitCodes
104:     assert info.returncode in allowable_exit_codes, 'Proc {} exited with code {}'.format(
104: AssertionError: Proc node_with_name_2-3 exited with code -2
```

MichaelOrlov, 2024-04-25:
@Yadunund A friendly ping to handle this issue.  
At the waffle meeting, we decided to assign it to you, but I don't have permission to do so.

Yadunund, 2024-05-09:
Unable to reproduce after latest zenoh-c bump #134.

closed 2024-05-09 by Yadunund