ros2/rmw_zenoh #165: Cannot compile in debug mode

Timple, 2024-04-29:
We typically have a colcon defaults file containing:
```
 - "-DCMAKE_BUILD_TYPE=Debug"
 - "-DCMAKE_CXX_FLAGS=-O2"
```
In order to still catch asserts in our code. However `rmw_zenoh` doesn't compile with these settings. With `RelWithDebInfo` it does.

clalancette, 2024-04-30:
Hm, I just tried it here, and it seems to compile for me:

```
colcon build --event-handler console_direct+ --cmake-args -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_FLAGS=-O2 --packages-select rmw_zenoh_cpp
```

In this test, I was on Ubuntu 22.04, building against Iron.  Can you give me more information about your setup?

Timple, 2024-04-30:
I can, I'll try to make a reproducable docker. 
But what I already notice is you're not compiling `zenoh_c_vendor`. So I'd make it `--packages-up-to`

clalancette, 2024-04-30:
> But what I already notice is you're not compiling `zenoh_c_vendor`. So I'd make it `--packages-up-to` in a clean workspace.

Ah, OK.  I'll give it a try.

clalancette, 2024-04-30:
OK, now I see the problem.  For some reason when you build `zenoh_c` in Debug mode, it actually changes the name of the CMake config file to `zenohc_debug`, which we weren't looking for.  I'll open a patch to fix this shortly.

Timple, 2024-04-30:
Awesome, thanks for diving in!

PR #166 (2024-05-02): Fix compilation when building zenohc in debug mode.
When zenohc is built in debug mode, it actually changes the name of the zenohcConfig.cmake file to zenohc_debugConfig.cmake. Luckily it does not change the name of the library.

To deal with this, first quietly look for the debug version. If we don't find that, then fall back to the release version. This should make it compile in both circumstances.

This should fix #165
closed 2024-05-02 by Yadunund