ros2/rmw_zenoh #175: Question: Plans to support loaned-messages shared-memory APIs?

mauropasse, 2024-05-17:
Hi! I've running some ROS2 benchmarks on `rmw_zenoh` using the iRobot [ros2-performance](https://github.com/irobot-ros/ros2-performance) framework, and it looks promising. However I noticed the lack of support for loaned messages APIs like:
https://github.com/ros2/rmw_zenoh/blob/c76c9627ca8cf8c133124368f2f1ad06092b7027/rmw_zenoh_cpp/src/rmw_zenoh.cpp#L758-L767

It seems zenoh-c already supports shared memory: https://github.com/eclipse-zenoh/zenoh-c/pull/156, https://zenoh.io/blog/2023-06-05-charmander2/.

So I'm wondering what's missing to fully implement shared memory on ROS2 systems, and if there is a plan in the roadmap to do so? This feature is key for performant multi-process applications.

imstevenpmwork, 2024-05-17:
Hello @mauropasse, would you mind sharing the benchmarks results with us? Just for curiosity :)
In the meantime I'll check with the rest of the team regarding your question

clalancette, 2024-05-17:
> So I'm wondering what's missing to fully implement shared memory on ROS2 systems, and if there is a plan in the roadmap to do so? This feature is key for performant multi-process applications.

We haven't prioritized loaned messages, mostly because they are disabled by default in Rolling and Jazzy (as they are currently unsafe to use).

At the moment, we don't have a plan to implement them, but it would be great to complete the API in this way.  If you are interested in looking into it, we'd be happy to review patches.  Otherwise, it will probably be a long-term goal.

mauropasse, 2024-09-13:
Hi guys! I've been working on a PR https://github.com/ros2/rclcpp/pull/2624 that aims to re-enable the use of loaned messages on rclcpp.

It contains a unit test to verify correct memory handling when the last owner of the loaned message is either the:

- User (storing messages beyond the subscription callback).
- Subscription (storing messages until it is executed).
- Publisher (storing messages for late joiner subscriptions).

By correctly memory handling I mean, memory should not be reused until all entities had returned the loans (all messages went ouf scope).

The test currently pass with CycloneDDS & FastDDS. I didn't test Zenoh since `rmw_zenoh_cpp` doesn't yet implement the loaned APIs.

My question is, supposing that the APIs were implemented, would Zenoh correctly handle the situations described above? Or that would depend actually on the implems on the missing APIs, described here https://design.ros2.org/articles/zero_copy.html

Besides that, are there any updates on using `zenoh-c > v1.0.0`, which comes with support for shared memory?