ros2/rmw_zenoh #182: Deadlock loading components

hwoithe, 2024-05-25:
I am experience a hang while loading many components with rmw_zenoh.

Terminal 1:
```
$ ros2 run rmw_zenoh_cpp rmw_zenohd
```

Terminal 2:
```
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run rclcpp_components component_container
```

Terminal 3:
```
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ for i in {1..100}; do ros2 component load --node-namespace /talker${i} /ComponentManager composition composition::Talker; done
...
Loaded component 75 into '/ComponentManager' container node as '/talker78/talker'
[INFO] [1716582392.589218332] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id e5c25c3ed59ffe52c2913bf5152eccf.
Loaded component 76 into '/ComponentManager' container node as '/talker79/talker'
[INFO] [1716582395.719657002] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id e5c25c3ed59ffe52c2913bf5152eccf.
Loaded component 77 into '/ComponentManager' container node as '/talker80/talker'
[INFO] [1716582398.886646585] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id e5c25c3ed59ffe52c2913bf5152eccf.
```

I would expect the for loop above to eventually finish but it does not.

Let's attach to the process and investigate:
```
$ sudo gdb attach -p `pgrep -f talker81`
...
(gdb) info threads
  Id   Target Id                                      Frame 
* 1    Thread 0x7fc68d86d740 (LWP 1481419) "ros2"     syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  2    Thread 0x7fc68bae8640 (LWP 1481420) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  3    Thread 0x7fc68b2e7640 (LWP 1481421) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  4    Thread 0x7fc6831a5640 (LWP 1481422) "ros2"     __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7fc68c98b280 <(anonymous namespace)::g_signal_handler_sem>) at ./nptl/futex-internal.c:57
  5    Thread 0x7fc6829a4640 (LWP 1481423) "app-0"    0x00007fc68d995e2e in epoll_wait (epfd=3, events=0x55d534d00580, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  6    Thread 0x7fc6827a3640 (LWP 1481424) "net-0"    0x00007fc68d995e2e in epoll_wait (epfd=8, events=0x55d534d03590, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  7    Thread 0x7fc6825a2640 (LWP 1481425) "acc-0"    0x00007fc68d995e2e in epoll_wait (epfd=12, events=0x55d534d065a0, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7fc6821a0640 (LWP 1481427) "tx-0"     0x00007fc68d995e2e in epoll_wait (epfd=16, events=0x7fc67000a830, maxevents=1024, timeout=399) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  9    Thread 0x7fc681f9f640 (LWP 1481428) "rx-0"     futex_wait (private=0, expected=2, futex_word=0x7fc68c98aa60 <rclpy::LoggingGuard::logging_mutex_>) at ../sysdeps/nptl/futex-internal.h:146
  10   Thread 0x7fc681d9e640 (LWP 1481429) "rx-1"     0x00007fc68d995e2e in epoll_wait (epfd=19, events=0x7fc670131ea0, maxevents=1024, timeout=8911) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  11   Thread 0x7fc681b9d640 (LWP 1481430) "ros2"     __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7fc681b9cde0, op=137, expected=0, futex_word=0x55d534fb3cc0) at ./nptl/futex-internal.c:57
...
(gdb) thread 1
[Switching to thread 1 (Thread 0x7fc68d86d740 (LWP 1481419))]
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
38  in ../sysdeps/unix/sysv/linux/x86_64/syscall.S
(gdb) bt
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007fc689411674 in std::sys::pal::unix::futex::futex_wait () at library/std/src/sys/pal/unix/futex.rs:62
#2  std::sys::sync::mutex::futex::Mutex::lock_contended () at library/std/src/sys/sync/mutex/futex.rs:68
#3  0x00007fc6898d6f05 in std::sys::sync::mutex::futex::Mutex::lock () at /rustc/9b00956e56009bab2aa15d7bff10916599e3d6d6/library/std/src/sys/sync/mutex/futex.rs:40
#4  std::sync::mutex::Mutex<alloc::boxed::Box<(dyn zenoh::net::routing::hat::HatTrait + core::marker::Send + core::marker::Sync), alloc::alloc::Global>>::lock<alloc::boxed::Box<(dyn zenoh::net::routing::hat::HatTrait + core::marker::Send + core::marker::Sync), alloc::alloc::Global>> (self=0x55d53487e9f0)
    at /rustc/9b00956e56009bab2aa15d7bff10916599e3d6d6/library/std/src/sync/mutex.rs:317
#5  0x00007fc6898e9259 in zenoh::net::routing::dispatcher::face::{impl#4}::send_declare (self=0x55d534bd0510, msg=...) at src/net/routing/dispatcher/face.rs:189
#6  0x00007fc6897d03b0 in zenoh::session::Session::declare_queryable_inner (self=0x55d534cb7700, key_expr=0x7ffe5e028130, complete=true, origin=zenoh::sample::Locality::Any, callback=...) at src/session.rs:1249
#7  0x00007fc689576ef6 in zenoh::queryable::{impl#20}::res_sync<zenoh::handlers::DefaultHandler> (self=...) at <***removed***>/.cargo/git/checkouts/zenoh-cc237f2570fab813/763a05f/zenoh/src/queryable.rs:632
#8  0x00007fc68956dcc2 in zenoh_ext::publication_cache::PublicationCache::new (conf=...) at src/publication_cache.rs:160
#9  0x00007fc68956cbdd in zenoh_ext::publication_cache::{impl#2}::res_sync (self=<error reading variable: Cannot access memory at address 0x89>) at src/publication_cache.rs:99
#10 0x00007fc68951061b in zenohcd::publication_cache::ze_declare_publication_cache (session=..., keyexpr=..., options=...) at <***removed***>/ws_rmw_zenoh/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/publication_cache.rs:136
#11 0x00007fc68aa8ba7d in rmw_create_publisher (node=0x55d534fb43d0, type_supports=0x7fc68bd05b80, topic_name=0x55d534fb5850 "/rosout", qos_profile=0x7ffe5e02d7b0, publisher_options=0x7ffe5e02d830) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:586
#12 0x00007fc68cda6046 in rcl_publisher_init () from /opt/ros/iron/lib/librcl.so
#13 0x00007fc68cda693a in ?? () from /opt/ros/iron/lib/librcl.so
#14 0x00007fc68cda70db in rcl_node_init () from /opt/ros/iron/lib/librcl.so
#15 0x00007fc68c7c2cec in rclpy::Node::Node (this=0x55d534ca3090, node_name=0x55d5349fcc40 "_ros2cli_1481419", namespace_=0x7ffe5e0343b8 "", context=..., pycli_args=..., use_global_arguments=true, enable_rosout=true) at <***removed***>/ws_rmw_zenoh/src/rclpy/rclpy/src/rclpy/node.cpp:464
#16 0x00007fc68c7cb7f5 in pybind11::detail::initimpl::construct_or_initialize<rclpy::Node, char const*, char const*, rclpy::Context&, pybind11::object, bool, bool, 0> () at /usr/include/pybind11/detail/init.h:61
#17 0x00007fc68c7c99cb in pybind11::detail::initimpl::constructor<char const*, char const*, rclpy::Context&, pybind11::object, bool, bool>::execute<pybind11::class_<rclpy::Node, rclpy::Destroyable, std::shared_ptr<rclpy::Node> >, , 0>(pybind11::class_<rclpy::Node, rclpy::Destroyable, std::shared_ptr<rclpy::Node> >&)::{lambda(pybind11::detail::value_and_holder&, char const*, char const*, rclpy::Context&, pybind11::object, bool, bool)#1}::operator()(pybind11::detail::value_and_holder&, char const*, char const*, rclpy::Context&, pybind11::object, bool, bool) const (__closure=0x55d534824ba8, v_h=..., args#0=0x55d5349fcc40 "_ros2cli_1481419", args#1=0x7ffe5e0343b8 "", args#2=..., args#3=..., args#4=true, 
[... 64 lines cut ...]
(gdb) up 3
#3  ___pthread_mutex_lock (mutex=0x7fc68c98aa60 <rclpy::LoggingGuard::logging_mutex_>) at ./nptl/pthread_mutex_lock.c:128
128 ./nptl/pthread_mutex_lock.c: No such file or directory.
(gdb) print mutex.__data.__owner
$1 = 1481419
(gdb) thread find 1481419
Thread 1 has target id 'Thread 0x7fc68d86d740 (LWP 1481419)'
```

Thread 9 is waiting on some resource held by thread 1.

Frame 9 on thread 9 wants to acquire ```logging_mutex_``` in ```rclpy_thread_safe_logging_output_handler()```:
https://github.com/ros2/rclpy/blob/52f2a69325721e1368ac3176adcc348e1f53f93e/rclpy/src/rclpy/logging.cpp#L51

Frame 15 on thread 1 should have acquired ```logging_mutex_``` right before ```rcl_node_init()```:
https://github.com/ros2/rclpy/blob/52f2a69325721e1368ac3176adcc348e1f53f93e/rclpy/src/rclpy/node.cpp#L464

This means that thread 1 is holding the ```logging_mutex_``` that thread 9 is waiting on.

Hopefully I have read the situation correctly here and, at the very least, others are able to reproduce the problem. I will continue this investigation after the US holiday weekend.

ROS Distribution: Iron 
OS: Ubuntu Jammy
rmw_zenohd: commit 93ba28e0959940200ebbd0066d3433a426088ef2.
rust: rustc 1.78.0 (9b00956e5 2024-04-29)

Yadunund, 2024-05-26:
I think this is similar to https://github.com/ros2/rmw_zenoh/issues/143 which may be a consequence of #170.

hwoithe, 2024-06-05:
I have updated my Iron environment and am still experiencing the same issue where a node loading a component in the bash for loop eventually hangs. I am observing a different issue in Jazzy and will follow up with a more detailed description.

Between the Iron and Jazzy releases, the use of the global logger mutex in ```rclpy``` has changed. In Iron we hold it for the entire execution of ```rcl_node_init()```:
https://github.com/ros2/rclpy/blob/52f2a69325721e1368ac3176adcc348e1f53f93e/rclpy/src/rclpy/node.cpp#L463

That is not the case for Jazzy:
https://github.com/ros2/rclpy/blob/e181737f19d6b8cb335b0326a40ec2075267c555/rclpy/src/rclpy/node.cpp#L503

This was likely changed in rclpy by:
https://github.com/ros2/rclpy/pull/1121
https://github.com/ros2/rclpy/commit/827d59729f266d9449d8fcb9752707bbd052527d

That pull request was to address a similar kind of issue, but in our case a different set of RMW specific locks:
https://github.com/ros2/rclcpp/issues/2147

hwoithe, 2024-06-05:
As mentioned above, in Iron, a node loading a component in the bash for loop will eventually experience a hang. In Jazzy, the component_container node will eventually experiences a hang.

```
$ sudo gdb -p `pgrep -f 'rclcpp_components/component_container'`
...
(gdb) info threads
  Id   Target Id                                             Frame 
* 1    Thread 0x7f557abe18c0 (LWP 3525805) "component_conta" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  2    Thread 0x7f5575f746c0 (LWP 3525848) "rx-2"            futex_wait (private=0, expected=2, futex_word=0x55e65761d250) at ../sysdeps/nptl/futex-internal.h:146
  3    Thread 0x7f557696b6c0 (LWP 3525819) "component_conta" 0x00007f557b10cd61 in __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55e65761eb78) at ./nptl/futex-internal.c:57
  4    Thread 0x7f557716c6c0 (LWP 3525818) "component_conta" 0x00007f557b10cd61 in __futex_abstimed_wait_common64 (private=1997974368, cancel=true, abstime=0x7f557716acd0, op=137, expected=0, futex_word=0x55e65761d4c0) at ./nptl/futex-internal.c:57
  5    Thread 0x7f557736d6c0 (LWP 3525817) "rx-1"            syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  6    Thread 0x7f557756e6c0 (LWP 3525816) "rx-0"            syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  7    Thread 0x7f557776f6c0 (LWP 3525815) "tx-0"            0x00007f557b19e042 in epoll_wait (epfd=16, events=0x7f556800a140, maxevents=1024, timeout=1894) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7f5577b716c0 (LWP 3525810) "acc-0"           syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  9    Thread 0x7f5577d726c0 (LWP 3525809) "net-0"           0x00007f557b19e042 in epoll_wait (epfd=8, events=0x55e65760fb10, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  10   Thread 0x7f5577f736c0 (LWP 3525808) "app-0"           0x00007f557b19e042 in epoll_wait (epfd=3, events=0x55e65760a580, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  11   Thread 0x7f557a3b56c0 (LWP 3525807) "component_c-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  12   Thread 0x7f557abb66c0 (LWP 3525806) "component_c-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
(gdb) thread 1
[Switching to thread 1 (Thread 0x7f557abe18c0 (LWP 3525805))]
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
38  in ../sysdeps/unix/sysv/linux/x86_64/syscall.S
(gdb) bt
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007f557816ba20 in std::sys::pal::unix::futex::futex_wait () at library/std/src/sys/pal/unix/futex.rs:62
#2  std::sys::sync::rwlock::futex::RwLock::read_contended () at library/std/src/sys/sync/rwlock/futex.rs:137
#3  0x00007f557838adc9 in std::sys::sync::rwlock::futex::RwLock::read (self=0x55e65760ee10) at /rustc/9b00956e56009bab2aa15d7bff10916599e3d6d6/library/std/src/sys/sync/rwlock/futex.rs:86
#4  0x00007f5578590115 in std::sync::rwlock::RwLock<zenoh::net::routing::dispatcher::tables::Tables>::read<zenoh::net::routing::dispatcher::tables::Tables> (self=0x55e65760ee10) at /rustc/9b00956e56009bab2aa15d7bff10916599e3d6d6/library/std/src/sync/rwlock.rs:273
#5  0x00007f5578628e4b in zenoh::net::routing::dispatcher::pubsub::full_reentrant_route_data (tables_ref=0x55e65760a410, face=0x55e65760a2d0, expr=0x7ffedb5ce938, ext_qos=..., ext_tstamp=..., payload=..., routing_context=0) at src/net/routing/dispatcher/pubsub.rs:436
#6  0x00007f557848a3a8 in zenoh::net::routing::dispatcher::face::{impl#4}::send_push (self=0x55e65760a410, msg=...) at src/net/routing/dispatcher/face.rs:246
#7  0x00007f557853da7d in zenoh::publication::resolve_put (publisher=0x55e657839110, value=..., kind=zenoh_protocol::core::SampleKind::Put, attachment=...) at src/publication.rs:841
#8  0x00007f557853bcb6 in zenoh::publication::{impl#13}::res_sync (self=...) at src/publication.rs:653
#9  0x00007f55781e8ef7 in zenohcd::publisher::z_publisher_put (publisher=..., payload=0x55e657bf72b0, len=200, options=...) at <***removed***>/ws_rmw_zenoh/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/publisher.rs:274
#10 0x00007f5579b5fbe9 in rmw_publish (publisher=0x55e6577e8d30, ros_message=0x7ffedb5d0240, allocation=0x0) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:945
#11 0x00007f557b025a15 in rcl_publish (publisher=0x7ffedb5d01e8, ros_message=0x7ffedb5d0240, allocation=0x0) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/publisher.c:283
#12 0x00007f557b01de82 in rcl_logging_rosout_output_handler (location=0x7f55760bb7b0 <composition::Talker::on_timer()::__rcutils_logging_location>, severity=20, name=0x55e6577d6280 "talker7.talker", timestamp=1717591806607772690, format=0x7f557609a644 "Publishing: '%s'", args=0x7ffedb5d2050) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/logging_rosout.c:354
#13 0x00007f557b01f0fe in rcl_logging_multiple_output_handler (location=0x7f55760bb7b0 <composition::Talker::on_timer()::__rcutils_logging_location>, severity=20, name=0x55e6577d6280 "talker7.talker", timestamp=1717591806607772690, format=0x7f557609a644 "Publishing: '%s'", args=0x7ffedb5d2050) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/logging.c:162
[... cut ...]
https://github.com/ros2/rclcpp/blob/0beba9715675e06bf63f041fd1046b2f23d3b3ce/rclcpp/src/rclcpp/context.cpp#L133

We can see that thread 1 is also in ```rclcpp_logging_output_handler()``` (frame 14) and has acquired ```logging_mutex``` before calling ```rcl_logging_multiple_output_handler()``` (frame 13).
https://github.com/ros2/rclcpp/blob/0beba9715675e06bf63f041fd1046b2f23d3b3ce/rclcpp/src/rclcpp/context.cpp#L134

rmw_zenoh: rolling (66ec842030fda0cd381835f8ed2577f331fb5729)
rclcpp: jazzy (0beba9715675e06bf63f041fd1046b2f23d3b3ce)
zenoh: 763a05f0d61a894cf3cbeb07223e2ea726777325

JEnoch, 2024-06-06:
I don't think the problem is related to #170 or to python.
In both the provided stacks the pattern is the same:
- the 2nd thread is reacting from an incoming Zenoh control message leading to call the liveliness subscription callback (zenohcd::liveliness::zc_liveliness_declare_subscriber::{closure#0}`, while retaining a lock in the routing layer (either `ctrl_lock ` or `tables.tables`). The callback is in `rmw_zenoh` and is updating the ROS graph, leading probably to a publication on `rosout`. Hence requiring the `logging_mutex `. 
- the 1st thread is in the other way around: during a call to `rcl_logging` the `logging_mutex` is taken and Zenoh is called for some declaration (`zenoh::session::Session::declare_queryable_inner `) or some publication (`zenoh::publication::resolve_put `). Hence requiring a lock in the routing layer which is taken by thread 2.

So the root of the problem is on Zenoh control messages (e.g. liveliness events) leading to call a user callback, Zenoh is keeping an internal lock that is blocking any other thread trying to send a message with Zenoh.
We'll investigate a solution to this issue in https://github.com/eclipse-zenoh/zenoh/issues/1092

@Yadunund in the meanwhile, a workaround could be to make sure no mutex is taken in liveliness callbacks.
Do you think deactivating the logging in the graph cache is possible and would be sufficient ?

Yadunund, 2024-06-07:
> @Yadunund in the meanwhile, a workaround could be to make sure no mutex is taken in liveliness callbacks.
Do you think deactivating the logging in the graph cache is possible and would be sufficient ?

As discussed yesterday, It is a bit strange that logging from the `rmw` layer is publishing to `rosout`. I will look into this.

JEnoch, 2024-06-08:
> As discussed yesterday, It is a bit strange that logging from the rmw layer is publishing to rosout. I will look into this.

Actually I was wrong, sorry for the confusion... I was confused with frame #13 in first stack:
`#13 0x00007f13aa25e0bb in rcl_logging_rosout_init_publisher_for_node (node=0x556946f65120) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/logging_rosout.c:276`
Which is probably be just the initialisation of `rosout` Publisher, and leads to creation of a Zenoh publisher, hence waiting a lock in routing layer.
Still this thread is holding the `logging_mutex`. And that's the common point with other stacks, even if publication to `rosout` is not occurring.

Yadunund, 2024-06-10:
That's good to know. Shall wait for the fix for https://github.com/eclipse-zenoh/zenoh/issues/1092.

JEnoch, 2024-06-10:
We have a fix candidate in a branch: https://github.com/eclipse-zenoh/zenoh/commits/dev/reentrant_declares/
#200 is tracking this branch.

@hwoithe I reproduced the same issue with your scenario, and switching to #200 fixes the issue for me.
Could you please test and confirm ?

hwoithe, 2024-06-10:
I am a bit short on time today, but I was able to quickly take a look and I am still experiencing the same problem under Jazzy where the ```component_container``` eventually hangs. Overall, it appears quite similar to the Jazzy test above.
https://github.com/ros2/rmw_zenoh/issues/182#issuecomment-2150633477

I have not yet retested Iron.

A quick debug session:
```
$ sudo gdb -p `pgrep -f 'rclcpp_components/component_container'`
...
(gdb) info threads
  Id   Target Id                                             Frame 
* 1    Thread 0x7fe64b65b8c0 (LWP 3416556) "component_conta" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  2    Thread 0x7fe63a2056c0 (LWP 3500878) "net-2"           syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  3    Thread 0x7fe63affa6c0 (LWP 3416696) "component_conta" 0x00007fe64bb86d61 in __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x559ddd95d368) at ./nptl/futex-internal.c:57
  4    Thread 0x7fe63b7fb6c0 (LWP 3416695) "component_conta" 0x00007fe64bb86d61 in __futex_abstimed_wait_common64 (private=998218592, cancel=true, abstime=0x7fe63b7f9cd0, op=137, expected=0, futex_word=0x559ddd95cef0) at ./nptl/futex-internal.c:57
  5    Thread 0x7fe63b9fc6c0 (LWP 3416584) "rx-1"            futex_wait (private=0, expected=2, futex_word=0x559ddd95b6f0) at ../sysdeps/nptl/futex-internal.h:146
  6    Thread 0x7fe63bbfd6c0 (LWP 3416583) "rx-0"            syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  7    Thread 0x7fe63bdfe6c0 (LWP 3416582) "tx-0"            0x00007fe64bc18042 in epoll_wait (epfd=16, events=0x7fe63c00a220, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7fe64839e6c0 (LWP 3416579) "acc-0"           syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  9    Thread 0x7fe6487a06c0 (LWP 3416577) "net-0"           syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  10   Thread 0x7fe6489a16c0 (LWP 3416576) "app-0"           0x00007fe64bc18042 in epoll_wait (epfd=3, events=0x559ddd948200, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  11   Thread 0x7fe64ae2f6c0 (LWP 3416575) "component_c-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  12   Thread 0x7fe64b6306c0 (LWP 3416574) "component_c-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
...
(gdb) thread 1
[Switching to thread 1 (Thread 0x7fe64b65b8c0 (LWP 3416556))]
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
38  in ../sysdeps/unix/sysv/linux/x86_64/syscall.S
(gdb) bt
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007fe648b9ca80 in std::sys::pal::unix::futex::futex_wait () at library/std/src/sys/pal/unix/futex.rs:62
#2  std::sys::pal::unix::locks::futex_rwlock::RwLock::read_contended () at library/std/src/sys/pal/unix/locks/futex_rwlock.rs:137
#3  0x00007fe649062a17 in std::sys::pal::unix::locks::futex_rwlock::RwLock::read (self=0x559ddd94cb50) at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/sys/pal/unix/locks/futex_rwlock.rs:86
#4  0x00007fe648dc8805 in std::sync::rwlock::RwLock<zenoh::net::routing::dispatcher::tables::Tables>::read<zenoh::net::routing::dispatcher::tables::Tables> (self=0x559ddd94cb50) at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/sync/rwlock.rs:210
#5  0x00007fe648f313ab in zenoh::net::routing::dispatcher::pubsub::full_reentrant_route_data (tables_ref=0x559ddd947a30, face=0x559ddd953d40, expr=0x7ffc82822868, ext_qos=..., ext_tstamp=..., payload=..., routing_context=0) at src/net/routing/dispatcher/pubsub.rs:451
#6  0x00007fe648e574e8 in zenoh::net::routing::dispatcher::face::{impl#4}::send_push (self=0x559ddd947a30, msg=...) at src/net/routing/dispatcher/face.rs:269
#7  0x00007fe6490b9f13 in zenoh::publication::resolve_put (publisher=0x559dddc42d90, value=..., kind=zenoh_protocol::core::SampleKind::Put, attachment=...) at src/publication.rs:841
#8  0x00007fe6490b8136 in zenoh::publication::{impl#13}::res_sync (self=...) at src/publication.rs:653
#9  0x00007fe648c28247 in zenohcd::publisher::z_publisher_put (publisher=..., payload=0x559dde1a98a0, len=208, options=...) at <***removed***>/ws_rmw_zenoh/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/publisher.rs:274
#10 0x00007fe64a5d9be9 in rmw_publish (publisher=0x559dddf46d90, ros_message=0x7ffc82824170, allocation=0x0) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:945
#11 0x00007fe64ba9fa15 in rcl_publish (publisher=0x7ffc82824118, ros_message=0x7ffc82824170, allocation=0x0) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/publisher.c:283
#12 0x00007fe64ba97e82 in rcl_logging_rosout_output_handler (location=0x7fe6480e87b0 <composition::Talker::on_timer()::__rcutils_logging_location>, severity=20, name=0x559dddf3dc70 "talker19.talker", timestamp=1718042153752681058, format=0x7fe6480c7651 "Publishing: '%s'", args=0x7ffc82825f80) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/logging_rosout.c:354
#13 0x00007fe64ba990fe in rcl_logging_multiple_output_handler (location=0x7fe6480e87b0 <composition::Talker::on_timer()::__rcutils_logging_location>, severity=20, name=0x559dddf3dc70 "talker19.talker", timestamp=1718042153752681058, format=0x7fe6480c7651 "Publishing: '%s'", args=0x7ffc82825f80) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/logging.c:162
[... cut ...]
https://github.com/ros2/rmw_zenoh/pull/200/commits/614954c248bdb7e528dca116512279edf7937a12#diff-7d5bcea24badbde5330943e2f5ff4c2234863cb59a9f75c049d1b659620d0ef9L24

My environment:

```
$ echo $ROS_DISTRO
jazzy
$ ls -1 ~/.cargo/git/checkouts/
zenoh-cc237f2570fab813
$ ls -1 ~/.cargo/git/checkouts/zenoh-cc237f2570fab813/
d070e7c
$ rustc -V
rustc 1.77.2 (25ef9e3d8 2024-04-09)
$ lsb_release -a
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:    24.04
Codename:   noble
```

rclcpp: 8ab10aabc0304fe2a26f54a67274cdbc51b65602

JEnoch, 2024-06-11:
Thanks for your testing and quick feedback. We'll analyse your backtraces.

> I don't know the codebase well enough yet and haven't found if Thread 5 is supposed to drop the write lock and/or `ctrl_lock` further in the call chain.

The purpose of the patch is precisely to release the `ctrl_lock` before propagating the liveliness message to the callbacks in `rmw_zenoh`. So either the patch is not complete, either another lock is in cause.
@OlivierHecart , isn't the problem in this line which still takes the `ctrl_lock`  while calling `wend_declare()`:
https://github.com/eclipse-zenoh/zenoh/blob/d070e7ce777222692d34f466c0f1a95f0de2f068/zenoh/src/net/routing/dispatcher/tables.rs#L177

> Also, is this referencing the correct issue number?
https://github.com/ros2/rmw_zenoh/commit/614954c248bdb7e528dca116512279edf7937a12#diff-7d5bcea24badbde5330943e2f5ff4c2234863cb59a9f75c049d1b659620d0ef9L24

It isn't since this patch is not in a pull request yet. I will update the comment once we have a zenoh PR number for it.

hwoithe, 2024-06-11:
> It isn't since this patch is not in a pull request yet. I will update the comment once we have a zenoh PR number for it.

Sounds great! I just wanted to make sure, since ```https://github.com/ros2/rmw_zenoh/issues/198``` refers to a benchmarking issue. Thank you for the quick response.

JEnoch, 2024-06-11:
There was indeed some oversights in `ctrl_lock` dropping. Those are now fixed in https://github.com/eclipse-zenoh/zenoh/commit/5191af258a1908dd2f28601ef4410554e97bb7e9.
I updated #200 to use https://github.com/eclipse-zenoh/zenoh-c/commit/87b515b4c3dd028f9e9476e3bb60016f7acc3489 which depends on https://github.com/eclipse-zenoh/zenoh/commit/5191af258a1908dd2f28601ef4410554e97bb7e9.
 
 I tested your scenario several times with this and I still don't experience any deadlock.
 Hopefully it will work for you now! 🤞 😅

hwoithe, 2024-06-12:
@JEnoch I am now able to make more progress with https://github.com/ros2/rmw_zenoh/pull/200. Thank you! Unfortunately, I'm still experiencing two scenarios which I will describe in separate comments. They do not seem to be Zenoh related.

hwoithe, 2024-06-12:
This is in Jazzy with rmw_zenow f697984f66373916bdc856f1acd213de16f53689 (https://github.com/ros2/rmw_zenoh/pull/200). This particular problem happens less often. The deadlock happens to a node loading a component in the bash for loop.

Debugging the node:
```
$ sudo gdb attach -p 370800
...
(gdb) info threads
  Id   Target Id                                     Frame 
* 1    Thread 0x7f5d575d5b80 (LWP 370800) "ros2"     futex_wait (private=0, expected=2, futex_word=0x1ce2598) at ../sysdeps/nptl/futex-internal.h:146
  2    Thread 0x7f5d46bf86c0 (LWP 370812) "ros2"     0x00007f5d5766fd61 in __futex_abstimed_wait_common64 (private=1186954624, cancel=true, abstime=0x7f5d46bf7e50, op=137, expected=0, futex_word=0x2628d00) at ./nptl/futex-internal.c:57
  3    Thread 0x7f5d46df96c0 (LWP 370811) "rx-1"     futex_wait (private=0, expected=2, futex_word=0x1ce2598) at ../sysdeps/nptl/futex-internal.h:146
  4    Thread 0x7f5d46ffa6c0 (LWP 370810) "rx-0"     futex_wait (private=0, expected=2, futex_word=0x7f5d567acde0 <rclpy::LoggingGuard::logging_mutex_>) at ../sysdeps/nptl/futex-internal.h:146
  5    Thread 0x7f5d471fb6c0 (LWP 370809) "tx-0"     0x00007f5d57701042 in epoll_wait (epfd=16, events=0x7f5d3800a410, maxevents=1024, timeout=2049) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  6    Thread 0x7f5d475fd6c0 (LWP 370807) "acc-0"    0x00007f5d57701042 in epoll_wait (epfd=12, events=0x2396020, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  7    Thread 0x7f5d4c3056c0 (LWP 370805) "net-0"    0x00007f5d57701042 in epoll_wait (epfd=8, events=0x2386360, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7f5d4c5066c0 (LWP 370804) "app-0"    0x00007f5d57701042 in epoll_wait (epfd=3, events=0x2383350, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  9    Thread 0x7f5d47fff6c0 (LWP 370803) "ros2"     0x00007f5d5766fd61 in __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7f5d567ad580 <(anonymous namespace)::g_signal_handler_sem>) at ./nptl/futex-internal.c:57
  10   Thread 0x7f5d54d846c0 (LWP 370802) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  11   Thread 0x7f5d555856c0 (LWP 370801) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38

(gdb) thread 1
[Switching to thread 1 (Thread 0x7f5d575d5b80 (LWP 370800))]
#0  futex_wait (private=0, expected=2, futex_word=0x1ce2598) at ../sysdeps/nptl/futex-internal.h:146
146 in ../sysdeps/nptl/futex-internal.h
(gdb) bt
#0  futex_wait (private=0, expected=2, futex_word=0x1ce2598) at ../sysdeps/nptl/futex-internal.h:146
#1  __GI___lll_lock_wait (futex=futex@entry=0x1ce2598, private=0) at ./nptl/lowlevellock.c:49
#2  0x00007f5d576770f1 in lll_mutex_lock_optimized (mutex=0x1ce2598) at ./nptl/pthread_mutex_lock.c:48
#3  ___pthread_mutex_lock (mutex=0x1ce2598) at ./nptl/pthread_mutex_lock.c:93
#4  0x00007f5d544cfc52 in __gthread_mutex_lock (__mutex=0x1ce2598) at /usr/include/x86_64-linux-gnu/c++/13/bits/gthr-default.h:749
#5  0x00007f5d544cfca6 in std::mutex::lock (this=0x1ce2598) at /usr/include/c++/13/bits/std_mutex.h:113
#6  0x00007f5d544cfe88 in std::lock_guard<std::mutex>::lock_guard (this=0x7fff9cb6c1d8, __m=...) at /usr/include/c++/13/bits/std_mutex.h:249
#7  0x00007f5d544d5eb5 in rmw_zenoh_cpp::GraphCache::parse_put (this=0x1ce2430, keyexpr="@ros2_lv/0/46f1237e44d9ea87e3aa8bf7ca9ccc/0/1/MP/%/_ros2cli_370800/%rosout/rcl_interfaces::msg::dds_::Log_/1:1:1,1000", ignore_from_current_session=false) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/graph_cache.cpp:347
#8  0x00007f5d545255fc in graph_sub_data_handler (sample=0x7fff9cb6d6a0, data=0x2363240) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_init.cpp:70
#9  0x00007f5d4e76a91d in zenohcd::closures::sample_closure::z_closure_sample_call (closure=0x23bf0f0, sample=0x7fff9cb6d6a0) at <***removed***>/ws_rmw_zenoh/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/closures/sample_closure.rs:52
#10 0x00007f5d4e723d71 in zenohcd::liveliness::zc_liveliness_declare_subscriber::{closure#0} (sample=...) at <***removed***>/ws_rmw_zenoh/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/liveliness.rs:186
#11 0x00007f5d4e921558 in zenoh::session::Session::handle_data (self=0x1e16380, local=false, key_expr=0x7fff9cb70b00, info=..., payload=..., attachment=...) at src/session.rs:1712
#12 0x00007f5d4e92ab20 in zenoh::session::{impl#13}::send_declare (self=0x1e16380, msg=...) at src/session.rs:2158
#13 0x00007f5d4e9372a4 in zenoh::session::{impl#16}::send_declare (self=0x1e16380, ctx=...) at src/session.rs:2701
#14 0x00007f5d4e9b9af6 in zenoh::net::routing::dispatcher::face::{impl#4}::send_declare (self=0x1d65740, msg=...) at src/net/routing/dispatcher/face.rs:210
#15 0x00007f5d4e961020 in zenoh::session::Session::declare_liveliness_inner (self=0x222fc40, key_expr=0x7fff9cb73760) at src/session.rs:1385
#16 0x00007f5d4e7422aa in zenoh::liveliness::{impl#10}::res_sync (self=...) at <***removed***>/.cargo/git/checkouts/zenoh-cc237f2570fab813/5191af2/zenoh/src/liveliness.rs:241
#17 0x00007f5d4e6ed6ed in zenoh_core::SyncResolve::res<zenoh::liveliness::LivelinessTokenBuilder> (self=<error reading variable: Cannot access memory at address 0x80>) at <***removed***>/.cargo/git/checkouts/zenoh-cc237f2570fab813/5191af2/commons/zenoh-core/src/lib.rs:57
[... cut ...]
Thread 4 has acquired ```graph_mutex_``` in frame 12 early in the ```parse_del()``` function:
https://github.com/ros2/rmw_zenoh/blob/f697984f66373916bdc856f1acd213de16f53689/rmw_zenoh_cpp/src/detail/graph_cache.cpp#L564

Thread 1 has acquired ```scoped_logging_guard``` in frame 20:
https://github.com/ros2/rclpy/blob/e181737f19d6b8cb335b0326a40ec2075267c555/rclpy/src/rclpy/node.cpp#L521

rmw_zenoh: f697984f66373916bdc856f1acd213de16f53689 (https://github.com/ros2/rmw_zenoh/pull/200)
rclpy: e181737f19d6b8cb335b0326a40ec2075267c555

hwoithe, 2024-06-12:
This is the second scenario in Jazzy with rmw_zenoh f697984f66373916bdc856f1acd213de16f53689 (https://github.com/ros2/rmw_zenoh/pull/200) where a node loading a component in the bash for loop hangs. It gets stuck in ```rmw_wait()```. I'll have to check this against the other ```rmw_wait()``` related issues and pull requests that are currently open to see how closely they match to this.

```
$ sudo gdb attach -p `pgrep -f talker80`
...
(gdb) info threads
  Id   Target Id                                     Frame 
* 1    Thread 0x7f8e1a5afb80 (LWP 966966) "ros2"     0x00007f8e1a649d61 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x3279be8) at ./nptl/futex-internal.c:57
  2    Thread 0x7f8e0db106c0 (LWP 966978) "ros2"     0x00007f8e1a649d61 in __futex_abstimed_wait_common64 (private=229703040, cancel=true, abstime=0x7f8e0db0fe50, op=137, expected=0, futex_word=0x31462f0) at ./nptl/futex-internal.c:57
  3    Thread 0x7f8e0de9a6c0 (LWP 966977) "rx-1"     0x00007f8e1a6db042 in epoll_wait (epfd=19, events=0x7f8dfc131e10, maxevents=1024, timeout=6753) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  4    Thread 0x7f8e0e09b6c0 (LWP 966976) "rx-0"     syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  5    Thread 0x7f8e0e29c6c0 (LWP 966975) "tx-0"     0x00007f8e1a6db042 in epoll_wait (epfd=16, events=0x7f8dfc00a290, maxevents=1024, timeout=2009) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  6    Thread 0x7f8e0e69e6c0 (LWP 966973) "acc-0"    0x00007f8e1a6db042 in epoll_wait (epfd=12, events=0x30f1650, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  7    Thread 0x7f8e0eaa06c0 (LWP 966971) "net-0"    0x00007f8e1a6db042 in epoll_wait (epfd=8, events=0x3103180, maxevents=1024, timeout=2147483647) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7f8e0eca16c0 (LWP 966970) "app-0"    0x00007f8e1a6db042 in epoll_wait (epfd=3, events=0x3100170, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  9    Thread 0x7f8e0f4a26c0 (LWP 966969) "ros2"     0x00007f8e1a649d61 in __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7f8e19789580 <(anonymous namespace)::g_signal_handler_sem>) at ./nptl/futex-internal.c:57
  10   Thread 0x7f8e17d616c0 (LWP 966968) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  11   Thread 0x7f8e185626c0 (LWP 966967) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
...
(gdb) bt
#0  0x00007f8e1a649d61 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x3279be8) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=0, abstime=0x0, clockid=0, expected=0, futex_word=0x3279be8) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x3279be8, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0) at ./nptl/futex-internal.c:139
#3  0x00007f8e1a64c7dd in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x3279bf0, cond=0x3279bc0) at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=0x3279bc0, mutex=0x3279bf0) at ./nptl/pthread_cond_wait.c:627
#5  0x00007f8e175194c5 in rmw_wait (subscriptions=0x32c0308, guard_conditions=0x32c0320, services=0x32c0350, clients=0x32c0338, events=0x32c0368, wait_set=0x32ce6a0, wait_timeout=0x0) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:3334
#6  0x00007f8e193e1b1b in rcl_wait (wait_set=0x3298600, timeout=-1) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/wait.c:670
#7  0x00007f8e1966999d in rclpy::WaitSet::wait (this=0x325d020, timeout=-1) at <***removed***>/ws_rmw_zenoh/src/rclpy/rclpy/src/rclpy/wait_set.cpp:247
#8  0x00007f8e1966e08d in pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}::operator()(rclpy::WaitSet*, long) const (__closure=0x2b227c8, c=0x325d020, args#0=-1) at /usr/include/pybind11/pybind11.h:111
#9  0x00007f8e1967387b in pybind11::detail::argument_loader<rclpy::WaitSet*, long>::call_impl<void, pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&, 0ul, 1ul, pybind11::detail::void_type>(pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&, std::integer_sequence<unsigned long, 0ul, 1ul>, pybind11::detail::void_type&&) && (this=0x7ffd239b50b0, f=...) at /usr/include/pybind11/cast.h:1480
#10 0x00007f8e19671da4 in pybind11::detail::argument_loader<rclpy::WaitSet*, long>::call<void, pybind11::detail::void_type, pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)( [... cut ...]
    return extension.main(args=args)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/jazzy/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.1', 'console_scripts', 'ros2')())
```

rcl: 00c29cd6e72caae36e1e749a8c02558ee709a275
rclpy: e181737f19d6b8cb335b0326a40ec2075267c555
rmw_zenoh: f697984f66373916bdc856f1acd213de16f53689 (https://github.com/ros2/rmw_zenoh/pull/200)

Yadunund, 2024-06-14:
I've also tested the scenario several times with #200 + Rolling and I can't reproduce the problem anymore.

It could be that something changed between Rolling and Jazzy too? I'm looking into that now.

@JEnoch could we merge the relevant upstream zenoh PRs and update the hash in #200 accordingly so that we can get that in?

JEnoch, 2024-06-18:
@Yadunund done! #200 is ready for review and merge.

hwoithe, 2024-06-19:
I have not yet retested in Iron.  In Jazzy, I am still experiencing deadlocks with the graph mutex and a logging mutex similar to https://github.com/ros2/rmw_zenoh/issues/182#issuecomment-2161919040. It seems to be easier to trigger if we use ```parallel```. In the debugging session below, the component_container node deadlocks which means the logging mutex is in ```rclcpp``` instead of ```rclpy```.

@Yadunund or @clalancette, could you please retest this with the ```parallel``` method in Jazzy?

Terminal 1:
```
ros2 run rmw_zenoh_cpp rmw_zenohd
```

Terminal 2:
```
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
ros2 run rclcpp_components component_container
```

Terminal 3 (now uses ```parallel```):
```
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
for i in {1..100}; do echo "ros2 component load --node-namespace /talker${i} /ComponentManager composition composition::Talker"; done | parallel
```

Debugging session:
```
(gdb) info threads
  Id   Target Id                                            Frame 
* 1    Thread 0x7f90e07c18c0 (LWP 610724) "component_conta" futex_wait (private=0, expected=2, futex_word=0x55d660c278f8) at ../sysdeps/nptl/futex-internal.h:146
  2    Thread 0x7f90bf7fe6c0 (LWP 611779) "rx-2"            futex_wait (private=0, expected=2, futex_word=0x55d660c3bd00) at ../sysdeps/nptl/futex-internal.h:146
  3    Thread 0x7f90bffff6c0 (LWP 610736) "component_conta" 0x00007f90e0cecd61 in __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55d660c3d908) at ./nptl/futex-internal.c:57
  4    Thread 0x7f90dcaea6c0 (LWP 610735) "component_conta" 0x00007f90e0cecd61 in __futex_abstimed_wait_common64 (private=-592540832, cancel=true, abstime=0x7f90dcae8cd0, op=137, expected=0, futex_word=0x55d660c3d3e0) at ./nptl/futex-internal.c:57
  5    Thread 0x7f90dcceb6c0 (LWP 610734) "rx-1"            futex_wait (private=0, expected=2, futex_word=0x55d660c278f8) at ../sysdeps/nptl/futex-internal.h:146
  6    Thread 0x7f90dd0ed6c0 (LWP 610732) "tx-0"            0x00007f90e0d7e042 in epoll_wait (epfd=16, events=0x7f90d0009880, maxevents=1024, timeout=505) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  7    Thread 0x7f90dd4ef6c0 (LWP 610730) "acc-0"           0x00007f90e0d7e042 in epoll_wait (epfd=12, events=0x55d660c348b0, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7f90dd8f16c0 (LWP 610728) "net-0"           futex_wait (private=0, expected=2, futex_word=0x55d660c278f8) at ../sysdeps/nptl/futex-internal.h:146
  9    Thread 0x7f90ddaf26c0 (LWP 610727) "app-0"           0x00007f90e0d7e042 in epoll_wait (epfd=3, events=0x55d660c27ec0, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  10   Thread 0x7f90dff956c0 (LWP 610726) "component_c-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  11   Thread 0x7f90e07966c0 (LWP 610725) "component_c-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38

(gdb) thread 1
[Switching to thread 1 (Thread 0x7f90e07c18c0 (LWP 610724))]
#0  futex_wait (private=0, expected=2, futex_word=0x55d660c278f8) at ../sysdeps/nptl/futex-internal.h:146
warning: 146    ../sysdeps/nptl/futex-internal.h: No such file or directory
(gdb) bt
#0  futex_wait (private=0, expected=2, futex_word=0x55d660c278f8) at ../sysdeps/nptl/futex-internal.h:146
#1  __GI___lll_lock_wait (futex=futex@entry=0x55d660c278f8, private=0) at ./nptl/lowlevellock.c:49
#2  0x00007f90e0cf40f1 in lll_mutex_lock_optimized (mutex=0x55d660c278f8) at ./nptl/pthread_mutex_lock.c:48
#3  ___pthread_mutex_lock (mutex=0x55d660c278f8) at ./nptl/pthread_mutex_lock.c:93
#4  0x00007f90df6e0c52 in __gthread_mutex_lock (__mutex=0x55d660c278f8) at /usr/include/x86_64-linux-gnu/c++/13/bits/gthr-default.h:749
#5  0x00007f90df6e0ca6 in std::mutex::lock (this=0x55d660c278f8) at /usr/include/c++/13/bits/std_mutex.h:113
#6  0x00007f90df6e0e88 in std::lock_guard<std::mutex>::lock_guard (this=0x7ffdab924e08, __m=...) at /usr/include/c++/13/bits/std_mutex.h:249
#7  0x00007f90df6e6eb5 in rmw_zenoh_cpp::GraphCache::parse_put (this=0x55d660c27790, keyexpr="@ros2_lv/0/70eb82fae64d6e3095520311f8d33d/271/272/MP/%talker24/talker/%rosout/rcl_interfaces::msg::dds_::Log_/1:1:1,1000", ignore_from_current_session=false) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/graph_cache.cpp:347
#8  0x00007f90df7365fc in graph_sub_data_handler (sample=0x7ffdab9262d0, data=0x55d660c25e30) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_init.cpp:70
#9  0x00007f90ddd06dcd in zenohcd::closures::sample_closure::z_closure_sample_call (closure=0x55d660c3eda0, sample=0x7ffdab9262d0) at <***removed***>/ws_rmw_zenoh/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/closures/sample_closure.rs:52
#10 0x00007f90ddd80e71 in zenohcd::liveliness::zc_liveliness_declare_subscriber::{closure#0} (sample=...) at <***removed***>/ws_rmw_zenoh/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/liveliness.rs:186
#11 0x00007f90ddf721c8 in zenoh::session::Session::handle_data (self=0x55d660c338c0, local=false, key_expr=0x7ffdab929730, info=..., payload=..., attachment=...) at src/session.rs:1712
#12 0x00007f90ddf7b790 in zenoh::session::{impl#13}::send_declare (self=0x55d660c338c0, msg=...) at src/session.rs:2158
#13 0x00007f90ddf87f14 in zenoh::session::{impl#16}::send_declare (self=0x55d660c338c0, ctx=...) at src/session.rs:2701
[... cut ...]
Thread 2 in frame 8 wants to acquire ```logging_mutex```:
https://github.com/ros2/rclcpp/blob/7f5b44ebe22d269d329d85b678f82ae96d66ad20/rclcpp/src/rclcpp/context.cpp#L133

Thread 2 has acquired ```graph_mutex_``` in frame 11 earlier in ```parse_del()```:
https://github.com/ros2/rmw_zenoh/blob/d3396a59f7b005e152244c212178c43951b2b113/rmw_zenoh_cpp/src/detail/graph_cache.cpp#L564

rmw_zenoh: d3396a59f7b005e152244c212178c43951b2b113
rclcpp: 7f5b44ebe22d269d329d85b678f82ae96d66ad20
rcl: 00c29cd6e72caae36e1e749a8c02558ee709a275

Yadunund, 2024-06-20:
@hwoithe I was able to successfully run `for i in {1..100}; do echo "ros2 component load --node-namespace /talker${i} /ComponentManager composition composition::Talker"; done | parallel` on Jazzy with #203 branch (It takes a while to complete) Could you please retry with this branch?

Yadunund, 2024-06-20:
Ah I am able to reproduce it in subsequent tries.

However if I add a sleep in between, I succeed each time.
```
for i in {1..100}; do echo "ros2 component load --node-namespace /talker${i} /ComponentManager composition composition::Talker"; sleep 1; done | parallel
```

hwoithe, 2024-06-20:
In general, I now run the parallel version to reproduce the graph lock issue and the non-parallel version to reproduce the rmw_wait issue. I am still experiencing a hang on rmw_wait from a node in the for loop, even with https://github.com/ros2/rmw_zenoh/pull/203.

With https://github.com/ros2/rmw_zenoh/pull/203:
```
(gdb) info threads
  Id   Target Id                                     Frame 
* 1    Thread 0x7f3b6c04ab80 (LWP 954695) "ros2"     0x00007f3b6c0e4d61 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x2f4cd70) at ./nptl/futex-internal.c:57
  2    Thread 0x7f3b4f3886c0 (LWP 954708) "ros2"     0x00007f3b6c0e4d61 in __futex_abstimed_wait_common64 (private=1329102208, cancel=true, abstime=0x7f3b4f387e50, op=137, expected=0, futex_word=0x31c1110) at ./nptl/futex-internal.c:57
  3    Thread 0x7f3b4f7fb6c0 (LWP 954707) "rx-1"     syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  4    Thread 0x7f3b4f9fc6c0 (LWP 954706) "rx-0"     0x00007f3b6c176042 in epoll_wait (epfd=19, events=0x7f3b50131d10, maxevents=1024, timeout=3732) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  5    Thread 0x7f3b4fbfd6c0 (LWP 954705) "tx-0"     0x00007f3b6c176042 in epoll_wait (epfd=16, events=0x7f3b50009f10, maxevents=1024, timeout=2387) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  6    Thread 0x7f3b4ffff6c0 (LWP 954703) "acc-0"    0x00007f3b6c176042 in epoll_wait (epfd=12, events=0x309c2d0, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  7    Thread 0x7f3b6055a6c0 (LWP 954701) "net-0"    0x00007f3b6c176042 in epoll_wait (epfd=8, events=0x2fbd190, maxevents=1024, timeout=2147483647) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  8    Thread 0x7f3b6075b6c0 (LWP 954700) "app-0"    0x00007f3b6c176042 in epoll_wait (epfd=3, events=0x2fec5f0, maxevents=1024, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  9    Thread 0x7f3b60f5c6c0 (LWP 954698) "ros2"     0x00007f3b6c0e4d61 in __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7f3b6b222580 <(anonymous namespace)::g_signal_handler_sem>) at ./nptl/futex-internal.c:57
  10   Thread 0x7f3b697f96c0 (LWP 954697) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  11   Thread 0x7f3b69ffa6c0 (LWP 954696) "ros2-ust" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
...
(gdb) thread 1
[Switching to thread 1 (Thread 0x7f3b6c04ab80 (LWP 954695))]
#0  0x00007f3b6c0e4d61 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x2f4cd70) at ./nptl/futex-internal.c:57
57  in ./nptl/futex-internal.c
(gdb) bt
#0  0x00007f3b6c0e4d61 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x2f4cd70) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=0, abstime=0x0, clockid=0, expected=0, futex_word=0x2f4cd70) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x2f4cd70, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0) at ./nptl/futex-internal.c:139
#3  0x00007f3b6c0e77dd in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x2f4cd78, cond=0x2f4cd48) at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=0x2f4cd48, mutex=0x2f4cd78) at ./nptl/pthread_cond_wait.c:627
#5  0x00007f3b68fb2276 in rmw_wait (subscriptions=0x31f7308, guard_conditions=0x31f7320, services=0x31f7350, clients=0x31f7338, events=0x31f7368, wait_set=0x32ba850, wait_timeout=0x0) at <***removed***>/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:3277
#6  0x00007f3b6ae7ab1b in rcl_wait (wait_set=0x32c26f0, timeout=-1) at <***removed***>/ws_rmw_zenoh/src/rcl/rcl/src/rcl/wait.c:670
#7  0x00007f3b6b10299d in rclpy::WaitSet::wait (this=0x31ed840, timeout=-1) at <***removed***>/ws_rmw_zenoh/src/rclpy/rclpy/src/rclpy/wait_set.cpp:247
#8  0x00007f3b6b10708d in pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}::operator()(rclpy::WaitSet*, long) const (__closure=0x2a048f8, c=0x31ed840, args#0=-1)
    at /usr/include/pybind11/pybind11.h:111
#9  0x00007f3b6b10c87b in pybind11::detail::argument_loader<rclpy::WaitSet*, long>::call_impl<void, pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&, 0ul, 1ul, pybind11::detail::void_type>(pybind11::cpp_function::cpp_function<void, rclpy::WaitSet, long, pybind11::name, pybind11::is_method, pybind11::sibling, char [48]>(void (rclpy::WaitSet::*)(long), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [48])::{lambda(rclpy::WaitSet*, long)#1}&, std::integer_sequence<unsigned long, 0ul, 1ul>, pybind11::detail::void_type&&) && (
    this=0x7fff5aa59cd0, f=...) at /usr/include/pybind11/cast.h:1480
[... cut ...]
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/jazzy/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.1', 'console_scripts', 'ros2')())
```

The node is waiting on frame 5:
https://github.com/ros2/rmw_zenoh/blob/45391cc378c9b27f00eeac8a32df743f386a81b9/rmw_zenoh_cpp/src/rmw_zenoh.cpp#L3277

I have not verified whether the appropriate events are being correct received in the first place.

rmw_zenoh: 45391cc378c9b27f00eeac8a32df743f386a81b9
rclpy: e181737f19d6b8cb335b0326a40ec2075267c555

Yadunund, 2024-06-21:
```
#8  0x00007f90e1c28c0b in rclcpp_logging_output_handler (location=0x7f90df793f10 <rmw_zenoh_cpp::GraphCache::parse_del(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::__rcutils_logging_location>, severity=30, name=0x7f90df7596ed "rmw_zenoh_cpp", timestamp=1718793033330027250, 
```

@clalancette and I took a closer look at the stack trace from https://github.com/ros2/rmw_zenoh/issues/182#issuecomment-2178911481 and discovered that it is indeed an ABBA deadlock resulting from the two threads trying to acquire both `graph_mutex_` and the `scoped_logging_guard`.  🤦🏼 

The problem arises from the `RCUTILS_LOG_*_NAMED` macros. By default [vrcutils_log_internal](https://github.com/ros2/rcutils/blob/f69abc1f9d5871843df2ad3870faed43cf35df6e/src/logging.c#L1200) logs to stdout/err via `g_rcutils_logging_output_handler` but this hander is overwritten via [rcutils_logging_set_output_handler](https://github.com/ros2/rcutils/blob/f69abc1f9d5871843df2ad3870faed43cf35df6e/src/logging.c#L858) in [rclcpp](https://github.com/ros2/rclcpp/blob/eeaa5222a1d11d8e83d4e1b9092b11ee7bc5f39d/rclcpp/src/rclcpp/context.cpp#L220-L223) and [rclpy](https://github.com/ros2/rclpy/blob/ee221485aef4a1e6551ede22be2990019694260d/rclpy/src/rclpy/logging.cpp#L68)  to also include publishing to `/rosout`.

I'm working on a fix where we rely on our own logging functions in `rmw_zenoh` that prints to stderr/stdout. 

The deadlock in `rmw_wait` will be addressed separately.

hwoithe, 2024-06-21:
> I'm working on a fix where we rely on our own logging functions in rmw_zenoh that prints to stderr/stdout.

Would you suggest that all RMW implementations migrate to their own logging mechanism to avoid potential logger related deadlocks?

rmw_fastrtps has encountered this as well:
https://github.com/ros2/rmw_fastrtps/blob/1417a44b5022264c00c5cf748eea00e3dac698e9/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L216

I can see the value of a RMW safe replacement (or fix) to the ```RCUTILS_LOG_*_NAMED``` macros.

hwoithe, 2024-06-21:
| ...we could add a new set of logging APIs to rcutils specifically for the RMW layer, that would only ever use the console.

I was thinking this as well and in the long term would be nice to have.

Thanks to all for the help on this issue!

clalancette, 2024-08-21:
I believe that the basic issue here was fixed by #223.  With that, I'm going to close this out.  If you are still having problems, please feel free to reopen or open a new issue.

closed 2024-08-21 by clalancette