ros2/rmw_zenoh #183: Missing install 'ros-jazzy-desktop'

niektuytel, 2024-05-28:
[missing the ros-jazzy-desktop package](https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debians.html)
[image]

The error message
[image]

niektuytel, 2024-05-28:
i have the same issue on `E: Unable to locate package ros-humble-desktop`

mjcarroll, 2024-05-28:
The zenoh repo is very likely the wrong place to ask this question as it doesn't have anything to specifically do with rmw_zenoh.

There are no active ROS releases support 20.04 as a Tier 1 platform.  You can probably compile `humble` from source, but binary `humble` packages require 22.04 and binary `jazzy` packages require 24.04.

closed 2024-05-28 by mjcarroll