ros2/rmw_zenoh #184: ros2 node list and ros2 topic list not working

berndpfrommer, 2024-05-29:
After your very nice youtube presentation today I took a stab at it, compiled and overlayed the rmw_zenoh_cpp workspace.
Talker and listener are working fine, but when I open a third terminal and try to run
``ros2 node list``
no nodes are shown and ``ros2 topic list`` shows only
```
/parameter_events
/rosout
```
Are those commands supposed to work?

Yadunund, 2024-05-29:
Hi @berndpfrommer,

Thanks for trying out `rmw_zenoh`.

That sounds really strange. 

This is silly of me to ask but did you `export RMW_IMPLEMENTATION=rmw_zenoh_cpp` in the third terminal? It will also help to ensure any `ros2 daemons` started previously are properly terminated before switching to `rmw_zenoh` since the cli tools query the `daemon` for ROS graph information.  Running the below command before running any nodes with `rmw_zenoh` would work.

```bash
pkill -9 -f ros && ros2 daemon stop
```

berndpfrommer, 2024-05-29:
RMW_IMPLEMENTATION was set correctly. Killing the ros2 daemon did the trick!
Thanks!

closed 2024-05-29 by berndpfrommer

Yadunund, 2024-05-29:
Glad to hear!