ros2/rmw_zenoh #187: MoveIt + zenoh : Topic name /xxx not found in topic map. Report this

CihatAltiparmak, 2024-05-29:
## Tested On
* OS : Ubuntu 22.04
* ROS Version: iron

## Description

Hello, 
i am trying to integrate rmw_zenoh to MoveIt. When i ran the demo launch of moveit, I observed some error logs related to rmw_zenoh and demo launch didn't run as expected. You can find this log [here](https://gist.github.com/CihatAltiparmak/dbbf4767a50061799c0aa92b0ce2cc9d#file-rmw_zenoh_spawner_crash-txt). Unfortunately, now i can only give this logs. But if i can achieve more information related to this problem, i will share this details here as well.

In addition, i also observed that MoveIt + zenoh runs without problem. but this occurs so rarely. 

## How To Reproduce

Apply the [get started tutorial of MoveIt](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html). Change RMW_IMPLEMENTATION with rmw_zenoh and then apply [MoveIt QuickStart in Rviz](https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) pages.

Yadunund, 2024-05-29:
Thanks for the ticket.

Related:
https://github.com/ros2/rmw_zenoh/issues/170

This is known issue where there is a `rmw_shutdown` and `rmw_context_fini` may be invoked in incorrect sequence somewhere in rclcpp/rclpy (pending further investigation). In your case you run a `spawner` node that shuts down after spawning the controller and we may be experiencing the above during this shutdown.

The issue may also be related to Lifecycle nodes and there was recently a fix upstream https://github.com/ros2/rclcpp/pull/2527.
Could you retry with the backported changes to `iron`? You will need to build `iron` from source.

CihatAltiparmak, 2024-05-29:
Thank you for your advice. Unfortunately i should say that i observed this bugs after applying your steps. Whereas this bug now rarely occurs compared to previous one. In addition, the rate of running moveit with rmw_ zenoh without problem has increased dramatically.  This is great for me. My other observations are that moveit sometimes runs without problem, sometimes its planning libraries are not loaded, sometimes robot planner runs but its planned trajectory is not executed.

Maybe i did something wrong while setting necessary workspace. I will continue to my work.

If you want, you can close this issue now. Because i can see something is worked. But it looks like there is a evil somewhere in controller spawner.

closed 2024-06-12 by CihatAltiparmak