ros2/rmw_zenoh #188: zenohd should log router ID to console

berndpfrommer, 2024-05-30:
Is there  any reason why not to have the router log its id to the console? It would make it easier to verify that the clients indeed connect to the right router. Right now, the zenohd only produces this output:
```
ros2 run rmw_zenoh_cpp rmw_zenohd
Enter 'q' to quit...
```
Whereas the client is more talkative:
```
 Successfully connected to a Zenoh router with id a9d4ace930f270fb60876a363d75844c.
```
When I had trouble getting things to work (see issue #184), the first question was: what router is the client connecting to? I know the daemon's log verbosity can be increased with the right env variable, but if it had just logged its ID on startup, that would have saved me from finding out how to change the log verbosity.

Yadunund, 2024-05-30:
Hi @berndpfrommer,

You can run the bridge with  envar `export RUST_LOG=zenoh=debug,zenoh::net::routing::queries=trace` and it should print out all the information you need. Let me know if that helps.

berndpfrommer, 2024-05-30:
Thanks, I got it from the logs already. But you can probably tell that the above line is not super intuitive, and if things don't work, people have to first find out how to enable the logging. If you want to keep the default output minimal, a "--debug" flag as argument to zenohd would also suffice.
At the moment I have no problems with rmw_zenoh_cpp. Works great for me. Just wanted to make a suggestion to improve the out-of-the-box experience of rmw_zenoh_cpp.

Yadunund, 2024-05-30:
Thanks for the valuable feedback! I'm okay with having the router print a log each time a session connects to it! We can have more details logged when `--debug` is passed.
I'd be more than happy to review a PR with these changes.

closed 2024-06-12 by Yadunund