ros2/rmw_zenoh #199: Is shm used in nodes connected in the same SOC

sgf201, 2024-06-08:
As I see,zenoh can use shm for msg ,Is shm used in rmw_zenoh?

Yadunund, 2024-06-08:
Preliminary efforts were made to support SHM but certain memory leaks were discovered. Hence, SHM is disabled by default in [the session config](https://github.com/ros2/rmw_zenoh/blob/1caf7f1c55e7f5f04c3511b19d172571e732f42e/rmw_zenoh_cpp/config/DEFAULT_RMW_ZENOH_SESSION_CONFIG.json5#L234-L235).

sgf201, 2024-06-08:
as I see, loan api of ros2 is UNSUPPORTED now, will it supported with shm in roadmap?

ruffsl, 2024-06-08:
> Hence, SHM is disabled by default

@Yadunund , is there some context for that. From git blame of that config line, is this perhaps all of it?
- https://github.com/ros2/rmw_zenoh/pull/91#issuecomment-1907298304
- https://github.com/ros2/rmw_zenoh/commit/a9b424505c55411dd37db402eae91663673980e8

Yadunund, 2024-06-10:
> > Hence, SHM is disabled by default
> 
> @Yadunund , is there some context for that. From git blame of that config line, is this perhaps all of it?
> 
> * [Support basic QoS settings #91 (comment)](https://github.com/ros2/rmw_zenoh/pull/91#issuecomment-1907298304)
> * [a9b4245](https://github.com/ros2/rmw_zenoh/commit/a9b424505c55411dd37db402eae91663673980e8)

It's an issue in the Zenoh implementation. See https://github.com/eclipse-zenoh/zenoh/issues/406.

My understanding is that Zettascale is working on a true-zero-copy alternative so we're likely going to wait for that to land and update our SHM implementation accordingly.

JEnoch, 2024-06-12:
That's right, a new shared memory will be available very soon with Zenoh 1.0.0.
The development in Rust is already completed, and the C API has been merged to the `dev/1.0.0` branch last week:
https://github.com/eclipse-zenoh/zenoh-c/pull/405

kikuyuta, 2024-06-12:
That's a great news!
According to my understanding, currently we can use zenoh-pico for poor capability chips.
Will zenoh 1.0.0, the Rust version with shared memory functionality, work on poor capacity chips? Or will zenoh-pico also have SHM functionality?

JEnoch, 2024-06-12:
There is no plan for a Zenoh Rust for constrained devices, neither for SHM in zenoh-pico.
But the Zenoh team is always eager to learn about new use cases and to get suggestions of new features. 😃 
Please, feel free to open a discussion here:
https://github.com/eclipse-zenoh/roadmap/discussions/categories/ideas-and-new-features

kikuyuta, 2024-06-12:
@JEnoch Thank you for your reply.
Okay I will open a new discussion there, but later, because it is past midnight in Japan.

sgf201, 2024-06-13:
> as I see, loan api of ros2 is UNSUPPORTED now, will it supported with shm in roadmap?

can anyone give me a bref info about this?
@JEnoch

closed 2024-06-24 by sgf201