ros2/rmw_zenoh #220: Subscriber terminating crashes publisher with SHM

ciandonovan, 2024-06-27:
When using shared-memory between a publisher and subscriber, when the subscriber terminates either gracefully or not, the publisher crashes with:

```
[node-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[node-1]   what():  failed to publish message: Failed to allocate a SHM buffer, even after GCing, at /opt/ros/dev_ws/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:949, at ./src/rcl/publisher.c:284
```

Tested with commit https://github.com/ros2/rmw_zenoh/commit/e73df29cf80723f0e49e09d1f72cff5f1d342744 of `rmw_zenoh` with the Jazzy distro.

clalancette, 2024-06-27:
Can you please give a configuration/setup where we could enable shared memory?  At present I'm not sure how to do this.

ciandonovan, 2024-06-27:
DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5
```
/// This file attempts to list and document available configuration elements.
/// For a more complete view of the configuration's structure, check out `zenoh/src/config.rs`'s `Config` structure.
/// Note that the values here are correctly typed, but may not be sensible, so copying this file to change only the parts that matter to you is not good practice.
{
  /// The identifier (as unsigned 128bit integer in hexadecimal lowercase - leading zeros are not accepted)
  /// that zenoh runtime will use.
  /// If not set, a random unsigned 128bit integer will be used.
  /// WARNING: this id must be unique in your zenoh network.
  // id: "1234567890abcdef",

  /// The node's mode (router, peer or client)
  mode: "router",

  /// Which endpoints to connect to. E.g. tcp/localhost:7447.
  /// By configuring the endpoints, it is possible to tell zenoh which router/peer to connect to at startup.
  connect: {
    endpoints: [
      // "<proto>/<address>"
    ],
  },

  /// Which endpoints to listen on. E.g. tcp/localhost:7447.
  /// By configuring the endpoints, it is possible to tell zenoh which are the endpoints that other routers,
  /// peers, or client can use to establish a zenoh session.
  listen: {
    endpoints: [
      "tcp/[::]:7447"
    ],
  },
  /// Configure the scouting mechanisms and their behaviours
  scouting: {
    /// In client mode, the period dedicated to scouting for a router before failing
    timeout: 3000,
    /// In peer mode, the period dedicated to scouting remote peers before attempting other operations
    delay: 1,
    /// The multicast scouting configuration.
    multicast: {
      /// Whether multicast scouting is enabled or not
      enabled: false,
      /// The socket which should be used for multicast scouting
      address: "224.0.0.224:7446",
      /// The network interface which should be used for multicast scouting
      interface: "auto", // If not set or set to "auto" the interface if picked automatically
      /// Which type of Zenoh instances to automatically establish sessions with upon discovery on UDP multicast.
      /// Accepts a single value or different values for router, peer and client.
      /// Each value is bit-or-like combinations of "peer", "router" and "client".
      autoconnect: { router: "", peer: "router|peer" },
      /// Whether or not to listen for scout messages on UDP multicast and reply to them.
      listen: true,
    },
    /// The gossip scouting configuration.
    gossip: {
      /// Whether gossip scouting is enabled or not
      enabled: true,
      /// When true, gossip scouting informations are propagated multiple hops to all nodes in the local network.
      /// When false, gossip scouting informations are only propagated to the next hop.
      /// Activating multihop gossip implies more scouting traffic and a lower scalability.
      /// It mostly makes sense when using "linkstate" routing mode where all nodes in the subsystem don't have
      /// direct connectivity with each other.
      multihop: false,
      /// Which type of Zenoh instances to automatically establish sessions with upon discovery on gossip.
      /// Accepts a single value or different values for router, peer and client.
      /// Each value is bit-or-like combinations of "peer", "router" and "client".
      autoconnect: { router: "", peer: "router|peer" },
    },
  },

  /// Configuration of data messages timestamps management.
[... 449 lines cut ...]
        private_key_pem: null,
        public_key_file: null,
        private_key_file: null,
        key_size: null,
        known_keys_file: null,
      },
    },
  },

  /// Configure the Admin Space
  /// Unstable: this configuration part works as advertised, but may change in a future release
  adminspace: {
    // read and/or write permissions on the admin space
    permissions: {
      read: true,
      write: false,
    },
  },
}

```

`export RMW_IMPLEMENTATION=rmw_zenoh_cpp`
`export ZENOH_ROUTER_CONFIG_URI=DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5`
`export ZENOH_SESSION_CONFIG_URI=DEFAULT_RMW_ZENOH_SESSION_CONFIG.json5`

ciandonovan, 2024-06-27:
I'm working on a reproducible container for this issue, but unfortunately it doesn't seem to be an issue with the simple pub/sub demo from the docs.

The issue occurred when streaming images so much larger bandwidth, working on a reproducible example that does something similar

Yadunund, 2024-06-27:
> I'm working on a reproducible container for this issue, but unfortunately it doesn't seem to be an issue with the simple pub/sub demo from the docs.
> 
> The issue occurred when streaming images so much larger bandwidth, working on a reproducible example that does something similar

Ah my hunch is that your payload exceeds the 10MB we allocate to the SHM buffer.
Can you retry after manually bumping this value to a larger size + re-compiling?
https://github.com/ros2/rmw_zenoh/blob/cdea502d5537f8ff22ae854568f59ce3bb09543b/rmw_zenoh_cpp/src/rmw_init.cpp#L41-L43

We tried making this configurable in the zenoh configs but at the time custom key-value pairs were not supported.

ciandonovan, 2024-06-29:
@Yadunund yes I can confirm bumping the `SHM_BUFFER_SIZE_MB` resolves the issue

ciandonovan, 2024-06-30:
However, it still occurs when a larger number of connections are made/unmade in rapid succession.

https://github.com/ros2/rmw_zenoh/assets/94260580/268e685c-d525-4a18-bf5e-9265519a2d39

Yadunund, 2024-07-01:
Thanks for the update. From the video attached, it seems like some memory is not being feed. We don't have any plans to get the existing SHM implementation working as of now as we will likely need to re-implement things once zenoh-1.0 is released. In the meanwhile, we'd be happy to review any of your PRs that fix this behavior.