ros2/rmw_zenoh #224: Shared-memory buffer memory-leak in /dev/shm/

ciandonovan, 2024-06-30:
When zenohd crashes or otherwise doesn't gracefully shutdown, the associated `shmem_XXXXXX` file in `/dev/shm/` isn't cleaned up. Nor is it cleaned up the next time a zenohd instance launches.

This leads to persistent useless memory usage that won't be resolved until the host reboots or the files are manually deleted. If there is an issue with zenohd that causes it to crash often and it is automatically restarted by a service manager, this could lead to memory exhaustion which Linux does not deal with well, causing system lockups and freezes.

ciandonovan, 2025-08-07:
Would a more robust standard modern way not be to use a memfd backend as an alternative to POSIX shm? It has inbuilt reference-counted cleanup based on open fds managed by the kernel, no manual managing required. Segments are inherently private too and only shared with other processes if passed by fd via a UNIX socket, better isolation than the globally shared /dev/shm.

Since rmw_zenoh already uses a UNIX socket to coordinate between processes it would seem a natural fit to share fds to memfd segments through that also?

YuanYuYuan, 2025-08-12:
Hi @ziyangfu & @ciandonovan! Thanks for reporting this. Our current mechanism always cleans up dangling SHM segments through garbage collection, either explicitly via an API call or automatically when a new SHM allocation occurs.

Regarding the suggestion to use a memfd backend, we’ll look into it. Thanks!