ros2/rmw_zenoh #236: ctrl+c not killing ros2 cli tools

jmount1992, 2024-07-03:
As a user I would expect that pressing `ctrl+c` when running a ros2 cli tool such as `ros2 topic list` or `ros2 node list` would kill the process. However, if a Zenoh router isn't running this signal isn't handled and it continues to print the default behaviour message of `[ERROR] [1719974909.135681607] [rmw_zenoh_cpp]: Unable to connect to a Zenoh router. Have you started a router with `ros2 run rmw_zenoh_cpp rmw_zenohd`?` I understand the default behaviour can be changed. However, I would expect users would still expect ctrl+c to kill this process.

Yadunund, 2024-07-03:
This is a known issue with signal handlers in rclpy + issues in rmw_shutdown. See https://github.com/ros2/rmw_zenoh/pull/135#pullrequestreview-1955014517

We don't have a solution yet but hoping this will be addressed when we resolve #170.

jmount1992, 2024-07-17:
Perfect. I hadn't seen any issue mentioning it so I wanted to bring it to your attention given the rmw_zenoh is in its infancy.

Yadunund, 2024-07-17:
Yep it's good to have this ticket! We are working on addressing it.

Yadunund, 2024-09-03:
Still an issue after https://github.com/ros2/rclpy/pull/1331

clalancette, 2024-12-17:
I believe that this is now fixed, so closing this out.  If you are still having this issue, please feel free to reopen.

closed 2024-12-17 by clalancette