ros2/rmw_zenoh #241: Support environment variables for dynamic discovery

Yadunund, 2024-07-11:
As documented here https://docs.ros.org/en/rolling/Tutorials/Advanced/Improved-Dynamic-Discovery.html

gavanderhoorn, 2025-02-13:
@Yadunund: somewhat related (hence my comment here): would it perhaps make sense to support an environment variable just for configuring / adding to [connect.endpoints](https://github.com/ros2/rmw_zenoh/blob/1601a69622cd913d0f34e64f1840f3b159ca5e75/rmw_zenoh_cpp/config/DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5#L36-L38) in the router config?

Right now, if a user wants to connect their router to an off-host router (ie: join/create a mesh), they have to copy the router config file, edit it and make `rmw_zenoh` use that copy, instead of the default one.

But having to duplicate the file (and manage it) just in order to be able to setup a multi-machine ROS 2 application -- which requires just a single line addition -- seems like it's more complex than it absolutely needs to be.

Support for something like `ROS_STATIC_PEERS` could help here perhaps: users would just need to set that environment variable. Especially when IPs/peers change that would be much easier than having to copy & edit the file.

Yadunund, 2025-02-14:
@gavanderhoorn,

What you suggested will be doable once we address https://github.com/ros2/rmw_zenoh/issues/262. The upstream Zenoh router, ie, `zenohd`, already parses various CLI args to inject/overwrite the config that may be loaded from the file. In `rmw_zenoh`, we're not running the upstream `zenohd` and instead have a custom executable [rmw_zenohd](https://github.com/ros2/rmw_zenoh/blob/rolling/rmw_zenoh_cpp/src/zenohd/main.cpp) which starts a router process with limited functionalities. Long story short, the c(pp) bindings of Zenoh did not distribute the zenohd binary but Zettascale has recently added some mechanisms to install the router when building the bindings. Once those changes go in, we'll bump the version of zenoh-cpp we vendor in rmw_zenoh and update our `rmw_zenohd` to simply run the upstream`zenohd` while passing along all the CLI args that the Zenoh router already parses including ones that specify additional endpoints to connect to 😄