ros2/rmw_zenoh #242: rmw_zenoh and ROS 2 daemon

Yadunund, 2024-07-12:
With ROS 2, there is a long-running daemon in the background that stores information about the ROS graph to provide faster responses to queries, e.g. the list of node names, without having to discovery every other peer in the network. The daemon is automatically started when the relevant command-line tools are used for the first time.

With `rmw_zenoh`, we expect users to run the `zenohd` router which largely does the same above. However, there are issues with the `zenohd` router and daemon coexisting (mostly related to graph cache queries).

For example, if a user runs `ros2 topic list` without the `zenohd` router running, the process will check indefinitely for a router until one is started. Once the command exists, and if one kills the router, running `ros2 topic list` again will instantly return the topics from the daemon's cache. However in such scenarios we would like the cli tool to look for a router first before returning results from a cache.

Starting a discussion on how to manage such scenarios.
1. systemd service: An extention to the discussion in https://github.com/ros2/rmw_zenoh/issues/231, the service could also kill/restart the daemon as appropriate.
2. Disable daemon:
  * Add mechanisms in `rmw` API to selectively enable/disable certain "features" including the daemon.
  * Update `rcl`, `rclpy` and `ros2cli` such that these features are checked  and only start the daemon if configured to do so.

clalancette, 2024-07-25:
We had a bit of a discussion about this at the Waffle meeting today.  Some notes:

* People were somewhat opposed to getting rid of the daemon, because it makes rmw_zenoh look "different" than other RMWs.
* One suggestion was to couple the lifetimes of the ROS 2 daemon and the Zenoh router.  That would probably mean that when we launched the Zenoh router, we would *also* launch the ROS 2 daemon at the same time, and if one died they both die.
* Another suggestion was to just replace the ROS 2 daemon with the Zenoh router.  There are a couple of major problems with that approach:
  * The protocol between the ROS 2 daemon the ros2cli tools is XMLRPC.  So we'd have to have some kind of "wrapper" around the Zenoh router that converted XMLRPC calls into something that the Zenoh router understands.
  * The ROS 2 daemon doesn't always run.  It is only started when certain ros2cli commands are invoked.  In contrast, the Zenoh router is required and needs to be running at all times for rmw_zenoh to succeed, so it has different lifetime requirements.

Pleune, 2024-11-01:
I would love to see zenohd automatically stop the ros2 daemon if it had been started with another rmw. These issues can be hard to track down if you read past the section in the readme.

peci1, 2026-11-13:
I think the problem with the current implementation is that there's only one cli daemon on one port. These issues where a user starts with FastDDS and later switches to Zenoh are real. Currently, the daemon is linked to FastDDS and Zenoh queries it without issues (if this hasn't been fixed already).

Either the daemon needs to become a multi-daemon (a hydra? :) ), or there needs to be a way how to compute its port from ROS distro, RMW and domain ID (and maybe security enclave).

So maybe a larger overhaul of the daemon idea is needed?

clalancette, 2026-11-13:
> Either the daemon needs to become a multi-daemon (a hydra? :) ), or there needs to be a way how to compute its port from ROS distro, RMW and domain ID (and maybe security enclave).

FWIW, this is exactly what the daemon is written to do (not the security enclave, but the rest).  If that is not working, then we probably need a bugfix somewhere in there.

peci1, 2026-11-13:
Are you sure?

https://github.com/ros2/ros2cli/issues/1009

https://github.com/ros2/ros2cli/pull/1270

clalancette, 2026-11-13:
> Are you sure?

Oh, interesting.  Looking again, it looks like it does launch one per domain ID: https://github.com/ros2/ros2cli/blob/254072b22ff58aa7aee20535c144b32a3c707697/ros2cli/ros2cli/daemon/__init__.py#L31-L34 .  And it also passes the rmw_implementation to the daemon, but it turns out it is only informational: https://github.com/ros2/ros2cli/blob/254072b22ff58aa7aee20535c144b32a3c707697/ros2cli/ros2cli/node/daemon.py#L189-L191 .

So I guess it has all of the information it needs, it just doesn't make new daemons.  I guess the big problem is discovering it; we wouldn't know which port to connect to, so we'd have to go scanning around.  Maybe there is a smarter way to do it.

peci1, 2026-11-13:
Choose a random port and write it down to `/run/ros2daemon/${ROS_DOMAIN_ID}_${RMW_IMPLEMENTATION}_${ENCLAVE_CA_SERIAL:-}`? (or to `/tmp` if `/run` is read-only). Would there be any downsides?

clalancette, 2026-11-13:
> Choose a random port and write it down to `/run/ros2daemon/${ROS_DOMAIN_ID}_${RMW_IMPLEMENTATION}_${ENCLAVE_CA_SERIAL:-}`? (or to `/tmp` if `/run` is read-only). Would there be any downsides?

I'm thinking something slightly more sophisticated (and cross-platform).  On startup, ros2 cli commands look to see if there is a `$ROS_HOME/ros2cli/daemons/<domain_id>-<rmw>.json` (by default, `$ROS_HOME` is `~/.ros`).  If that file does not exist, it creates a new ros2 daemon, which is responsible for creating that file.  If it does exist, it uses the data in that file to find where to connect.  The contents of that file would be something like: 

`{"port": 53122, "pid": 4711, "rmw_implementation": "rmw_zenoh_cpp", "ros_domain_id": 0, "token": "<random>"}`

The port is totally random (open the socket with 0 so the OS gives us a random one).  The token is to ensure we are talking to a not-stale daemon; all ros2 cli commands start by verifying that the token they read is the same as the live server they are talking to.  On startup, daemons overwrite the file atomically.  If a daemon ever finds that it is stale itself (this can happen if two ros2 cli commands race to start it), it assumes that the "other" daemon running is not stale and exits itself.

What do you think?

peci1, 2026-11-13:
Sounds good to me. Maybe we should move this discussion to https://github.com/ros2/ros2cli/issues/1009 ? There were more people with interest into looking into this.

clalancette, 2026-11-14:
> Sounds good to me. Maybe we should move this discussion to [ros2/ros2cli#1009](https://github.com/ros2/ros2cli/issues/1009) ? There were more people with interest into looking into this.

Good idea.  I'm not going to move this issue over; this is more specific to rmw_zenoh itself, rather than the more generic "each RMW should have its own daemon".  I'll either attach the conversation here to an existing issue, or open a new one over there.

Yadunund, 2026-11-16:
I like this approach discussed above! 

I also think having a custom router plugin like what @josephduchesne described would not be too difficult and could perhaps make the experience better on older distros.