ros2/rmw_zenoh #254: Add example router configuration file for 'peer' topologies

josborja7castillo, 2024-07-29:
Hi, I am attempting to use the Zenoh RMW as I was used to with DDS, that is, being a distributed service. So far, I have not been lucky with that, although the "router/client" topology works like a charm :). Please, could you provide an example of a JSON5 configuration file that enables ROS 2 nodes to communicate in the "peer to peer" (distributed/non-centralized) mode?

I am using Ubuntu 24.04 LTS with ROS 2 Jazzy.

Thanks in advance!

Timple, 2024-07-29:
The process is doable, one one of the hosts (e.g. your laptop), you need to copy this file: https://github.com/ros2/rmw_zenoh/blob/rolling/rmw_zenoh_cpp/config/DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5 for example to `~/my_router_config.json5`. Open it and modify the `endpoints` key according to these instructions: https://github.com/ros2/rmw_zenoh?tab=readme-ov-file#connecting-multiple-hosts.

After that, you point to the file and run the router:
```bash
export ZENOH_SESSION_CONFIG_URI=~/my_router_config.json5
ros2 run rmw_zenoh_cpp rmw_zenohd
```

Now in another terminal you should be able to see the nodes.

---

It would really be nice if the router had a nice argument such that one could do only this:

```
ros2 run rmw_zenoh_cpp rmw_zenohd --listen tcp/my-robot.local:7447  # This does not work now!
```
So perhaps that idea can be entertained rather than writing instructions :slightly_smiling_face:

josborja7castillo, 2024-07-29:
Thank you for your reply @Timple! However, I need to be a bit more specific. 
This approach does not suit very well in case the addresses are changing frequently, so I was considering employing the UDP multicast scouting. I think this will avoid me from needing to specify the endpoints and still it will use a de-centralized topology, is it true?

At my current setup, I can freely use Wireshark as ROS 2 is running on LXC containers.

Cheers!

Yadunund, 2024-07-30:
To have nodes discover each other over multicast discovery, you will need to 
1) enable multicast scouting [here](https://github.com/ros2/rmw_zenoh/blob/fcfed30ca7ba629a137bc48fab0502c71e515839/rmw_zenoh_cpp/config/DEFAULT_RMW_ZENOH_SESSION_CONFIG.json5#L42)
2) Set envar `ZENOH_ROUTER_CHECK_ATTEMPTS` to `-1` as described here https://github.com/ros2/rmw_zenoh/tree/rolling#checking-for-a-zenoh-router

This should work for nodes on the same host. I've not tried it for nodes across different hosts but I imagine it would work the same. You can find better documentation of various topologies in upstream Zenoh.

We don't plan to add such a config file to the repo here since the goal of `rmw_zenoh` is to provide a good out-of-the-box experience with a single default config file. However, advanced users are free to specify custom configs. Hence, I will close this ticket.

Feel free to ask more questions like this on https://robotics.stackexchange.com/

closed 2024-07-30 by Yadunund