ros2/rmw_zenoh #257: rmw_destroy_guard_condition segfault

CihatAltiparmak, 2024-08-05:
#### Versions
rmw_zenoh : fcfed30ca7ba629a137bc48fab0502c71e515839

#### Description
Hello, 
I'm trying to benchmark the middleware implementations like rmw_zenoh, rmw_cyclonedds and rmw_fastrtps. I have implemented some benchmark codes for comparing how middlewares behave basic at topic subscription-publishing scenario. My benchmark code exits with SIGSEV exit code whereas rmw_fastrtps exits cleanly. Here is backtrace. Seems problem occurs while trying to deallocate some data. 

Bactrace:

```
(gdb) backtrace 
#0  0x0000000000000000 in ?? ()
#1  0x00007ffff6363aac in rmw_destroy_guard_condition (
    guard_condition=0x555555943e90)
    at /home/cihat/ws_moveit/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:3282
#2  0x00007ffff7f253e0 in rcl_guard_condition_fini ()
   from /opt/ros/rolling/lib/librcl.so
#3  0x00007ffff7d28191 in rclcpp::GuardCondition::~GuardCondition() ()
   from /opt/ros/rolling/lib/librclcpp.so
#4  0x00007ffff7cf3bea in ?? () from /opt/ros/rolling/lib/librclcpp.so
#5  0x00007ffff7d02731 in rclcpp::Executor::~Executor() ()
   from /opt/ros/rolling/lib/librclcpp.so
#6  0x000055555578d473 in __gnu_cxx::new_allocator<rclcpp::executors::SingleThreadedExecutor>::destroy<rclcpp::executors::SingleThreadedExecutor> (
    this=0x555555947f20, __p=0x555555947f20)
    at /usr/include/c++/11/ext/new_allocator.h:168
#7  0x000055555578b3fd in std::allocator_traits<std::allocator<rclcpp::executors::SingleThreadedExecutor> >::destroy<rclcpp::executors::SingleThreadedExecutor>
    (__a=..., __p=0x555555947f20)
    at /usr/include/c++/11/bits/alloc_traits.h:535
#8  0x000055555578742f in std::_Sp_counted_ptr_inplace<rclcpp::executors::SingleThreadedExecutor, std::allocator<rclcpp::executors::SingleThreadedExecutor>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x555555947f10)
    at /usr/include/c++/11/bits/shared_ptr_base.h:528
--Type <RET> for more, q to quit, c to continue without paging--
#9  0x000055555572e53d in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x555555947f10) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#10 0x000055555572e2db in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x5555558346d8, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:705
#11 0x0000555555735c62 in std::__shared_ptr<rclcpp::executors::SingleThreadedExecutor, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x5555558346d0, 
    __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#12 0x0000555555735c82 in std::shared_ptr<rclcpp::executors::SingleThreadedExecutor>::~shared_ptr (this=0x5555558346d0, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr.h:122
#13 0x0000555555735ce8 in moveit::middleware_benchmark::ScenarioBasicSubPubFixture::~ScenarioBasicSubPubFixture (this=0x5555558345f0, 
    __in_chrg=<optimized out>)
    at /home/cihat/ws_moveit/src/moveit2_stack/moveit_middleware_benchmark/include/moveit_middleware_benchmark/scenarios/scenario_basic_subscription.hpp:70
#14 0x0000555555783544 in moveit::middleware_benchmark::ScenarioBasicSubPubFixture_test_scenario_basic_sub_pub_Benchmark::~ScenarioBasicSubPubFixture_test_scenario_basic_sub_pub_Benchmark (this=0x5555558345f0, __in_chrg=<optimized out>)
    at /home/cihat/ws_moveit/src/moveit2_stack/moveit_middleware_benchmark/src/scenarios/scenario_basic_subscription.cpp:107
#15 0x0000555555783564 in moveit::middleware_benchmark::ScenarioBasicSubPubFixtu--Type <RET> for more, q to quit, c to continue without paging--
re_test_scenario_basic_sub_pub_Benchmark::~ScenarioBasicSubPubFixture_test_scenario_basic_sub_pub_Benchmark (this=0x5555558345f0, __in_chrg=<optimized out>)
    at /home/cihat/ws_moveit/src/moveit2_stack/moveit_middleware_benchmark/src/scenarios/scenario_basic_subscription.cpp:107
#16 0x00007ffff7f78daf in ?? () from /opt/ros/rolling/lib/libbenchmark.so.1
#17 0x00007ffff7445495 in __run_exit_handlers (status=0, 
    listp=0x7ffff761a838 <__exit_funcs>, 
    run_list_atexit=run_list_atexit@entry=true, run_dtors=run_dtors@entry=true)
    at ./stdlib/exit.c:113
#18 0x00007ffff7445610 in __GI_exit (status=<optimized out>)
    at ./stdlib/exit.c:143
#19 0x00007ffff7429d97 in __libc_start_call_main (
    main=main@entry=0x55555572de89 <main(int, char**)>, argc=argc@entry=13, 
    argv=argv@entry=0x7fffffffaa68)
    at ../sysdeps/nptl/libc_start_call_main.h:74
#20 0x00007ffff7429e40 in __libc_start_main_impl (
    main=0x55555572de89 <main(int, char**)>, argc=13, argv=0x7fffffffaa68, 
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffaa58) at ../csu/libc-start.c:392
#21 0x000055555572ddc5 in _start ()
```
The crashed code line:

https://github.com/ros2/rmw_zenoh/blob/c12ff3ed2b73169ff0bd7a2a341f8911082fc1e9/rmw_zenoh_cpp/src/rmw_zenoh.cpp#L3282 

#### How to reproduce
I actually work on source installation, but maybe you want to try it. Ping me in any case you struggle, because i haven't run this docker for long time.

Pull this docker image
```sh
docker pull ghcr.io/cihataltiparmak/moveit_middleware_benchmark:latest
```
inside docker:

Terminal 1
```
cd ws_moveit/src/moveit_middleware_benchmark
git checkout feature/benchmark_simple_sub_pub_topics
cd ../../
colcon build --packages-select rmw_zenoh_cpp moveit_middleware_benchmark --cmake-args -DCMAKE_BUILD_TYPE=Debug
source /opt/ros/rolling/setup.bash
source install/setup.bash
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
# don't forget to run zenoh_router in Terminal 2
ros2 launch moveit_middleware_benchmark scenario_basic_subscription_benchmark.launch.py
```

Terminal 2
```
docker exec -it <your_container_id> bash
# inside docker
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
ros2 run rmw_zenoh_cpp rmw_zenohd
```

CihatAltiparmak, 2024-08-05:
After some invastigation, i found that `&guard_condition->context->options.allocator;` is NULL and rmw_zenoh tries to use `allocator->deallocate(guard_condition->data, allocator->state); ` in order to shutdown node.

CihatAltiparmak, 2024-08-08:
Hello @Yadunund ,
I feel like it's not necessary to add `RCUTILS_CHECK_ALLOCATOR` to `rmw_destroy_guard_condition` . Because i guess there are still some dataraces and this dataraces give rise to write to some forbidden sections of memory. This is one of the backtrace logs. I have a lot of backtrace log. Some of them points out `rmw_destroy_node` and some of them is related to sub_data_handler. I have seen your PR's. Let's be patient. After merging  your PRs about the thread safety, i will try rmw_zenoh on my code again and i will give feedback. I found an error in my code but this error isn't triggered at other middleware implementation.(rmw_fastrtps and rmw_cyclonedds). The part is [here](https://github.com/CihatAltiparmak/moveit_middleware_benchmark/blob/bc0d06ad9a6e37c855fdb58f3e2e5b0c32ee53f9/src/scenarios/scenario_basic_subscription.cpp#L96-L105). (It should be added `executor_.reset()`) . I will send other backtrace logs after a detailed look. Let's wait your PRs.

```sh
[INFO] [1723137814.597346998] [benchmark_main]: Subscribing to topic : /benchmarked_topic1 with hz 10000
[INFO] [1723137814.599325017] [benchmark_main]: Successfully subscribed to topic /benchmarked_topic1 with hz 10000! When received msg number is bigger than 100, benchmark will be finished!
malloc(): unsorted double linked list corrupted

Thread 10 "rx-0" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe77fb640 (LWP 57183)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737077294656) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) backtrace 
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737077294656)
    at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737077294656)
    at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737077294656, signo=signo@entry=6)
    at ./nptl/pthread_kill.c:89
#3  0x00007ffff7442476 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
#4  0x00007ffff74287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff7489676 in __libc_message (action=action@entry=do_abort, 
    fmt=fmt@entry=0x7ffff75dbb77 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff74a0cfc in malloc_printerr (
    str=str@entry=0x7ffff75dec48 "malloc(): unsorted double linked list corrupted") at ./malloc/malloc.c:5664
#7  0x00007ffff74a42dc in _int_malloc (av=av@entry=0x7fffc8000030, 
    bytes=bytes@entry=17) at ./malloc/malloc.c:4010
#8  0x00007ffff74a5139 in __GI___libc_malloc (bytes=17)
    at ./malloc/malloc.c:3329
#9  0x00007ffff78ae98c in operator new(unsigned long) ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x000055555572e410 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*> (this=0x7fffe77eedf0, 
    __beg=0x7ffff63701be "source_timestamp", __end=0x7ffff63701ce "")
--Type <RET> for more, q to quit, c to continue without paging--
    at /usr/include/c++/11/bits/basic_string.tcc:219
#11 0x00007ffff6307a99 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<std::allocator<char> > (
    this=0x7fffe77eedf0, __s=0x7ffff63701be "source_timestamp", __a=...)
    at /usr/include/c++/11/bits/basic_string.h:539
#12 0x00007ffff6339735 in rmw_zenoh_cpp::sub_data_handler (
    sample=0x7fffe77ef190, data=0x55555592ab30)
    at /home/cihat/ws_moveit/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/rmw_data_types.cpp:474
#13 0x00007ffff488f700 in zenohcd::closures::sample_closure::z_closure_sample_call (closure=0x55555592ba40, sample=0x7fffe77ef190)
    at /home/cihat/ws_moveit/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/closures/sample_closure.rs:52
#14 0x00007ffff48bb212 in zenohcd::subscriber::z_declare_subscriber::{closure#0} (sample=...)
    at /home/cihat/ws_moveit/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/subscriber.rs:189
#15 0x00007ffff4bba88f in zenoh::session::Session::handle_data (
    self=0x555555843f00, local=false, key_expr=0x7fffe77f2f38, info=..., 
    payload=..., attachment=...) at src/session.rs:1712
#16 0x00007ffff4bc679c in zenoh::session::{impl#13}::send_push (
    self=0x555555843f00, msg=...) at src/session.rs:2240
#17 0x00007ffff4bcf80b in zenoh::session::{impl#16}::send_push (
--Type <RET> for more, q to quit, c to continue without paging--
    self=0x555555843f00, 
    msg=<error reading variable: Cannot access memory at address 0xe047>)
    at src/session.rs:2706
#18 0x00007ffff4a95d36 in zenoh::net::routing::dispatcher::pubsub::full_reentrant_route_data (tables_ref=0x7fffec13e920, face=0x7fffec0111a0, 
    expr=0x7fffe77f4e08, ext_qos=..., ext_tstamp=..., payload=..., 
    routing_context=0) at src/net/routing/dispatcher/pubsub.rs:494
#19 0x00007ffff4a16048 in zenoh::net::routing::dispatcher::face::{impl#4}::send_push (self=0x7fffec13e920, msg=...) at src/net/routing/dispatcher/face.rs:269
#20 0x00007ffff4c64bd0 in zenoh::net::primitives::demux::{impl#1}::handle_message (self=0x7fffec13e920, msg=...) at src/net/primitives/demux.rs:68
#21 0x00007ffff4b2b48e in zenoh::net::runtime::{impl#6}::handle_message (
    self=0x7fffec13a5a0, msg=...) at src/net/runtime/mod.rs:385
#22 0x00007ffff4de85a9 in zenoh_transport::unicast::universal::transport::TransportUnicastUniversal::trigger_callback (self=0x7fffec13f040, callback=..., 
    msg=...) at src/unicast/universal/rx.rs:49
[... cut ...]
#87 alloc::boxed::{impl#47}::call_once<(), alloc::boxed::Box<dyn core::ops::function::FnOnce<(), Output=()>, alloc::alloc::Global>, alloc::alloc::Global> (
    self=0x7fffec0117e0, args=<optimized out>)
    at library/alloc/src/boxed.rs:2007
#88 std::sys::unix::thread::{impl#2}::new::thread_start (main=0x7fffec0117e0)
    at library/std/src/sys/unix/thread.rs:108
#89 0x00007ffff7494ac3 in start_thread (arg=<optimized out>)
    at ./nptl/pthread_create.c:442
#90 0x00007ffff7526850 in clone3 ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

    at src/runtime/context/scoped.rs:40

clalancette, 2024-12-17:
All of the PRs to address this should have landed now, so I'm going to close this out.  Please feel free to retest and reopen if you are still running into trouble.

closed 2024-12-17 by clalancette