ros2/rmw_zenoh #267: Make rmw_zenoh_cpp pass all core tests

ahcorde, 2024-08-23:
This issue will track the tests failures in some ros core packages:

| Package | Failing test | Flaky |
|----------|----------| ----------|
| rcl   | test_events__rmw_zenoh_cpp  |  |
| rcl_action    | :heavy_check_mark:    | |
| rcl_lifecycle   |  :heavy_check_mark:   | |
| rclcpp   | test_qos_event__rmw_zenoh_cpp  | |
| rclcpp_lifecycle   |  :heavy_check_mark:    | |
| rclcpp_action  |  :heavy_check_mark:   | |
| test_rclcpp   |    |  test_services_cpp__rmw_zenoh_cpp |

Yadunund, 2024-09-20:
With https://github.com/ros2/rclcpp/pull/2626, I can get all rclcpp and rclcpp_lifecycle tests to run without any crashes!

However these tests from `rclcpp` still fail. Some should be fixed by https://github.com/ros2/rclcpp/pull/2626 while others require some fixes to `rmw_zenoh_cpp`.
```
The following tests FAILED:
	 33 - test_intra_process_manager (Failed)
	 41 - test_node_interfaces__get_node_interfaces (Failed)
	 44 - test_node_interfaces__node_graph (Failed)
	 62 - test_publisher (Failed)
	 67 - test_qos (Failed)
	 95 - test_executors (Failed)
	105 - test_events_executor (Timeout)
	108 - test_wait_set (Failed)
	132 - test_qos_event__rmw_zenoh_cpp (Failed)
```

Yadunund, 2024-09-26:
Superseded by https://github.com/ros2/rmw_zenoh/issues/286

closed 2024-09-26 by Yadunund

MrBlenny, 2025-03-11:
@Yadunund 

Have all of these been resolved? Using the latest binaries for `rmw_zenoh_cpp` I am seeing this issue when using zenoh with foxglove bridge:

```
[foxglove_bridge-2] [ERROR] [1741672280.071278525] [rclcpp]: caught std::exception exception in GraphListener thread: rcl_wait unexpectedly timed out
[foxglove_bridge-2] terminate called after throwing an instance of 'std::runtime_error'
[foxglove_bridge-2]   what():  rcl_wait unexpectedly timed out
[ERROR] [foxglove_bridge-2]: process has died [pid 77, exit code -6, cmd '/opt/ros/jazzy/lib/foxglove_bridge/foxglove_bridge --ros-args --log-level error --ros-args -r __node:=data_bridge -r __ns:=/groundstation --params-file /tmp/launch_params_veefrsbb'].
```

JEnoch, 2025-03-11:
Hi @MrBlenny ,
I created https://github.com/ros2/rmw_zenoh/issues/527 for this specific "rcl_wait unexpectedly timed out" issue.