ros2/rmw_zenoh #268: Towards Windows support

ahcorde, 2024-08-23:
I installed `rust` and `mingw` using `chocolatey`

```bash
choco install -y rust mingw
```

But I'm getting this error:

```
    CMake Error at install/cmake_install.cmake:59 (file):
      file INSTALL cannot find
      "C:/rolling/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor-build/release/target/release/zenohc.dll.lib":
      No error.
    Call Stack (most recent call first):
      cmake_install.cmake:37 (include)
```

Not sure why `.lib` is added at the end of the file name

Yadunund, 2024-08-28:
Perhaps you can try with this commit of zenoh-c?
https://github.com/eclipse-zenoh/zenoh-c/pull/471

ahcorde, 2024-08-28:
https://github.com/ros2/rmw_zenoh/pull/273

ahcorde, 2024-12-17:
the zenoh_cpp port is compiling on windows https://github.com/ros2/rmw_zenoh/pull/327

closing this issue

closed 2024-12-17 by ahcorde