ros2/rmw_zenoh #270: Value too large for defined data type - Compiling for ARM-HF using Docker Emulation

jmount-boss, 2024-08-27:
Attempting to compile RMW Zenoh for ARM-HF using Docker Emulation. After creating a ARM-HF (Linux/arm/v7) docker image and running an interactive container emulating the ARM architecture. I get the following error when attempting to build the zenoh_c_vendor package.

### Dockerfile

```
# syntax=docker/dockerfile:1

# ARGUMENTS
ARG DEBIAN_DIST=bookworm
ARG ROS2_DISTRO=jazzy

# ROS ARM-HF Image
FROM --platform=linux/armhf debian:${DEBIAN_DIST} AS ros-armhf
ARG DEBIAN_DIST
ARG ROS2_DISTRO

# Update and upgrade
RUN apt-get update && apt-get upgrade -y

# Add system dependencies folder from host
# allow installing of dependencies as well as
# keeps the list of deps installed within the image
ADD sys_deps /sys_deps

# Install ROS Build Dependencies
RUN /sys_deps/ros_build_${ROS2_DISTRO}.deps

# Debian distribution specifc setup
RUN apt-get remove -y rustc && apt-get install -y curl && /sys_deps/rustup_init.sh -y
```

### Colcon Build Error

```
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> zenoh_c_vendor
--- stderr: zenoh_c_vendor                                
    Updating crates.io index
    Updating git repository `https://github.com/eclipse-zenoh/zenoh.git`
warning: spurious network error (3 tries remaining): could not read directory '/root/.cargo/git/db/zenoh-cc237f2570fab813/refs': Value too large for defined data type; class=Os (2)
warning: spurious network error (2 tries remaining): could not read directory '/root/.cargo/git/db/zenoh-cc237f2570fab813/refs': Value too large for defined data type; class=Os (2)
warning: spurious network error (1 tries remaining): could not read directory '/root/.cargo/git/db/zenoh-cc237f2570fab813/refs': Value too large for defined data type; class=Os (2)
error: failed to get `zenoh` as a dependency of package `zenoh-c v0.11.0-dev (/opt/armhf-sysroot/data/ros2_zenoh_rmw/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor-build/release)`

Caused by:
  failed to load source for dependency `zenoh`

Caused by:
  Unable to update https://github.com/eclipse-zenoh/zenoh.git?branch=main#93f93d2d

Caused by:
  failed to clone into: /root/.cargo/git/db/zenoh-cc237f2570fab813

Caused by:
  could not read directory '/root/.cargo/git/db/zenoh-cc237f2570fab813/refs': Value too large for defined data type; class=Os (2)
gmake[5]: *** [CMakeFiles/cargo.dir/build.make:111: release/target/release/libzenohc.so] Error 101
gmake[4]: *** [CMakeFiles/Makefile2:180: CMakeFiles/cargo.dir/all] Error 2
gmake[3]: *** [Makefile:146: all] Error 2
gmake[2]: *** [CMakeFiles/zenoh_c_vendor.dir/build.make:85: zenoh_c_vendor-prefix/src/zenoh_c_vendor-stamp/zenoh_c_vendor-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/zenoh_c_vendor.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zenoh_c_vendor [28.5s, exited with code 2]

Summary: 0 packages finished [29.6s]
  1 package failed: zenoh_c_vendor
  1 package had stderr output: zenoh_c_vendor
  1 package not processed
```

I believe it has to do with the issue outlined [here](https://github.com/rust-lang/cargo/issues/8719). However, I am not familiar with cargo and unsure on how to apply a solution to the CMakeList.txt within the zenoh_c_vendor package.

jmount-boss, 2024-08-27:
One potential is to cross-compile, rather than use emulation. Zenoh-C cross compile steps are provided [here](https://github.com/eclipse-zenoh/zenoh-c?tab=readme-ov-file#cross-compilation). Summary of steps here:

1. Edit [zenoh_c_vendor/CMakeLists.txt](https://github.com/ros2/rmw_zenoh/blob/rolling/zenoh_c_vendor/CMakeLists.txt#L30) by adding following `"-DZENOHC_CUSTOM_TARGET=armv7-unknown-linux-gnueabihf"` top the CMAKE_ARGS

```cmake
ament_vendor(zenoh_c_vendor
  VCS_URL https://github.com/eclipse-zenoh/zenoh-c.git
  VCS_VERSION 548ee8dde0f53a58c06e68a2949949b31140c36c
  CMAKE_ARGS
    "-DZENOHC_CARGO_FLAGS=${ZENOHC_CARGO_FLAGS}" 
    "-DZENOHC_CUSTOM_TARGET=armv7-unknown-linux-gnueabihf"
)
```
2. Setup system ready for compilation
```bash
# Install arm gcc compiler
apt-get install gcc-arm-linux-gnueabihf

# install rustup - remove any rustc currently existing on system first
apt-get remove rustc
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y

# Source 
source $HOME/.carg/env

# Add rustup target
rustup target add armv7-unknown-linux-gnueabihf
```

3. Build
```
# Source ros2 
source <path/to/ros2/jazzy>/install/setup.bash

# Export rust flags
export RUSTFLAGS="-Clinker=arm-linux-gnueabihf-gcc -Car=arm-linux-gnueabihf-ar"

# Run colcon build
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```

Can, and should, the CMakeLists.txt be altered in some way to allow for a command line argument to be provided when running the colcon build command to set the `DZENOHC_CUSTOM_TARGET` flag?

clalancette, 2024-08-27:
> ```
> 
> ```
> 
> 
>     
>   
> 
> Can, and should, the CMakeLists.txt be altered in some way to allow for a command line argument to be provided when running the colcon build command to set the `DZENOHC_CUSTOM_TARGET` flag?

I think that is a fine option to have in `zenoh_c_vendor`.  We do something similar in https://github.com/ros2/yaml_cpp_vendor/blob/cb9b3a0cfd15a27e8362ce6deb183e486796f8ec/CMakeLists.txt#L29, for example.

A PR adding that option would be welcome.

jmount-boss, 2024-08-28:
Pull request created. Given a solution has been found to the issue, happy for conversation to occur in the PR instead as it is now more around implementation.

jmount-boss, 2024-08-28:
Closed with #272