ros2/rmw_zenoh #275: poradic

YuanYuYuan, 2024-09-02:


YuanYuYuan, 2024-09-02:
The analysis has been posted in rclpy https://github.com/ros2/rclpy/issues/1352

Yadunund, 2024-09-03:
@YuanYuYuan thanks for the detailed analysis! Curious if this issue only arises with `rclpy` or also `rclcpp`? Also, would this be related to https://github.com/ros2/rmw_zenoh/issues/170?

I still think shutting down the Zenoh session within `rmw_shutdown` is the right thing to do. If downstream packages such as `rclpy` and `rclcpp` are not equipped to handle cases where this function takes a while to execute, then we should fix expectations in those downstream expectations to handle these cases.

YuanYuYuan, 2024-09-03:
Hi @Yadunund, I believe this issue is mainly caused by rclpy and the fix is coming https://github.com/ros2/rclpy/pull/1353.

> Also, would this be related to https://github.com/ros2/rmw_zenoh/issues/170?

Actually, it would be a possible consequence once we fix rclpy. I think there has already been many debates in Rust community about this problem. See [here](https://github.com/rust-lang/rust/issues/126600) for more details. In short, Rust suggests not doing this way.

YuanYuYuan, 2024-11-18:
This issue has been fixed by https://github.com/ros2/rclpy/pull/1353

closed 2024-11-18 by YuanYuYuan