ros2/rmw_zenoh #279: rclcpp test_subscription_content_filter segfault

ahcorde, 2024-09-11:
I'm getting a segfault running the test `test_subscription_content_filter` in particular `TestContentFilterSubscription.content_filter_get_begin`.

```bash
Thread 1 "test_subscripti" received signal SIGSEGV, Segmentation fault.
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:317
317	../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory.
(gdb) bt
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:317
#1  0x00007ffff7b43bd7 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*> (this=0x7fffffff20a0, 
    __beg=0x64636635383239 <error: Cannot access memory at address 0x64636635383239>, __end=0x646366353832ca <error: Cannot access memory at address 0x646366353832ca>)
    at /usr/include/c++/11/bits/basic_string.tcc:225
#2  0x00007ffff532bda5 in rmw_zenoh_cpp::liveliness::Entity::id[abi:cxx11]() const (this=0x555555b46590) at /home/ahcorde/ros2_rolling/src/ros2/rmw_zenoh/rmw_zenoh_cpp/src/detail/liveliness_utils.cpp:582
#3  0x00007ffff535b466 in operator() (__closure=0x555555b325b0, queryable_prefix="c67248845cb613836ba09f84f75fca9d") at /home/ahcorde/ros2_rolling/src/ros2/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:1510
```
The memory address of `sub_data->entity` looks wrong `0x4

```bash
(gdb) print sub_data
$1 = (rmw_zenoh_cpp::rmw_subscription_data_t * const) 0x555555afdd30
(gdb) print sub_data->entity
$2 = std::shared_ptr<rmw_zenoh_cpp::liveliness::Entity> (expired, weak count 0) = {get() = 0x4}
```

Any thoughts @clalancette or @Yadunund ?

ahcorde, 2024-09-11:
Same issue is happening with `test_client` in `rclcpp_action`

ahcorde, 2024-09-11:
digging a into the code, I see that `set_querying_subscriber_callback` is adding the callback to a structure, but this structure is not clean it when the topic is removed.

@FYI @Yadunund

Yadunund, 2024-09-17:
I've opened a PR with a fix https://github.com/ros2/rmw_zenoh/pull/282

commit 25e51f27 (2024-09-17): Fix #279 by removing querying sub callback after subscription is destroyed (#282)
closed 2024-09-17 by ahcorde