ros2/rmw_zenoh #280: rclcpp `test_qos_event__rmw_zenoh_cpp` generating backtrace when test is finished

ahcorde, 2024-09-11:
While testing `test_qos_event__rmw_zenoh_cpp` I found this ugly error. Even If I skipped all tests the error is still there

```bash
130: thread '<unnamed>' panicked at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/thread/local.rs:246:26:
130: cannot access a Thread Local Storage value during or after destruction: AccessError
130: stack backtrace:
130:    0: rust_begin_unwind
130:              at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:645:5
130:    1: core::panicking::panic_fmt
130:              at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/core/src/panicking.rs:72:14
130:    2: core::result::unwrap_failed
130:              at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/core/src/result.rs:1649:5
130:    3: core::result::Result<T,E>::expect
130:              at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/core/src/result.rs:1030:23
130:    4: std::thread::local::LocalKey<T>::with
130:              at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/thread/local.rs:246:9
130:    5: tokio::runtime::context::runtime_mt::current_enter_context
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/context/runtime_mt.rs:5:5
130:    6: tokio::runtime::scheduler::multi_thread::worker::block_in_place::{{closure}}
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/scheduler/multi_thread/worker.rs:355:13
130:    7: tokio::runtime::scheduler::multi_thread::worker::with_current::{{closure}}
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/scheduler/multi_thread/worker.rs:1223:14
130:    8: tokio::runtime::context::with_scheduler::{{closure}}
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/context.rs:183:33
130:    9: core::result::Result<T,E>::unwrap_or_else
130:              at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/core/src/result.rs:1426:23
130:   10: tokio::runtime::context::with_scheduler
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/context.rs:182:9
130:   11: tokio::runtime::scheduler::multi_thread::worker::with_current
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/scheduler/multi_thread/worker.rs:1221:5
130:   12: tokio::runtime::scheduler::multi_thread::worker::block_in_place
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/scheduler/multi_thread/worker.rs:353:24
130:   13: tokio::runtime::scheduler::block_in_place::block_in_place
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/runtime/scheduler/block_in_place.rs:20:5
130:   14: tokio::task::blocking::block_in_place
130:              at /home/ahcorde/.cargo/registry/src/index.crates.io-6f17d22bba15001f/tokio-1.36.0/src/task/blocking.rs:78:9
130:   15: zenoh_runtime::ZRuntime::block_in_place
130:              at /home/ahcorde/.cargo/git/checkouts/zenoh-cc237f2570fab813/e66745e/commons/zenoh-runtime/src/lib.rs:130:9
130:   16: <zenoh_core::ResolveFuture<F,To> as zenoh_core::SyncResolve>::res_sync
130:              at /home/ahcorde/.cargo/git/checkouts/zenoh-cc237f2570fab813/e66745e/commons/zenoh-core/src/lib.rs:166:9
130:   17: zenoh_core::SyncResolve::res
130:              at /home/ahcorde/.cargo/git/checkouts/zenoh-cc237f2570fab813/e66745e/commons/zenoh-core/src/lib.rs:57:9
130:   18: z_close
130:              at /home/ahcorde/ros2_rolling/build/zenoh_c_vendor/zenoh_c_vendor-prefix/src/zenoh_c_vendor/src/session.rs:144:11
130:   19: rmw_shutdown
130:              at /home/ahcorde/ros2_rolling/src/ros2/rmw_zenoh/rmw_zenoh_cpp/src/rmw_init.cpp:415:14
130:   20: rmw_shutdown
130:              at /home/ahcorde/ros2_rolling/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:248:1
130:   21: rcl_shutdown
130:              at /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl/src/rcl/init.c:294:23
130:   22: _ZN6rclcpp7Context8shutdownERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
130:              at /home/ahcorde/ros2_rolling/src/ros2/rclcpp/rclcpp/src/rclcpp/context.cpp:320:31
130:   23: _ZN6rclcpp7ContextD2Ev
130:              at /home/ahcorde/ros2_rolling/src/ros2/rclcpp/rclcpp/src/rclcpp/context.cpp:159:22
130:   24: _ZN6rclcpp8contexts14DefaultContextD2Ev
130:              at /home/ahcorde/ros2_rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/contexts/default_context.hpp:26:7
130:   25: _ZN9__gnu_cxx13new_allocatorIN6rclcpp8contexts14DefaultContextEE7destroyIS3_EEvPT_
130:              at /usr/include/c++/11/ext/new_allocator.h:168:13
130:   26: _ZNSt16allocator_traitsISaIN6rclcpp8contexts14DefaultContextEEE7destroyIS2_EEvRS3_PT_
130:              at /usr/include/c++/11/bits/alloc_traits.h:535:15
130:   27: _ZNSt23_Sp_counted_ptr_inplaceIN6rclcpp8contexts14DefaultContextESaIS2_ELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv
130:              at /usr/include/c++/11/bits/shared_ptr_base.h:528:35
130:   28: _ZNSt16_Sp_counted_baseILN9__gnu_cxx12_Lock_policyE2EE10_M_releaseEv
130:              at /usr/include/c++/11/bits/shared_ptr_base.h:168:16
130:   29: _ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE2EED2Ev
130:              at /usr/include/c++/11/bits/shared_ptr_base.h:705:21
130:   30: _ZNSt12__shared_ptrIN6rclcpp8contexts14DefaultContextELN9__gnu_cxx12_Lock_policyE2EED2Ev
130:              at /usr/include/c++/11/bits/shared_ptr_base.h:1154:7
130:   31: _ZNSt10shared_ptrIN6rclcpp8contexts14DefaultContextEED2Ev
130:              at /usr/include/c++/11/bits/shared_ptr.h:122:11
[... 111 lines cut ...]
130:                                at /usr/include/c++/11/bits/alloc_traits.h:535:15
130:   50:     0x7f9cb79a69a7 - _ZNSt23_Sp_counted_ptr_inplaceIN6rclcpp8contexts14DefaultContextESaIS2_ELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv
130:                                at /usr/include/c++/11/bits/shared_ptr_base.h:528:35
130:   51:     0x559b0a4ade9d - _ZNSt16_Sp_counted_baseILN9__gnu_cxx12_Lock_policyE2EE10_M_releaseEv
130:                                at /usr/include/c++/11/bits/shared_ptr_base.h:168:16
130:   52:     0x559b0a4a7621 - _ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE2EED2Ev
130:                                at /usr/include/c++/11/bits/shared_ptr_base.h:705:21
130:   53:     0x7f9cb7997370 - _ZNSt12__shared_ptrIN6rclcpp8contexts14DefaultContextELN9__gnu_cxx12_Lock_policyE2EED2Ev
130:                                at /usr/include/c++/11/bits/shared_ptr_base.h:1154:7
130:   54:     0x7f9cb7997390 - _ZNSt10shared_ptrIN6rclcpp8contexts14DefaultContextEED2Ev
130:                                at /usr/include/c++/11/bits/shared_ptr.h:122:11
130:   55:     0x7f9cb6645495 - __run_exit_handlers
130:                                at ./stdlib/exit.c:113:8
130:   56:     0x7f9cb6645610 - __GI_exit
130:                                at ./stdlib/exit.c:143:3
130:   57:     0x7f9cb6629d97 - __libc_start_call_main
130:                                at ./csu/../sysdeps/nptl/libc_start_call_main.h:74:3
130:   58:     0x7f9cb6629e40 - __libc_start_main_impl
130:                                at ./csu/../csu/libc-start.c:392:3
130:   59:     0x559b0a484b35 - _start
130:   60:                0x0 - <unknown>
130: thread '<unnamed>' panicked at library/core/src/panicking.rs:163:5:
130: panic in a destructor during cleanup
130: thread caused non-unwinding panic. aborting.
```

ahcorde, 2024-09-11:
rclcpp_lifecycle
	 - test_state_machine_info (Failed)
	 - test_register_custom_callbacks (Failed)
	 - test_callback_exceptions (Failed)

ahcorde, 2024-09-11:
Related PR https://github.com/ros2/rclcpp/pull/2626

Yadunund, 2024-09-20:
This should be fixed by https://github.com/ros2/rclcpp/pull/2633

closed 2024-09-24 by ahcorde