ros2/rmw_zenoh #286: Test failures with rmw_zenoh

Yadunund, 2024-09-26:
This ticket will track the status of test failures with `rmw_zenoh`.

Below is the template to be used with an example. Let's copy paste this each time we make changes to the `ros2.repos` file from 
https://github.com/Yadunund/ros2/blob/yadu/rmw_zenoh_testing/ros2.repos

## Test setup
* Date: 26-09-2024
* ros2.repos file:  https://github.com/Yadunund/ros2/blob/c425fa9d0ced5bfca6e530141fc1f1629be7e720/ros2.repos

## rmw_zenoh/rolling
* rmw_zenoh commit: b3e4a6378063d4f73def3069a1f23c9e41fd4d89

| ROS Package | Failing test name | Failing test output |
|-------------|-------------------|----------------------------|
| rclcpp           |  33 - test_intra_process_manager |                        [gist](https://gist.github.com/Yadunund/0453d07786bb8b6d0b9b2155f63f241d) |
|                      |                                |                                                |
|                      |                                |                                                |

## rmw_zenoh/dev/1.0.0
* rmw_zenoh commit: 6a48919727226ebad853373e8a1eec0ee22c3be4

| ROS Package | Failing test name | Failing test output |
|-------------|-------------------|----------------------------|
|                      |                                |                                                |
|                      |                                |                                                |
|                      |                                |                                                |

ahcorde, 2024-09-27:
## rmw_zenoh/dev/1.0.0
* rmw_zenoh commit: * rmw_zenoh commit: 8d3fce9378636615a647f28c3f481dde7a54b7df and cherry-pick 8e883bcf64f6f023c621fb9ee11e5586e4e76150

| ROS Package | Failing test name | Failing test output |
|-------------|-------------------|----------------------------|
|          rcl            |              test_graph__rmw_zenoh_cpp                  |                       description below                         |
|          rcl_action            |             test_action_communication__rmw_zenoh_cpp                   |                      description below                           |
|          rcl_action            |                 test_action_communication__rmw_zenoh_cpp               |                       description below                          |
|          rcl_action            |                 test_graph__rmw_zenoh_cpp               |                       description below                          |
|          rcl_action            |                 test_action_server               |                       timeout                          |
|          rcl_lifecyle            |                 test_rcl_lifecycle               |                    Timeout                             |
| rcl_yaml_param_parser | None |description below |
| rclcpp| test_allocator_memory_strategy | timeout |
| rclcpp| test_intra_process_manager | description below |
| rclcpp| test_node_interfaces__node_graph | description below |
| rclcpp| test_parameter_client |description below  |
| rclcpp| test_publisher | description below |
| rclcpp| test_qos | description below |
| rclcpp| test_wait_for_message | description below |
| rclcpp| test_events_executor | timeout |
| rclcpp| test_wait_set | description below |
| rclcpp_Action| test_client | timeout |

### test_action_communication (rcl_action)

```bash
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:561: Failure
11: Expected equality of these values:
11:   ret
11:     Which is: 2
11:   0
11: error not set
11: 
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:573: Failure
11: Value of: this->is_status_ready
11:   Actual: false
11: Expected: true
11: 
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:581: Failure
11: Expected equality of these values:
11:   ret
11:     Which is: 4103
11:   0
11: error not set
11: 
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:584: Failure
11: Expected equality of these values:
11:   status_array.msg.status_list.size
11:     Which is: 1
11:   incoming_status_array.status_list.size
11:     Which is: 0

```

###  test_wait_set (rclcpp)

```bash
108: [ RUN      ] TestWaitSet.add_guard_condition_to_two_different_wait_set
108: unknown file: Failure
108: C++ exception with description "Failed to initialize event: provided event_type 7 is not supported by rmw_zenoh_cpp, at /home/ahcorde/ros2_rolling/src/ros2/rmw_zenoh/rmw_zenoh_cpp/src/rmw_event.cpp:53" thrown in the test body.
108: 
108: [  FAILED  ] TestWaitSet.add_guard_condition_to_two_different_wait_set (16 ms)
108: [ RUN      ] TestWaitSet.add_remove_wait
108: unknown file: Failure
108: C++ exception with description "Failed to initialize event: provided event_type 1 is not supported by rmw_zenoh_cpp, at /home/ahcorde/ros2_rolling/src/ros2/rmw_zenoh/rmw_zenoh_cpp/src/rmw_event.cpp:108" thrown in the test body.
```

### test_wait_for_message (rclcpp)

segfault
[... 214 lines cut ...]
13: [INFO] [1727443822.650237406] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443822.760814004] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443822.870295696] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_graph.cpp:347: Failure
13: Expected: (attempts) <= (max_attempts), actual: 5 vs 4
13: Unable to attain all required nodes
13: 
13: [  FAILED  ] TestActionGraphMultiNodeFixture.action_client_init_maybe_fail (5378 ms)
13: [ RUN      ] TestActionGraphMultiNodeFixture.rcl_get_client_names_and_types_by_node_maybe_fail
13: [INFO] [1727443828.031448905] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443828.145644359] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443828.256353051] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_graph.cpp:347: Failure
13: Expected: (attempts) <= (max_attempts), actual: 5 vs 4
13: Unable to attain all required nodes
13: 
13: [  FAILED  ] TestActionGraphMultiNodeFixture.rcl_get_client_names_and_types_by_node_maybe_fail (5339 ms)
13: [ RUN      ] TestActionGraphMultiNodeFixture.rcl_get_server_names_and_types_by_node_maybe_fail
13: [INFO] [1727443833.367802389] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443833.480293497] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443833.590140755] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_graph.cpp:347: Failure
13: Expected: (attempts) <= (max_attempts), actual: 5 vs 4
13: Unable to attain all required nodes
```

ahcorde, 2024-09-30:
## rmw_zenoh/dev/1.0.0
* rmw_zenoh commit: * rmw_zenoh commit: 63530b717681b15a6b84f49adfaa0ed78fed902a

| ROS Package | Failing test name | Failing test output |
|-------------|-------------------|----------------------------|
|          rcl            |              test_graph__rmw_zenoh_cpp                  |                       description below                         |
|          rcl_action            |             test_action_communication__rmw_zenoh_cpp                   |                      description below                           |
|          rcl_action            |                 test_graph__rmw_zenoh_cpp               |                       description below                          |
|          rcl_action            |                 test_action_server               |                       timeout                          |
|          rcl_lifecyle            |                 test_rcl_lifecycle               |                    Timeout                             |
| rcl_yaml_param_parser | None |description below |
| rclcpp| test_allocator_memory_strategy | timeout |
| rclcpp| test_clock | flaky |
| rclcpp| test_intra_process_manager | description below |
| rclcpp| test_node_interfaces__node_graph | description below |
| rclcpp| test_parameter_client |description below  |
| rclcpp| test_publisher | description below |
| rclcpp| test_qos | description below |
| rclcpp| test_wait_for_message | description below |
| rclcpp| test_events_executor | timeout |
| rclcpp| test_wait_set | description below |
| rclcpp_action| test_client | timeout |
| rclcpp_action| test_server | timeout |

### test_action_communication (rcl_action)

```bash
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:561: Failure
11: Expected equality of these values:
11:   ret
11:     Which is: 2
11:   0
11: error not set
11: 
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:573: Failure
11: Value of: this->is_status_ready
11:   Actual: false
11: Expected: true
11: 
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:581: Failure
11: Expected equality of these values:
11:   ret
11:     Which is: 4103
11:   0
11: error not set
11: 
11: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_action_communication.cpp:584: Failure
11: Expected equality of these values:
11:   status_array.msg.status_list.size
11:     Which is: 1
11:   incoming_status_array.status_list.size
11:     Which is: 0

```

###  test_wait_set (rclcpp)

```bash
108: [ RUN      ] TestWaitSet.add_guard_condition_to_two_different_wait_set
108: unknown file: Failure
108: C++ exception with description "Failed to initialize event: provided event_type 7 is not supported by rmw_zenoh_cpp, at /home/ahcorde/ros2_rolling/src/ros2/rmw_zenoh/rmw_zenoh_cpp/src/rmw_event.cpp:53" thrown in the test body.
108: 
108: [  FAILED  ] TestWaitSet.add_guard_condition_to_two_different_wait_set (16 ms)
108: [ RUN      ] TestWaitSet.add_remove_wait
108: unknown file: Failure
108: C++ exception with description "Failed to initialize event: provided event_type 1 is not supported by rmw_zenoh_cpp, at /home/ahcorde/ros2_rolling/src/ros2/rmw_zenoh/rmw_zenoh_cpp/src/rmw_event.cpp:108" thrown in the test body.
```

### test_wait_for_message (rclcpp)
segfault
[... 226 lines cut ...]
13: [INFO] [1727443822.650237406] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443822.760814004] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443822.870295696] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_graph.cpp:347: Failure
13: Expected: (attempts) <= (max_attempts), actual: 5 vs 4
13: Unable to attain all required nodes
13: 
13: [  FAILED  ] TestActionGraphMultiNodeFixture.action_client_init_maybe_fail (5378 ms)
13: [ RUN      ] TestActionGraphMultiNodeFixture.rcl_get_client_names_and_types_by_node_maybe_fail
13: [INFO] [1727443828.031448905] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443828.145644359] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443828.256353051] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_graph.cpp:347: Failure
13: Expected: (attempts) <= (max_attempts), actual: 5 vs 4
13: Unable to attain all required nodes
13: 
13: [  FAILED  ] TestActionGraphMultiNodeFixture.rcl_get_client_names_and_types_by_node_maybe_fail (5339 ms)
13: [ RUN      ] TestActionGraphMultiNodeFixture.rcl_get_server_names_and_types_by_node_maybe_fail
13: [INFO] [1727443833.367802389] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443833.480293497] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: [INFO] [1727443833.590140755] [rmw_zenoh_cpp]: Successfully connected to a Zenoh router with id ffa65399a158d8ccd21b64105f6c7af4.
13: /home/ahcorde/ros2_rolling/src/ros2/rcl/rcl_action/test/rcl_action/test_graph.cpp:347: Failure
13: Expected: (attempts) <= (max_attempts), actual: 5 vs 4
13: Unable to attain all required nodes
```

YuanYuYuan, 2024-09-30:
The failures of test_graph should be resolved once the RCL changes the logic in [this PR](https://github.com/ros2/rcl/pull/1192).

Yadunund, 2024-10-01:
@ahcorde can we update the ros2.repos file in https://github.com/yadunund/ros2 with the latest commits in rclcpp/rcl and re-run tests for both `rolling` and `dev/1.0.0`?

ahcorde, 2024-10-02:
## Test setup
* Date: 02-10-2024
* ros2.repos file:  https://gist.github.com/ahcorde/dccfd8cf6e102baf10d45bda22995f68

## rmw_zenoh/rolling
* rmw_zenoh commit: cf8e8a2eccdf2d4bcce6858d35ed1eac14ce0a0a

| ROS Package | Failing test name | Failing test output |
|-------------|-------------------|----------------------------|
| rcl | None | |
| rcl_action | None  | |
| rcl_lifecyle | test_rcl_lifecycle | segfault |
| rcl_yaml_param_parser | None | |
| rclcpp| test_intra_process_manager | description below |
| rclcpp| test_node_interfaces__node_graph | description below |
| rclcpp| test_publisher | description below |
| rclcpp| test_qos | description below |
| rclcpp| test_events_executor | Timeout |
| rclcpp| test_wait_set | description below |
| rclcpp_action| None |  |
| rclcpp_lifecycle| None |  |
| rclcpp_lifecycle| None |  |

@Yadunund There is a new regression on `rcl_lifecycle`. A segfault in this test `test_rcl_lifecycle`

ahcorde, 2024-10-02:
PR to fix rclcpp.test_qos https://github.com/ros2/rclcpp/pull/2639

Yadunund, 2024-10-02:
@ahcorde thanks for reporting the regression. I can reproduce it. seems to be have caused by #278.

clalancette, 2024-10-02:
> @ahcorde thanks for reporting the regression. I can reproduce it. seems to be have caused by #278.

Yes.  Looking at it, the problem is that `PublisherData::make` can create a "partially-initialized" `PublisherData` object.  Basically it can create the object, and then something further down can fail.  At that point, the destructor for `PublisherData` will be called, at which point it will try to do `shutdown()`.  Since `shutdown()` expects a fully-initialized object, it can access invalid memory.

One way to solve this would be to not actually create the `PublisherData` object until we have already initialized everything.  That is, we would initialize everything into stack variables, and then do something like:

```
return std::make_shared<PublisherData>(node, gid, type_support_impl, type_support, entity, pub_cache, pub, token, events_mgr);
```

YuanYuYuan, 2024-10-03:
There is a mistake in dev/1.0.0 which has been fixed in this [commit](https://github.com/ZettaScaleLabs/rmw_zenoh/commit/c386a3dd1678c817d8c4aa698ee312d04c787025). Now it can pass all the rcl_action test.

```console
100% tests passed, 0 tests failed out of 22

Label Time Summary:
copyright     =   1.32 sec*proc (1 test)
cppcheck      =   0.26 sec*proc (1 test)
cpplint       =   2.40 sec*proc (1 test)
gtest         =  65.53 sec*proc (16 tests)
lint_cmake    =   0.27 sec*proc (1 test)
linter        =   6.41 sec*proc (6 tests)
uncrustify    =   0.81 sec*proc (1 test)
xmllint       =   1.35 sec*proc (1 test)

Total Test time (real) =  71.95 sec
---
Finished <<< rcl_action [1min 12s]

Summary: 1 package finished [1min 13s]
```

imstevenpmwork, 2024-10-03:
Hello guys, here's the latest update of the week.

# Test Setup
* Date: 03-10-2024
* ros2.repos file: TBD
* ROS Rolling
* Ubuntu 24 (Containerised CI)

# Version
* rolling rmw_zenoh: https://github.com/ros2/rmw_zenoh/commit/cf8e8a2eccdf2d4bcce6858d35ed1eac14ce0a0a
* dev/1.0.0 rmw_zenoh: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/c386a3dd1678c817d8c4aa698ee312d04c787025
* zenoh-c/zenoh: https://github.com/eclipse-zenoh/zenoh-c/commit/278726e50f31c3df1e89331a9a58f7d304131599 (1.0.0.10)
* rclcpp: https://github.com/ros2/rclcpp/commit/1a0092a65b2d0f97b57a31321d44827f07b0320e
* rcl: https://github.com/ros2/rcl/commit/d680ff3a72cd2bdf90f8aeccf2ead38171ecfb84
* system_tests: https://github.com/ros2/system_tests/commit/5f73908afa3e9b86dbc42163e25e6ceba6f7e77c

# Logs
rolling rmw_zenoh: [logs_system_test_rolling.zip](https://github.com/user-attachments/files/17241930/logs_system_test_rolling.zip)
dev/1.0.0 rmw_zenoh: [logs_system_test_dev100.zip](https://github.com/user-attachments/files/17242268/logs_system_test_dev100.zip)

# rmw_zenoh/rolling
* rmw_zenoh commit: https://github.com/ros2/rmw_zenoh/commit/cf8e8a2eccdf2d4bcce6858d35ed1eac14ce0a0a

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | None       | None   |
| rcl_action   | None        | None      |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param   | None        | None      |
| rclcpp   | test_intra_process_manager        | TBD      |
| rclcpp   | test_node_interfaces__node_graph        | TBD      |
| rclcpp   | test_publisher        | TBD      |
| rclcpp   | test_events_executor        | TBD      |
| rclcpp   | test_wait_set        | TBD      |
| rclcpp_action   | test_client        | TBD      |
| rclcpp_components   | None        | None      |
| rclcpp_lifecycle   | test_lifecycle_node        | TBD      |
| rmw_zenoh_cpp   | None        | None      |
| test_cli   | None        | None      |
| test_cli_remap   | test_cli_remapping        | TBD      |
| test_communication   | test_requester_replier__rclpy__rmw_zenoh_cpp        | TBD      |
| test_communication   | test_action_client_server__rclpy__rmw_zenoh_cpp        | TBD      |
| test_communication   | test_requester_replier__rclcpp__rclpy__rmw_zenoh_cpp        | TBD      |
| test_communication   | test_action_client_server__rclcpp__rclpy__rmw_zenoh_cpp        | TBD      |
| test_qos   | test_deadline__rmw_zenoh_cpp        | TBD      |
| test_qos   | test_liveliness__rmw_zenoh_cpp        | TBD      |
| test_qos   | test_best_available__rmw_zenoh_cpp        | TBD      |
| test_rclcpp   | None        | None      |
| test_security   | None        | None      |

### Nav2 Demo: Working

# rmw_zenoh/dev/1.0.0
* rmw_zenoh commit: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/c386a3dd1678c817d8c4aa698ee312d04c787025

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | None       | None   |
| rcl_action   | test_graph__rmw_zenoh_cpp        | TBD      |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param   | None        | None      |
| rclcpp   | test_intra_process_manager        | TBD      |
| rclcpp   | test_node_interfaces__node_graph        | TBD      |
| rclcpp   | test_publisher        | TBD      |
| rclcpp   | test_publisher_subscription_count_api        | TBD      |
| rclcpp   | test_wait_for_message        | TBD      |
| rclcpp   | test_events_executor        | TBD      |
| rclcpp   | test_wait_set        | TBD      |
| rclcpp_action   | None        | None      |
| rclcpp_components   | test_component_manager        | TBD      |
| rclcpp_lifecycle   | test_lifecycle_node        | TBD      |
| rclcpp_lifecycle   | test_lifecycle_service_client        | TBD      |
| rmw_zenoh_cpp   | None        | None      |
| test_cli   | None        | None      |
| test_cli_remap   | test_cli_remapping        | TBD      |
| test_communication   | test_requester_replier__rclpy__rmw_zenoh_cpp        | TBD      |
| test_communication   | test_action_client_server__rclpy__rmw_zenoh_cpp        | TBD      |
| test_communication   | test_requester_replier__rclcpp__rclpy__rmw_zenoh_cpp        | TBD      |
| test_communication   | test_action_client_server__rclcpp__rclpy__rmw_zenoh_cpp        | TBD      |
| test_communication   | test_publisher_subscriber__rclcpp__rmw_zenoh_cpp        | TBD      |
| test_qos   | test_deadline__rmw_zenoh_cpp        | TBD      |
| test_qos   | test_liveliness__rmw_zenoh_cpp        | TBD      |
| test_qos   | test_best_available__rmw_zenoh_cpp        | TBD      |
| test_rclcpp   | test_services_cpp__rmw_zenoh_cpp        | TBD      |
| test_rclcpp   | test_services_cpp__rmw_zenoh_cpp        | TBD      |
| test_security   | None        | None      |

### Nav2 Demo: Not Working

# Summary
rmw_zenoh/rolling: 15
rmw_zenoh/dev/1.0.0: 22

YuanYuYuan, 2024-10-03:
The reason to 1.0 is failing in rcl/test_wait_for_message is we try to fetch the z_id info of the session while creating a subscription AFTER the shutdown is called. Since the session is not alive anymore, it would cause a segfault.

Yadunund, 2024-10-03:
@imstevenpmwork can you retry the tests with Rolling built from source using the latest `ros2.repos` from [here](https://github.com/Yadunund/ros2/blob/yadu/rmw_zenoh_testing/ros2.repos)?

ahcorde, 2024-10-04:
This [PR](https://github.com/ros2/rclcpp/pull/2643) should fix  `test_events_executor` in `rclcpp`

ahcorde, 2024-10-04:
This [PR](https://github.com/ros2/system_tests/pull/550) should fix some of the failures in test_communication

ahcorde, 2024-10-04:
Thi [PR](https://github.com/ros2/system_tests/pull/551) should fix `test_deadline__rmw_zenoh_cpp` and `test_liveliness__rmw_zenoh_cpp`

ahcorde, 2024-10-04:
This [PR](https://github.com/ros2/system_tests/pull/552) should fix `test_cli_remapping`

ahcorde, 2024-10-04:
This [PR](https://github.com/ros2/rclcpp/pull/2644) should fix `test_wait_set` in `rclcpp`

ahcorde, 2024-10-04:
I found some issue on `rclpy` this [PR](https://github.com/ros2/rclpy/pull/1369) should fix then

imstevenpmwork, 2024-10-10:
Hello guys, here's the latest update of the week.

# Test Setup
* Date: 09-10-2024
* ros2.repos file: TBD
* ROS Rolling
* Ubuntu 24 (Containerised CI)

# Version
* rolling rmw_zenoh: https://github.com/ros2/rmw_zenoh/commit/439d6dc6b0325811793533414dee89196cd72186
* dev/1.0.0 rmw_zenoh: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/6ba342bea55de168ca44b4e7facc25b14d6cb4e5
* zenoh-c: https://github.com/eclipse-zenoh/zenoh-c/commit/e270137cbe109032feecf06389b0560e75fc8c84
* zenoh: https://github.com/eclipse-zenoh/zenoh/commit/7e32e7295f85f5fd196d94afff26cc48584c938c
* rclcpp: https://github.com/ros2/rclcpp/commit/41d3a27437973e85cf8fbb5612beb82fb6811e98
* rcl: https://github.com/ros2/rcl/commit/d9daca7c0af7ecdcd103ece30bb7c64bc919ef24
* system_tests: https://github.com/ros2/system_tests/commit/b698ef5329de45a167a451206a46ec2075350921
* nav2: https://github.com/ros-navigation/navigation2/commit/6311978e0902b807a05883b3764e72f731d446f8

# Logs
rolling rmw_zenoh: TBD
dev/1.0.0 rmw_zenoh: TBD

# rmw_zenoh/rolling
* rmw_zenoh commit: https://github.com/ros2/rmw_zenoh/commit/439d6dc6b0325811793533414dee89196cd72186

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | None       | None   |
| rcl_action   | None        | None      |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param   | None        | None      |
| rclcpp   | test_intra_process_manager        | Failed (Check logs)      |
| rclcpp   | test_node_interfaces__node_graph        | Failed (Check logs)      |
| rclcpp   | test_publisher        | Failed (Check logs)      |
| rclcpp   | test_events_executor        | Timeout (Check logs)      |
| rclcpp_action   | None        | None      |
| rclcpp_components   | None        | None      |
| rclcpp_lifecycle  | test_lifecycle_node       | Failed (Check logs)      |
| rmw_zenoh_cpp   | None        | None      |
| test_cli   | None        | None      |
| test_cli_remap   | test_cli_remapping        | Failed (Check logs)      |
| test_communication   | test_requester_replier__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_communication   | test_action_client_server__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_communication   | test_requester_replier__rclcpp__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_communication   | test_action_client_server__rclcpp__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_qos   | test_deadline__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_qos   | test_liveliness__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_qos   | test_best_available__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_rclcpp   | None        | None      |
| test_security   | None        | None      |

### Nav2 Demo: TBD

# rmw_zenoh/dev/1.0.0
* rmw_zenoh commit: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/6ba342bea55de168ca44b4e7facc25b14d6cb4e5

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | None       | None   |
| rcl_action   | None        | None      |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param   | None        | None      |
| rclcpp   | test_intra_process_manager        | Failed (Check logs)      |
| rclcpp   | test_node_interfaces__node_graph        | Failed (Check logs)      |
| rclcpp   | test_publisher        | Failed (Check logs)      |
| rclcpp   | test_executors_timer_cancel_behavior        | Timeout (Check logs)      |
| rclcpp   | test_events_executor        | Timeout (Check logs)      |
| rclcpp   | test_executor        | Timeout (Check logs)      |
| rclcpp_action   | test_client        | Timeout (Check logs)      |
| rclcpp_components   | None        | None     |
| rclcpp_lifecycle   | test_lifecycle_node        | Failed (Check logs)      |
| rclcpp_lifecycle   | test_lifecycle_service_client        | Timeout (Check logs)      |
| rmw_zenoh_cpp   | cpplint        | Failed (Check logs)      |
| rmw_zenoh_cpp   | uncrustify        | Failed (Check logs)      |
| test_cli   | None        | None      |
| test_cli_remap   | test_cli_remapping        | Failed (Check logs)      |
| test_communication   | test_requester_replier__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_communication   | test_action_client_server__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_communication   | test_requester_replier__rclcpp__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_communication   | test_action_client_server__rclcpp__rclpy__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_communication   | test_publisher_subscriber__rclcpp__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_qos   | test_deadline__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_qos   | test_liveliness__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_qos   | test_best_available__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_rclcpp   | gtest_spin__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_rclcpp   | test_n_nodes__rmw_zenoh_cpp        | Failed (Check logs)      |
| test_rclcpp   | test_node_name__rmw_zenoh_cpp        | Timeout (Check logs)      |
| test_security   | None        | None      |

### Nav2 Demo: TBD

# Summary
rmw_zenoh/rolling: 13 failing tests
rmw_zenoh/dev/1.0.0: 22 failing tests
Delta: 9 tests (-2 format)
Note: mind some of them might/are flaky

ahcorde, 2024-10-31:
# [rmw_zenoh/dev/1.0.0](https://github.com/ros2/rmw_zenoh/tree/ahcorde/rolling/1.0.0_windows)
* rmw_zenoh commit: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/59cbc5966c15c4bb915d8df77a970950340acd73

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | test_timer__rmw_zenoh_cpp       | None   |
| rcl      | test_arguments       | None   |
| rcl      | test_secutiry       | None   |
| rcl_action   | test_graph__rmw_zenoh_cpp        | Timeout (Check logs)      |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param_parser   | None        | None      |
| rclcpp   | test_generic_client        |       |
| rclcpp   | test_generic_service        |       |
| rclcpp   | test_client_common       |       |
| rclcpp   | test_node       |       |
| rclcpp   | test_publisher        |       |
| rclcpp   | test_typesupport_helpers        |       |
| rclcpp   | test_executors        | Failed (Check logs)      |
| rclcpp   | test_executor        | Failed (Check logs)      |
| rclcpp   | test_generic_pubsub__rmw_zenoh_cpp        | Failed (Check logs)      |
| rclcpp_action   | None        |       |
| rclcpp_lifecycle   | None        | Failed (Check logs)      |
| rmw_zenoh_cpp   | None        | None      |

Yadunund, 2024-10-31:
@ahcorde did you get a chance to run the tests on Windows with the `rolling` branch?

ahcorde, 2024-11-04:
> @ahcorde did you get a chance to run the tests on Windows with the `rolling` branch?

When I try to execute the `rclcpp` test my virtual machine just dead. I'm looking in to it, but `rcl` and `rcl_action` have more failing test

Yadunund, 2024-11-07:
Maybe we just need to put existing information together but can we have a single follow-up comment in the format [above](https://github.com/ros2/rmw_zenoh/issues/286#issue-2551475063) where there are two tables (rolling vs dev/1.0.0) along with the ros2.repos file used to generate test results for both tables? Such a view would help provide a clear overview of the state of the two branches.

ahcorde, 2024-11-08:
# Windows 

## zenoh 1.0.0

# [rmw_zenoh/dev/1.0.0](https://github.com/ros2/rmw_zenoh/tree/ahcorde/rolling/1.0.0_windows)
* rmw_zenoh commit: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/2fa15bb1861982edf538fc986f486010224ad9f1

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | test_service__rmw_zenoh_cpp| None   |
| rcl_action   | test_graph__rmw_zenoh_cpp        |  flaky  |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param_parser   | None        | None      |
| rclcpp   | test_publisher|       |
| rclcpp_action   | None        |       |
| rclcpp_lifecycle   | None        | Failed (Check logs)      |

ahcorde, 2024-11-08:
[ros2.repos](https://gist.github.com/ahcorde/9eb30a197bbd8e6f576d5f5372ddf93a)

# rmw_zenoh/dev/1.0.0
* rmw_zenoh commit: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/28d917e0532fed6e9703f044662298f2b3553716

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | None       | None   |
| rcl_action   | test_graph__rmw_zenoh_cpp        | flaky      |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param   | None        | None      |
| rclcpp   | test_publisher        | Failed (Check logs)      |
| rclcpp   | test_events_executor        | Timeout (Check logs)      |
| rclcpp_action   | test_server        |       |
| rclcpp_components   | test_component_manager        | Failed (Check logs)      |
| rclcpp_lifecycle   | test_lifecycle_node        | Failed (Check logs)      |
| rclcpp_lifecycle   | test_lifecycle_service_client        | Timeout (Check logs)      |

# clalancette/cleanup-rmw-context-impl
* rmw_zenoh commit: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/e16b6d78d68ac524ff7b103bc6cfa27ca2b9cf3a

| ROS Package      | Failing test name | Failing test output     |
| :---:        |    :----:   |          :---: |
| rcl      | None       | None   |
| rcl_action   | None | None |
| rcl_lifecycle   | None        | None      |
| rcl_yaml_param   | None        | None      |
| rclcpp   | test_publisher        | Failed (Check logs)      |
| rclcpp   | test_events_executor        | Timeout (Check logs)      |
| rclcpp_action   | None   |   |
| rclcpp_components   | None  |   |
| rclcpp_lifecycle   | None  |  |
| rclcpp_lifecycle   | None |None |

YuanYuYuan, 2024-11-18:
- Action logs: https://github.com/ZettaScaleLabs/zenoh_rmw_eval/actions/runs/11886816016/job/33118736023
- Commit: https://github.com/ZettaScaleLabs/rmw_zenoh/commit/2743344795dfbaf7bf5447be373de9c534858af2

| Pacakge                 | Test                                                                  | Reason                                                    |
| -                       | -                                                                     | -                                                         |
| test_quality_of_service | 9 - test_deadline__rmw_zenoh_cpp (Failed)                             | Will be skipped                                           |
| test_quality_of_service | 11 - test_liveliness__rmw_zenoh_cpp (Failed)                          | Will be skipped                                           |
| test_quality_of_service | 12 - test_best_available__rmw_zenoh_cpp (Failed)                      | Also failed in 0.11.0                                     |
| rclcpp_lifecycle        | 6 - test_lifecycle_node (Failed)                                      | Hard to reproduce                                         |
| test_communication      | 80 - test_requester_replier__rclcpp__rclpy__rmw_zenoh_cpp (Failed)    | Flaky                                                     |
| test_communication      | 81 - test_action_client_server__rclcpp__rclpy__rmw_zenoh_cpp (Failed) | Flaky                                                     |
| test_rclcpp             | 70 - test_parameter_server_cpp__rmw_zenoh_cpp (Timeout)               | Hard to reproduce                                         |
| rclcpp                  | 62 - test_publisher (Failed)                                          | Also failed in 0.11.0                                     |
| rclcpp                  | 86 - test_time (Failed)                                               | Flaky                                                     |
| rclcpp                  | 118 - test_executor (Failed)                                          | Flaky                                                     |
| rclpy                   | 3 - test_action_client (Failed)                                       | [Issue](https://github.com/ros2/rmw_zenoh/issues/315)     |

clalancette, 2024-12-17:
While I know for sure that we aren't yet passing all tests, I think we are down to few enough that we should track them individually.  Thus I'm going to close out this tracking bug, and instead opt for individual bug reports.  Please feel free to reopen if you disagree.

closed 2024-12-17 by clalancette