ros2/rmw_zenoh #303: [Question] Humble support

jelavice, 2024-10-29:
Hi folks,

we saw zenoh demo on roscon 2024 and we were convinced by its reliability and simplicity of use. Are there any releases for humble?

Thanks for the info.

Edo

christophebedard, 2024-10-29:
The README mentions that the `rolling` branch supports Rolling/Jazzy/Iron: https://github.com/ros2/rmw_zenoh#requirements. It doesn't mention Humble, and there is no `humble` branch.

I personally do not know if there are any plans to support Humble, but I would let a maintainer chime in.

clalancette, 2024-10-29:
> I personally do not know if there are any plans to support Humble, but I would let a maintainer chime in.

We've done testing on Iron, Jazzy, and Rolling so far.  We know that it doesn't currently compile for Humble, but we haven't looked into why.  It is probably something relatively easy to fix, but it isn't one of our priorities right now.  Once we get it into Rolling as a Tier-1, we may have time to look into it.  In the meantime, if you'd like to contribute a fix, we'd be happy to review it.

jelavice, 2024-11-07:
Thanks for the info

0xjairo, 2024-11-07:
When attempting to build on humble, the top error is related to using a `rosidl_runtime_c` api that is [not present in Humble](https://docs.ros.org/en/humble/p/rosidl_runtime_c/generated/index.html#full-c-api):

```sh
Starting >>> rmw_zenoh_cpp
--- stderr: rmw_zenoh_cpp
/home/0xjairo/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/graph_cache.cpp:35:10: fatal error: rosidl_runtime_c/type_hash.h: No such file or directory
   35 | #include "rosidl_runtime_c/type_hash.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/rmw_zenoh_cpp.dir/build.make:132: CMakeFiles/rmw_zenoh_cpp.dir/src/detail/graph_cache.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/0xjairo/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/rmw_data_types.cpp:26:
/home/0xjairo/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/rmw_data_types.hpp:31:10: fatal error: rosidl_runtime_c/type_hash.h: No such file or directory
   31 | #include "rosidl_runtime_c/type_hash.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

```

clalancette, 2025-01-02:
This was fixed in #388, so closing.

closed 2025-01-02 by clalancette