ros2/rmw_zenoh #323: [Feature] Introduce include-what-you-use into the build pipeline

YuanYuYuan, 2024-11-22:
I see. I'm unaware of this pattern. We usually trust the compiler in Rust. But when I studied this pattern and tried to apply the tool [include-what-you-use](https://github.com/include-what-you-use/include-what-you-use), I got many warnings. What do you think about introducing the check of include-what-you-use?

[iwyu.txt](https://github.com/user-attachments/files/17827869/iwyu.txt)

_Originally posted by @YuanYuYuan in https://github.com/ros2/rmw_zenoh/pull/276#discussion_r1849899825_

YuanYuYuan, 2024-11-22:
Furthermore, I think we need to wrap the ROS dependencies with brackets rather than quotes as include-what-you-use suggsted. And this way is also recommended in the ROS community. https://answers.ros.org/question/348742/double-quotes-vs-angled-brackets-for-rosrosh/

Yadunund, 2025-01-16:
We're relying on standard ROS linters like `ament_uncrustify` and `ament_cpplint` in `rmw_zenoh` so any changes to conventions or linters that enforce conventions should be discussed upstream.

closed 2025-01-16 by Yadunund