ros2/rmw_zenoh #324: The solution proposal for panic in the program termination

evshary, 2024-11-26:
I would like to raise a discussion about the possible solution for https://github.com/ros2/rmw_zenoh/tree/8306a63313cd9e21883344deff9abc782636a464?tab=readme-ov-file#crash-when-program-terminates
I thought we had reached an agreement on this in the meeting before. Nevertheless, it might be worth writing it down here to make it clearer.

I know we expect users to shutdown the ROS Context explicitly, but it would be good if there is a way to detect this, or we might need to add shutdown to all the packages without it. For example, in `ros_gz` used by nav2
https://github.com/ZettaScaleLabs/rmw_zenoh/issues/26#issuecomment-2423370366

## The proposal:

The Zenoh session can't be closed while terminating the program due to the reason described in README. However, while calling `rmw_shutdown`, there is no available way to know if we can close the Zenoh session. Therefore, we come up with a mechanism to detect the termination of the program. We can register the `atexit` function, and while the entering the termination stage, it will toggle a flag. If the flag is set, we shouldn't touch the Zenoh session to avoid panic.
For more information, please check the PR here: https://github.com/ZettaScaleLabs/rmw_zenoh/pull/37/files

## Some questions:

1. Is this logic only necessary with 1.0.0?
    - Not really. I can do the porting to rolling if we all agree with the changes.
2. How do we ensure two or more threads do not access is_exiting concurrently?
    - I thought it didn't cause any issue since this is a one-way flag, which means we don't toggle it back to false. However, I can add atomic here to avoid possible concurrency issues.
3. When the program terminates, static ordering is unknown so there is no guarantee that is_exiting is valid. 
    - I'm not sure whether I misunderstood the spec or not. According to [the spec here](https://en.cppreference.com/w/cpp/utility/program/exit), the destructor will be called in a reverse order. I think that means the global variable `static bool is_exiting` should be valid before the `Context` is shutdown. Feel free to correct me if I'm wrong.

Let me know your thoughts. Thank you!

evshary, 2024-11-28:
After [the PR](https://github.com/eclipse-zenoh/zenoh-c/pull/835), we might not need this workaround
cc @yellowhatter here

YuanYuYuan, 2024-12-05:
This should resolve https://github.com/ZettaScaleLabs/rmw_zenoh/issues/35

YuanYuYuan, 2024-12-05:
@yellowhatter Please check if it can fix this https://github.com/Yadunund/rmw_zenoh_shutdown_test

Yadunund, 2025-04-05:
it sounds like #591 resolves this issue on systems other than windows. Do we still want to keep this ticket open?

closed 2025-04-06 by evshary