ros2/rmw_zenoh #337: Clock topic in bag negative

thomasemter, 2024-12-11:
Hi,

we are running some test with rmw_zenoh and the results are very promising. Somehow peculiar is the `/clock` when playing back a bag file since it shows negative timestamps.

# Steps to reproduce
## Record bag file:

```bash
# terminal 1
ros2 run rmw_zenoh_cpp rmw_zenohd
```

```bash
# terminal 2
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
ros2 run demo_nodes_cpp talker
```

```bash
# terminal 3
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
ros2 bag record -a
```

## Play back and echo clock:

```bash
# terminal 3
ros2 bag play rosbag2_...
```

```bash
# terminal 2
ros2 topic echo /clock
```

## Result:

```bash
# terminal 2
clock:
  sec: -1142771387
  nanosec: 756485177
---
clock:
  sec: -1142771387
  nanosec: 781476412
---
clock:
  sec: -1142771387
  nanosec: 806413161
---
clock:
  sec: -1142771387
  nanosec: 831491893
---
clock:
  sec: -1142771387
  nanosec: 856471721
```

clalancette, 2024-12-11:
Can you please explain how you are getting the `/clock` topic?  That is not published when you just run `ros2 run demo_nodes_cpp talker`.

thomasemter, 2024-12-11:
Oh, I am so sorry. The `--clock` got lost in copying. It should indeed be
```
# terminal 3
ros2 bag play rosbag2_... --clock
```
I fixed the original post, too.

clalancette, 2024-12-11:
OK, thanks.  Now I can reproduce.

So I see what the problem is.  The problem is that during the reception of a message, we are using the timestamp that is coming from Zenoh for the delivery time.  But that time is actually an NTP64 time, not a number of nanoseconds since the epoch.  Since the rest of our system depends on nanoseconds since the epoch, we have a mis-match.

Tomorrow I'll work on a fix to convert between them.  Thanks for bringing this up, it is an important problem.

thomasemter, 2024-12-12:
Thank you for looking into it.

closed 2024-12-13 by clalancette

thomasemter, 2024-12-13:
Thank you very much :-)

PR #338 (2024-12-13): fix: correct the received timestamp
Closes #337. This timestamp should be the one when received rather than the one when sent.

```console
➜ ros2 topic echo /clock
clock:
  sec: 1734081513
  nanosec: 389786241
---
clock:
  sec: 1734081513
  nanosec: 414703983
---
clock:
  sec: 1734081513
  nanosec: 439739606
---

```