ros2/rmw_zenoh #349: Shared Memory Support for Large Messages

cesurs, 2024-12-19:
Hello, I realized that when i use shared memory with less than ~4.2 MB of data Zenoh can use shared memory but when I start publishing large messages I get the error:
```
  WARN ThreadId(06) zenoh_shm: SharedMemoryManager does not have sufficient free memory to allocate 5000010 bytes, try de-fragmenting! at /root/.cargo/git/checkouts/zenoh-cc237f2570fab813/e66745e/commons/zenoh-shm/src/lib.rs:409.
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: Failed to allocate a SHM buffer, even after GCing, at /root/ws_rmw_zenoh/src/rmw_zenoh/rmw_zenoh_cpp/src/rmw_zenoh.cpp:917, at ./src/rcl/publisher.c:284
```
 I tried to use zc_shm_defrag (By the way I cannot find any sort of documentation or reference for zc_shm_gc or zc_shm_defrag functions) function but it did not help. Is this expected behaviour and how can I solve this issue. When I use default fastrtps niddleware it can share the messages through shared memory with no problems.

Yadunund, 2024-12-19:
Hi there,

rmw_zenoh does not support SHM just yet but it should be coming soon now that we've migrated to Zenoh-1.0.

I'll close this issue as it is a duplicate of https://github.com/ros2/rmw_zenoh/issues/220 and relates to https://github.com/ros2/rmw_zenoh/issues/175

closed 2024-12-19 by Yadunund

cesurs, 2024-12-20:
Thanks, changing 

 #define SHM_BUFFER_SIZE_MB 10 

value (which I for some reason set to 4 previously) at rmw_zenoh/rmw_zenoh_cpp/src/rmw_init.cpp made it possible to share larger messages but messages arrive very sparsely and slowly. I will be looking forward to full shm support.