ros2/rmw_zenoh #350: Missing <functional> include when building on 22.04

christophebedard, 2024-12-19:
I rebased #294 on `rolling` yesterday (potentially after #347), and then `rmw_zenoh_cpp` started failing to build on my 22.04 system:

<details>

<summary>Click to expand</summary>

```
Starting >>> rmw_zenoh_cpp
--- stderr: rmw_zenoh_cpp                                      
In file included from /usr/include/c++/11/bits/move.h:57,
                 from /usr/include/c++/11/bits/stl_pair.h:59,
                 from /usr/include/c++/11/utility:70,
                 from /usr/include/c++/11/array:38,
                 from /home/christophe.bedard/ros2_ws/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/attachment_helpers.cpp:15:
/usr/include/c++/11/type_traits: In instantiation of ‘struct std::__is_trivially_copy_constructible_impl<std::reference_wrapper<zenoh::ZShmMut>, true>’:
/usr/include/c++/11/type_traits:1254:12:   required from ‘struct std::is_trivially_copy_constructible<std::reference_wrapper<zenoh::ZShmMut> >’
/usr/include/c++/11/type_traits:3200:43:   required from ‘constexpr const bool std::is_trivially_copy_constructible_v<std::reference_wrapper<zenoh::ZShmMut> >’
/usr/include/c++/11/optional:662:11:   required from ‘class std::optional<std::reference_wrapper<zenoh::ZShmMut> >’
/home/christophe.bedard/ros2_ws/install/zenoh_cpp_vendor/opt/zenoh_cpp_vendor/include/zenoh/api/shm/buffer/zshm.hxx:67:83:   required from here
/usr/include/c++/11/type_traits:1249:25: error: invalid use of incomplete type ‘class std::reference_wrapper<zenoh::ZShmMut>’
 1249 |                         __is_trivially_constructible(_Tp, const _Tp&)>>
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/11/type_traits:48:11: note: declaration of ‘class std::reference_wrapper<zenoh::ZShmMut>’
   48 |     class reference_wrapper;
      |           ^~~~~~~~~~~~~~~~~
/usr/include/c++/11/type_traits: In instantiation of ‘struct std::is_trivially_copy_constructible<std::reference_wrapper<zenoh::ZShmMut> >’:
/usr/include/c++/11/type_traits:3200:43:   required from ‘constexpr const bool std::is_trivially_copy_constructible_v<std::reference_wrapper<zenoh::ZShmMut> >’
/usr/include/c++/11/optional:662:11:   required from ‘class std::optional<std::reference_wrapper<zenoh::ZShmMut> >’
/home/christophe.bedard/ros2_ws/install/zenoh_cpp_vendor/opt/zenoh_cpp_vendor/include/zenoh/api/shm/buffer/zshm.hxx:67:83:   required from here
/usr/include/c++/11/type_traits:1257:52: error: static assertion failed: template argument must be a complete class or an unbounded array
 1257 |       static_assert(std::__is_complete_or_unbounded(__type_identity<_Tp>{}),
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/11/type_traits:1257:52: note: ‘std::__is_complete_or_unbounded<std::__type_identity<std::reference_wrapper<zenoh::ZShmMut> > >((std::__type_identity<std::reference_wrapper<zenoh::ZShmMut> >{}, std::__type_identity<std::reference_wrapper<zenoh::ZShmMut> >()))’ evaluates to false
/usr/include/c++/11/type_traits: In instantiation of ‘constexpr const bool std::is_trivially_copy_constructible_v<std::reference_wrapper<zenoh::ZShmMut> >’:
/usr/include/c++/11/optional:662:11:   required from ‘class std::optional<std::reference_wrapper<zenoh::ZShmMut> >’
/home/christophe.bedard/ros2_ws/install/zenoh_cpp_vendor/opt/zenoh_cpp_vendor/include/zenoh/api/shm/buffer/zshm.hxx:67:83:   required from here
/usr/include/c++/11/type_traits:3200:43: error: ‘value’ is not a member of ‘std::is_trivially_copy_constructible<std::reference_wrapper<zenoh::ZShmMut> >’
 3200 |     is_trivially_copy_constructible<_Tp>::value;
      |                                           ^~~~~
/usr/include/c++/11/type_traits: In instantiation of ‘struct std::__is_trivially_move_constructible_impl<std::reference_wrapper<zenoh::ZShmMut>, true>’:
/usr/include/c++/11/type_traits:1277:12:   required from ‘struct std::is_trivially_move_constructible<std::reference_wrapper<zenoh::ZShmMut> >’
/usr/include/c++/11/type_traits:3203:43:   required from ‘constexpr const bool std::is_trivially_move_constructible_v<std::reference_wrapper<zenoh::ZShmMut> >’
/usr/include/c++/11/optional:662:11:   required from ‘class std::optional<std::reference_wrapper<zenoh::ZShmMut> >’
/home/christophe.bedard/ros2_ws/install/zenoh_cpp_vendor/opt/zenoh_cpp_vendor/include/zenoh/api/shm/buffer/zshm.hxx:67:83:   required from here
/usr/include/c++/11/type_traits:1272:25: error: invalid use of incomplete type ‘class std::reference_wrapper<zenoh::ZShmMut>’
 1272 |                         __is_trivially_constructible(_Tp, _Tp&&)>>
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/11/type_traits:48:11: note: declaration of ‘class std::reference_wrapper<zenoh::ZShmMut>’
   48 |     class reference_wrapper;
      |           ^~~~~~~~~~~~~~~~~
/usr/include/c++/11/type_traits: In instantiation of ‘struct std::is_trivially_move_constructible<std::reference_wrapper<zenoh::ZShmMut> >’:
/usr/include/c++/11/type_traits:3203:43:   required from ‘constexpr const bool std::is_trivially_move_constructible_v<std::reference_wrapper<zenoh::ZShmMut> >’
/usr/include/c++/11/optional:662:11:   required from ‘class std::optional<std::reference_wrapper<zenoh::ZShmMut> >’
/home/christophe.bedard/ros2_ws/install/zenoh_cpp_vendor/opt/zenoh_cpp_vendor/include/zenoh/api/shm/buffer/zshm.hxx:67:83:   required from here
/usr/include/c++/11/type_traits:1280:52: error: static assertion failed: template argument must be a complete class or an unbounded array
 1280 |       static_assert(std::__is_complete_or_unbounded(__type_identity<_Tp>{}),
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/11/type_traits:1280:52: note: ‘std::__is_complete_or_unbounded<std::__type_identity<std::reference_wrapper<zenoh::ZShmMut> > >((std::__type_identity<std::reference_wrapper<zenoh::ZShmMut> >{}, std::__type_identity<std::reference_wrapper<zenoh::ZShmMut> >()))’ evaluates to false
/usr/include/c++/11/type_traits: In instantiation of ‘constexpr const bool std::is_trivially_move_constructible_v<std::reference_wrapper<zenoh::ZShmMut> >’:
/usr/include/c++/11/optional:662:11:   required from ‘class std::optional<std::reference_wrapper<zenoh::ZShmMut> >’
/home/christophe.bedard/ros2_ws/install/zenoh_cpp_vendor/opt/zenoh_cpp_vendor/include/zenoh/api/shm/buffer/zshm.hxx:67:83:   required from here
/usr/include/c++/11/type_traits:3203:43: error: ‘value’ is not a member of ‘std::is_trivially_move_constructible<std::reference_wrapper<zenoh::ZShmMut> >’
 3203 |     is_trivially_move_constructible<_Tp>::value;
      |                                           ^~~~~
/usr/include/c++/11/type_traits: In instantiation of ‘struct std::is_copy_constructible<std::reference_wrapper<zenoh::ZShmMut> >’:
/usr/include/c++/11/type_traits:3180:33:   required from ‘constexpr const bool std::is_copy_constructible_v<std::reference_wrapper<zenoh::ZShmMut> >’
/usr/include/c++/11/optional:666:2:   required from ‘class std::optional<std::reference_wrapper<zenoh::ZShmMut> >’
/home/christophe.bedard/ros2_ws/install/zenoh_cpp_vendor/opt/zenoh_cpp_vendor/include/zenoh/api/shm/buffer/zshm.hxx:67:83:   required from here
/usr/include/c++/11/type_traits:987:52: error: static assertion failed: template argument must be a complete class or an unbounded array
  987 |       static_assert(std::__is_complete_or_unbounded(__type_identity<_Tp>{}),
[... 399 lines cut ...]
@@ -15,6 +15,8 @@
 #ifndef DETAIL__ZENOH_UTILS_HPP_
 #define DETAIL__ZENOH_UTILS_HPP_
 
+#include <functional>
+
 #include <zenoh.hxx>
 
 #include <array>
diff --git a/rmw_zenoh_cpp/src/rmw_init_options.cpp b/rmw_zenoh_cpp/src/rmw_init_options.cpp
index 15e06e2..760f904 100644
--- a/rmw_zenoh_cpp/src/rmw_init_options.cpp
+++ b/rmw_zenoh_cpp/src/rmw_init_options.cpp
@@ -12,6 +12,8 @@
 // See the License for the specific language governing permissions and
 // limitations under the License.
 
+#include <functional>
+
 #include <zenoh.hxx>
 
 #include "detail/identifier.hpp"
```

Note that `<zenoh.hxx>` is sometimes included before the STL includes. For example, see `zenoh_utils.hpp`. In any case, this should probably instead be fixed in zenoh.

clalancette, 2024-12-19:
Interestingly, we didn't see this because it doesn't fail on Ubuntu 24.04 for whatever reason.  Regardless, as I wrote in https://github.com/ros2/rmw_zenoh/pull/351#issuecomment-2555191062 , we should fix this in upstream zenoh-cpp.

ahcorde, 2024-12-19:
Related PR https://github.com/eclipse-zenoh/zenoh-cpp/pull/342 in zenoh_cpp

YuanYuYuan, 2024-12-20:
Thanks, @ahcorde and @christophebedard! The PR looks good to me. By the way, I did a test within Iron and Humble dockers (22.04). It surprised me that it passed in Humble but failed in Iron.

clalancette, 2024-12-23:
This was fixed by #353, so closing.

closed 2024-12-23 by clalancette