ros2/rmw_zenoh #364: build error on ros rolling: ros_trace_rmw_send_response was not declared in this scope

berndpfrommer, 2024-12-27:
Commit 79fc5ce304fa4ef9a40ab497e71d8a12d91a2449  breaks the build for me.
I'm getting a build error like this on rolling when using the apt-packages for base ros rolling

```
In file included from /opt/ros/src/rolling/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/rmw_service_data.cpp:43:
/opt/ros/src/rolling/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/rmw_service_data.cpp: In member function ‘rmw_ret_t rmw_zenoh_cpp::ServiceData::send_response(rmw_request_id_t*, void*)’:
/opt/ros/src/rolling/src/rmw_zenoh/rmw_zenoh_cpp/src/detail/rmw_service_data.cpp:460:3: error: ‘ros_trace_rmw_send_response’ was not declared in this scope
  460 |   TRACETOOLS_TRACEPOINT(
      |   ^~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/rmw_zenoh_cpp.dir/build.make:272: CMakeFiles/rmw_zenoh_cpp.dir/src/detail/rmw_service_data.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
```

This commit still works: 65e0cabeb7bc45ec317ca4f714f8a86df5de9a0b

I assume you guys are compiling against some later rclcpp version. Just documenting this in case other people hit the same problem.

clalancette, 2024-12-27:
Yes, this will be fixed by the next Rolling sync.

clalancette, 2025-01-02:
This should be [fixed now](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2025-01-02/41376), so closing.

closed 2025-01-02 by clalancette

cesurs, 2025-01-20:
> This should be [fixed now](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2025-01-02/41376), so closing.

I think the issue persists.