ros2/rmw_zenoh #390: Plugin Support for RMW_Zenoh zenohd

S3ckShUn21, 2025-01-03:
This may be a silly question, but does the custom `zenohd` provided by this codebase load plugins during the router runtime build step?

I've been trying to load the REST API plugin (compiled with the same git commit from the Rust codebase) to no avail.

I've done some cursory looks around the cpp, c, and rs codebases to try and understand where plugins are being loaded. I see where they're being loaded in the Rust implementation of `zenohd`. But I haven't found yet if/where that's happening with this `zenohd` impl.

If it doesn't exist, should I try and add that functionality in to this custom `zenohd`? I see that a couple people have spent some time looking into using the upstream version of `zenohd` through the `zenoh-c` build step.

clalancette, 2025-01-03:
> This may be a silly question, but does the custom `zenohd` provided by this codebase load plugins during the router runtime build step?

No, it does not.  That's one of the big downsides to `rmw_zenohd`.

> I see that a couple people have spent some time looking into using the upstream version of `zenohd` through the `zenoh-c` build step.

I think we should attempt to use the upstream `zenohd`, rather than slowly replicating its functionality here.  I don't know if that is now easier or harder with the switch to `zenoh_cpp_vendor`, but I still think it should be possible somehow.

S3ckShUn21, 2025-01-03:
Thanks for the quick reply. I'll look into the work being done to move to the upstream zenohd

Yadunund, 2025-01-16:
Closing as the effort to rely on upstream `zenohd` is being discussed in https://github.com/ros2/rmw_zenoh/issues/262

closed 2025-01-16 by Yadunund