ros2/rmw_zenoh #433: Core dumped  with the turtlebot4 simulation

YuanYuYuan, 2025-01-23:
A core dumped was found when launching the turtlebot4 bringup with slam enabled.

```console
❯ ros2 launch turtlebot4_gz_bringup turtlebot4_gz.launch.py slam:=true rviz:=true
\[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-01-23-06-26-28-141401-ZenS13-92691
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): failed to initialize rcl: discovery_options must be zero initialized, at /workspace/src/rmw/rmw/src/discovery_options.c:44, at /workspace/src/rcl/rcl/src/rcl/discovery_options.c:147
free(): invalid size
[1]    92691 IOT instruction (core dumped)  ros2 launch turtlebot4_gz_bringup turtlebot4_gz.launch.py slam:=true
```

## Environment

- ROS2: Jazzy docker Ubuntu 24.04
- rmw_zenoh: bda2d7181d7e1ea2f7d143fabf7c4ffc6c670b20

Yadunund, 2025-01-23:
Please add instructions to build this package along with a link to the source repository.

YuanYuYuan, 2025-01-23:
The core dumped was due to interference from the local rcl & rclcpp.

closed 2025-01-23 by YuanYuYuan