ros2/rmw_zenoh #442: README.md has bringup order wrong, unclear on connecting to

PaulBouchier, 2025-02-02:
A minor documentation error, but README.md walks you through bringup by first having you "Start the Zenoh Router", then "Terminate ROS2 daemon started with another RMW", but this second step kills the Zenoh router. It seems to me the "Terminate ROS2 daemon" step should be placed before the "Start Zenoh" step. It confused me for a little bit.

Also, the section "Connecting multiple hosts" is confusing. It says to put the provided section: {connect: {endpoints: }} into one of the host's configuration file, but leaves it unclear as to whether you need to copy down the whole config file from source and modify just the {mode: router: router} section and then set the env var, or maybe just make a file that contains only the provided snippet (what I tried first). It seems to require providing the whole file AFICT. This could be clarified by ellipses indicating the rest of the file comes before & after the provided snippet, and maybe a few words of explanation connecting the previous section about environment variables to the modified file.

If this is a documentation error, I can submit a PR to fix it. 

Sorry if this seems too dumb and someone ought to be able to figure it out (I did after some experimentation), but I'm a Zenoh n00b and had to dig deeper than I expected because it seemed like it would be an easy recipe, but instead I had to experiment to find a working solution. The "Design" link wasn't helpful because it went into contexts & sessions & a bunch of other stuff that didn't address a dumb "follow these steps" need.

Yadunund, 2025-02-02:
hi @PaulBouchier you're right the instruction to `Terminate ROS 2 daemon started with another RMW` should come before starting the router. Please submit a PR!

PaulBouchier, 2025-02-02:
One thing is still unclear in the instructions, and I can update the PR if
you tell me the answer.

When you have n hosts should n-1 of them point to n-1 other hosts? Should
there be one host that doesn't point to any others? Or does it not matter,
as long as there is at least one unidirectional link? Are redundant links
ok? (I don't really want to get into redundant configurations, just to
accurately describe multiple host configurations.

PR #443 (2025-02-06): Update documentation to install binaries and connect to multiple hosts
Improve README.md by correcting mis-ordering of killing other rmw's. Also clarify procedure for modifying router config file

Fix #442
PR #452 (2025-02-06): Update documentation to install binaries and connect to multiple hosts (backport #443)
Improve README.md by correcting mis-ordering of killing other rmw's. Also clarify procedure for modifying router config file

Fix #442 <hr>This is an automatic backport of pull request #443 done by [Mergify](https://mergify.com).
PR #453 (2025-02-06): Update documentation to install binaries and connect to multiple hosts (backport #443)
Improve README.md by correcting mis-ordering of killing other rmw's. Also clarify procedure for modifying router config file

Fix #442 <hr>This is an automatic backport of pull request #443 done by [Mergify](https://mergify.com).
closed 2025-02-06 by Yadunund