ros2/rmw_zenoh #447: Excessive disk usage in build directory - 4.4 GB

PaulBouchier, 2025-02-04:
When building using the command line specified in the README.md, 
`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`
the build directory grows to 4.4 GB. This seems to be due to stuff in a couple of directories called debug way down in the build tree. 

This problem causes issues on small platforms like Raspberry Pi (I ran out of space), and the requirement to give --cmake-args makes it not fit well into a standard workspace build, where you can just type colcon build.

Is there some short-term workaround to avoid the bloat? And can the build be made so that the default colcon build just does what it should?

Yadunund, 2025-02-04:
@PaulBouchier thanks for the ticket. I can confirm the build dir for `zenoh_cpp_vendor` is ~4.3GB.

It's unclear whether this is "bloat" introduced by our build system or whether building Zenoh-c/cpp takes up this much space. We'd appreciate if you could investigate by building zenoh-cpp following the instructions upstream https://github.com/eclipse-zenoh/zenoh-cpp?tab=readme-ov-file#building-and-running-tests while passing along the flags as we do in rmw_zenoh, https://github.com/ros2/rmw_zenoh/blob/2e43a482ce82a475f4b561f0722bdbe7becd18c3/zenoh_cpp_vendor/CMakeLists.txt#L18-L40

Anyways, building code from source on RPi will likely lead to resource constraints. Why not rely on `rmw_zenoh` binaries instead?

PaulBouchier, 2025-02-05:
Thanks for the suggestion to use binaries. I did that and they work fine. I added README.md documentation about using binary packages to my PR. I'll respond here with findings from the investigation you requested.

Yadunund, 2025-02-14:
This is just a byproduct of rust builds which end up building all dependencies from source (no dynamic linking) so there's not much we can do about it. I'll close this out as there are workarounds like installing binaries, upgrading storage, etc.

closed 2025-02-14 by Yadunund

PaulBouchier, 2025-02-15:
Sounds good. Thanks for the explanation. I'm sorry I didn't get to my
assignment.

Paul

Yadunund, 2025-02-15:
No worries! Thanks for your help with updating the README! Let us know if you run into any other issues with rmw_zenoh!