ros2/rmw_zenoh #450: Add instructions to install rmw_zenoh binaries in README

Yadunund, 2025-02-04:
The README only has a setup section to inform users to build from source. But `rmw_zenoh` binaries are available on all Tier-1 platforms so users can just install binaries. We should update the README to mention the availability of binaries and install from source only if latest features are required.

PaulBouchier, 2025-02-05:
You're ahead of the game - I was going to suggest including in my README change, instructions on how to install binaries. I didn't know they were available. I'm happy to update my PR with that, as well as your requested changes to the README PR. I just need to know one thing:
The x86 packages available are ros-[jazzy | rolling]-cpp and -cpp-vendor. The arm packages are only -cpp - no cpp-vendor. What needs to get installed?

PaulBouchier, 2025-02-05:
Updated PR #443 with binary installation instructions

closed 2025-02-07 by Yadunund