ros2/rmw_zenoh #451: `rmw_zenohd` crashes on IPv4-only systems

gavanderhoorn, 2025-02-05:
As per title.

On systems that have no IPv6 support at all, starting `rmw_zenohd` results in:

```
2025-02-05T17:23:30.329605Z  WARN ThreadId(03) zenoh::net::runtime::orchestrator: Unable to open listener tcp/[::]:7447: Can not create a new TCP listener bound to tcp/[::]:7447: [Os { code: 97, kind: Uncategorized, message: "Address family not supported by protocol" }] at /home/buildfarm/.cargo/git/checkouts/zenoh-cc237f2570fab813/9640d22/io/zenoh-links/zenoh-link-tcp/src/unicast.rs:326.
2025-02-05T17:23:30.329864Z ERROR ThreadId(03) zenohc::session: Error opening session: Can not create a new TCP listener bound to tcp/[::]:7447: [Os { code: 97, kind: Uncategorized, message: "Address family not supported by protocol" }] at /home/buildfarm/.cargo/git/checkouts/zenoh-cc237f2570fab813/9640d22/io/zenoh-links/zenoh-link-tcp/src/unicast.rs:326.
Error opening Session!\n[ros2run]: Process exited with failure 1
```

Note the `tcp/[::]:7447` in the error message, which indicates it's trying to bind to `ANY` but IPv6.

This seems to be due to https://github.com/eclipse-zenoh/zenoh/pull/366, which was also discussed in https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/issues/84.

So technically not a bug, but at least surprising.

Perhaps the error message could be improved? Anyone not into networking (which I guess would be at least part of the audience of `rmw_zenoh`) would likely not immediately understand the error.

Yadunund, 2025-02-05:
@gavanderhoorn thanks for reporting the issue. I would imagine a more description error message would be something to add to upstream Zenoh.
What we could do in `rmw_zenoh` for now is add an entry for this under the [Known issues](https://github.com/ros2/rmw_zenoh?tab=readme-ov-file#known-issues) section of the README to inform users of the issue and possible workarounds. 

Please open a PR and i'll be happy to review it!

gavanderhoorn, 2025-02-05:
> I would imagine a more descriptive error message would be something to add to upstream Zenoh.

could be. I'm a user of `rmw_zenoh` though, hence I report it here.

> Please open a PR and i'll be happy to review it!

this may take a while, but I'll put it on the TODO list.

closed 2025-02-13 by Yadunund