ros2/rmw_zenoh #468: Access control not working as expected

jrached, 2025-02-15:
I have two machines, both running their own zenoh router using the binary release for humble. One machine is publishing several topics and I only want to let the ones containing the word "pose" through. When I run the zenoh router on the local machine the zenoh log outputs the following,

```
2025-02-15T00:03:58.461765Z  INFO net-0 ThreadId(03) zenoh::net::routing::interceptor::access_control: c52dd07b5dcf950d476cd6d642fedb0f did not match any configured ACL subject. Default permission `Deny` will be applied on all messages
```
Where c52dd07b5dcf950d476cd6d642fedb0f is the identifier of the remote router. And I'm unable to subscribe to any remote topics.

Here is my access control configuration, based on https://zenoh.io/docs/manual/access-control/ and https://youtu.be/m5x7QABH1xI?t=2807 

```
access_control: {
   "enabled": true,
   "default_permission": "deny",
   "rules":
   [
     {"id": "pose_rule",
      "permission": "allow", "actions": ["put", "query", "declare_subscriber"], "messages": ["put", "query", "declare_subscriber"],
      "flows":["ingress", "egress"], "interfaces": ["enp3s0", "wlo1", "wlp1s0"],
      "key_exprs": ["*/$*%pose", "pose"],
     }
   ],
   "subjects":
   [
      {
        "id": "subject1",
        "interfaces": [
          "enp3s0",
        ],
        "usernames": [
          "px01",
        ],
      },
      ]
   "policies":
   [
     {
       "rules": ["pose_rule"],
       "subjects": ["subject1"],
     },
   ]
  },
```

Both subjects are configured through username in the transport section of the config file, where credentials.txt contains the username and passwords of the two routers involved, following the following docs: https://zenoh.io/docs/manual/user-password/ 

```
auth: {
      usrpwd: {
        user: "al00",
        password: "uavswarm",
        dictionary_file: "/home/user/project/credentials.txt",
      },
    },
```

I have tried defining a wildcard subject in the local machine's config file, but I think all that achieves is that the local router identifier is taken in as the wildcard subject and the remote one is ignored.

 Any advice is greatly appreciated. If there is a more direct way to configure the subjects, say using the router identifier, I'd love to learn how that's done.

Yadunund, 2025-02-15:
Hi @jrached,

In general, the humble support for rmw_zenoh is largely untested and we're expecting the community to help identify and patch any issues. 

While this issue may not be a humble specific, I'd still appreciate if you could try with the latest Jazzy or Rolling binaries and report back. 

I'd like to point out that we're still playing around with security ourselves in various open tickets like https://github.com/ros2/rmw_zenoh/pull/411 and is not something we're promoting just yet with rmw_zenoh although it's supported in Zenoh. 

But if I had to guess looking at the config, the issue could be because you're not allowing liveliness tokens through which is what we use for discovery. Without those key expressions bring allowed to cross hosts, I don't expect your nodes in a different host to discover and receive any messages from a node on a different host. In the ticket linked above, I've highlighted how to allow liveliness key expressions. Perhaps you could try it out and let me know.

Yadunund, 2025-03-20:
You may reference https://github.com/ZettaScaleLabs/roscon2024_workshop/blob/main/exercises/ex-6.md

And we'd really appreciate if you can try out the config generator from this PR https://github.com/ros2/rmw_zenoh/pull/495 and help us review the same.

cgeller, 2025-03-24:
Thanks for the hint! I had a look to PR #495 and successfully tested it (see https://github.com/ros2/rmw_zenoh/pull/495#issuecomment-2747586584).

Only defined messages are published. However, `ros2 topic list` shows all topics. Can we somehow also deny the "discovery traffic" so that we allow only custom defined topics?

Yadunund, 2025-04-05:
@cgeller thanks for trying out that PR and providing feedback. https://github.com/ros2/rmw_zenoh/pull/595 builds off that PR to also add authentication & encryption which is documented in the added README. If your publisher is configured with the auth enclaves, `ros2 topic list` will not show the topic unless also run from a session that is authenticated.

Yadunund, 2025-05-18:
Closing this ticket as we've finally added support for generating zenoh configs with access control settings. Feel free to reopen if the issue persists.

closed 2025-05-18 by Yadunund