ros2/rmw_zenoh #479: [Design] rmw_zenoh payload documentation

PetervdPerk-NXP, 2025-02-26:
I'm trying to communicate from zenoh-pico directly to rmw_zenoh_cpp.

I can create topic using `domain_id/topic_name/message_type/RIHS01_hash` and subscribe to them using `ros2 topic echo` however I'm getting these errors.

```
[ERROR] [1740587814.207593503] [rmw_zenoh_cpp]: Unable to obtain attachment
```

With the https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds I could make the following payload
```<CDR Xtypes header><IDL payload>```

I presume for rmw_zenoh_cpp this different but I can't find any documentation for it.

Yadunund, 2025-02-26:
> I'm trying to communicate from zenoh-pico directly to rmw_zenoh_cpp.

We don't yet support interoperability between zenoh-pico and rmw_zenoh_cpp. See https://github.com/ros2/rmw_zenoh/issues/125 for more information.

> With the https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds I could make the following payload
<CDR Xtypes header><IDL payload>

I'm sorry but I'm a bit confused as to why you're relying on `zenoh-plugin-ros2dds` to communicate between pico and `rmw_zenoh`? Could you clarify? 

> I presume for rmw_zenoh_cpp this different but I can't find any documentation for it.

We rely on the type_support to serialize the ROS message into bytes which we directly send over Zenoh. See https://github.com/ros2/rmw_zenoh/blob/4fc36b75606911c2bbd2467667827f04edc7254f/rmw_zenoh_cpp/src/detail/rmw_publisher_data.cpp#L232.
Similarly on the subscription side, the type_support deserializes the bytes into a ROS message.

PetervdPerk-NXP, 2025-02-26:
Okay I presumed with the bump to Zenoh 1.2.1 the version of Zenoh matches between rmw_zenoh and zenoh-pico and thus it would be interoperable.

> I'm sorry but I'm a bit confused as to why you're relying on zenoh-plugin-ros2dds to communicate between pico and rmw_zenoh? Could you clarify?

I'm relying on zenoh-plugin-ros2dds to go from zenoh-pico to ros2 using rmw_cyclonedds_cpp. But the idea is to switching to rmw_zenoh_cpp directly removing the need of the bridge.

> We rely on the type_support to serialize the ROS message into bytes which we directly send over Zenoh. See 

I'm doing standalone CDR (from CycloneDDS) with Xtypes, I presume that's the same as type_support, but I can look into that further.

It seems that error `Unable to obtain attachment` I'm getting is from the Zenoh incompatibility and not from the serialization. Any roadmap on when zenoh-pico and rmw_zenoh should be interoperable?

JEnoch, 2025-02-27:
Hi @PetervdPerk-NXP 

> It seems that error Unable to obtain attachment I'm getting is from the Zenoh incompatibility and not from the serialization.

No, it's coming from rmw_zenoh Subscriber implementation that expects to receive an attachment along the payload:
https://github.com/ros2/rmw_zenoh/blob/4fc36b75606911c2bbd2467667827f04edc7254f/rmw_zenoh_cpp/src/detail/rmw_subscription_data.cpp#L198-L204

You can make your zenoh-pico Publisher compatible with an rmw_zenoh Subscriber just adding a compliant attachment. See the [z_pub_attachment.c](https://github.com/eclipse-zenoh/zenoh-pico/blob/128f980a30e3155bdc9477c2b4de07ca7c2dc241/examples/unix/c11/z_pub_attachment.c#L119) as an example.
The attachment has to contain what the [rmw_zenoh Subscriber expects](https://github.com/ros2/rmw_zenoh/blob/4fc36b75606911c2bbd2467667827f04edc7254f/rmw_zenoh_cpp/src/detail/attachment_helpers.cpp#L79):
- a sequence number as `int64_t`
- a timestamp ([number of ns since EPOCH](https://github.com/ros2/rmw_zenoh/blob/4fc36b75606911c2bbd2467667827f04edc7254f/rmw_zenoh_cpp/src/detail/zenoh_utils.cpp#L68)) as `int64_t`
- a unique publisher identifier (DDS-like GID) as 16*`uint8_t`

All serialized with the Zenoh [Serialization API](https://zenoh-pico.readthedocs.io/en/1.2.1/api.html#serialization).

Yadunund, 2025-03-19:
Closing as a solutions is provided above. Feel free to reopen if some other issue persists.

closed 2025-03-19 by Yadunund

g4sp3r, 2025-05-18:
Just leaving this here for the next person who might experience the same issue...

To support communication from zenoh-pico I have added the following structure to my embedded project:
```c
typedef struct __attribute__((__packed__)){
    int64_t sequence_n;
    int64_t time;
    uint8_t rmw_gid_size;
    uint8_t rmw_gid[16];
}rmw_attachment_t;
```
Serialization is done simply by using `z_bytes_from_static_buf` and moving bytes to attachment.

During first tests I had forget to set `rmw_gid_size` to 16 and I had panic and crash of ROS subscriber node:
```
thread 'rx-1' panicked at library/core/src/panicking.rs:136:5:
panic in a function that cannot unwind
stack backtrace:
   0: rust_begin_unwind
             at /build/rustc-jTnCR5/rustc-1.75.0+dfsg0ubuntu1/library/std/src/panicking.rs:645:5
   1: core::panicking::panic_nounwind_fmt
             at /build/rustc-jTnCR5/rustc-1.75.0+dfsg0ubuntu1/library/core/src/panicking.rs:106:14
   2: core::panicking::panic_nounwind
             at /build/rustc-jTnCR5/rustc-1.75.0+dfsg0ubuntu1/library/core/src/panicking.rs:136:5
   3: core::panicking::panic_cannot_unwind
             at /build/rustc-jTnCR5/rustc-1.75.0+dfsg0ubuntu1/library/core/src/panicking.rs:223:5
   4: std::sys::unix::thread::Thread::new::thread_start
             at /build/rustc-jTnCR5/rustc-1.75.0+dfsg0ubuntu1/library/std/src/sys/unix/thread.rs:102:9
   5: <unknown>
   6: <unknown>
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
thread caused non-unwinding panic. aborting.
Aborted (core dumped)
```

I took me more time than I would like to admit to find and fix this user error on my side.

As a suggestion, maybe some error message can be shown to user when attachment format is not as expected.