ros2/rmw_zenoh #482: zenoh-cpp flags such as Z_FEATURE_SHARED_MEMORY are not defined

Pleune, 2025-02-27:
I am coming from building this using nix, so apologies for the unusual build environment. 

When I build the versions of `zenoh-c` and `zenoh-cpp` that are mentioned in the `zenoh-cpp-vendor/CMakeListst.txt`, everything goes fine until the the rmw_zenoh package build and I get errors such as below.

```
build/rmw_zenoh_cpp/src/detail/rmw_publisher_data.hpp:59:26: error: 'ShmProvider' is not a member of 'zenoh'
   59 |     std::optional<zenoh::ShmProvider> & shm_provider);
      |                          ^~~~~~~~~~~
/build/rmw_zenoh_cpp/src/detail/rmw_publisher_data.hpp:59:37: error: template argument 1 is invalid
   59 |     std::optional<zenoh::ShmProvider> & shm_provider);
      |                                     ^
/build/rmw_zenoh_cpp/src/detail/rmw_publisher_data.hpp:64:26: error: 'ShmProvider' is not a member of 'zenoh'
   64 |     std::optional<zenoh::ShmProvider> & shm_provider);
      |                          ^~~~~~~~~~~
/build/rmw_zenoh_cpp/src/detail/rmw_publisher_data.hpp:64:37: error: template argument 1 is invalid
   64 |     std::optional<zenoh::ShmProvider> & shm_provider);
``` 

This seems to be because `zenoh.hxx` guards some features behind enable flags like `Z_FEATURE_SHARED_MEMORY` and `Z_FEATURE_UNSTABLE_API` https://github.com/eclipse-zenoh/zenoh-cpp/blob/5dfb68c9ac966925e59bcb52f39b9bc26c0ad6d3/include/zenoh/api.hxx#L42

In the normal ros distribution builds, how are these flags passed to the build environment? I don't see any documentation or mention of these flags... anywhere actually.

Thanks for the awesome work on zenoh

YuanYuYuan, 2025-02-27:
Hi @Pleune! Can you check if you built zenoh-c vendor with unstable flags that are equivalent to what we do in CMakeLists.txt?

```txt
# Disable default features and enable only the most useful ones. This reduces build time and footprint.
# For a complete list of features see: https://github.com/eclipse-zenoh/zenoh/blob/main/zenoh/Cargo.toml
# Note: We separate the two args needed for cargo with "$<SEMICOLON>" and not ";" as the
# latter is a list separater in cmake and hence the string will be split into two
# when expanded.
set(ZENOHC_CARGO_FLAGS "--no-default-features$<SEMICOLON>--features=shared-memory zenoh/transport_compression zenoh/transport_tcp zenoh/transport_tls")

ament_vendor(zenoh_c_vendor
  VCS_URL https://github.com/eclipse-zenoh/zenoh-c.git
  VCS_VERSION 261493682c7dc54db3a07079315e009a2e7c1573
  CMAKE_ARGS
    "-DZENOHC_CARGO_FLAGS=${ZENOHC_CARGO_FLAGS}"
    "-DZENOHC_BUILD_WITH_UNSTABLE_API=TRUE"
    "-DZENOHC_CUSTOM_TARGET=${ZENOHC_CUSTOM_TARGET}"
)
```

Pleune, 2025-02-27:
Hi thanks for looking at this! Yeah I think I have properly built zenoh-c with these flags. I think they are not being properly propagated to the compilation of `rmw_zenoh_cpp`. I found out I can get a little further by patching `rmw_zenoh_cpp/CMakeLists.txt` like so:

```cmake
target_compile_definitions(rmw_zenoh_cpp
  PRIVATE
    RMW_VERSION_MAJOR=${rmw_VERSION_MAJOR}
    RMW_VERSION_MINOR=${rmw_VERSION_MINOR}
    RMW_VERSION_PATCH=${rmw_VERSION_PATCH}
    Z_FEATURE_UNSTABLE_API Z_FEATURE_SHARED_MEMORY Z_FEATURE_QUERY
)
```

I then get errors like
```
[ 92%] Building CXX object CMakeFiles/rmw_zenoh_cpp.dir/src/rmw_qos.cpp.o
In file included from /nix/store/spzviai36pxbfq8s6i1pm2nnvgbdysls-zenoh-c/include/zenoh.h:29,
                 from /nix/store/92drrlpjdj8g9fb63vg95wq5knrc04y5-zenoh-cpp/include/zenoh/zenohc.hxx:14,
                 from /nix/store/92drrlpjdj8g9fb63vg95wq5knrc04y5-zenoh-cpp/include/zenoh/api/base.hxx:21,
                 from /nix/store/92drrlpjdj8g9fb63vg95wq5knrc04y5-zenoh-cpp/include/zenoh/api/bytes.hxx:18,
                 from /nix/store/92drrlpjdj8g9fb63vg95wq5knrc04y5-zenoh-cpp/include/zenoh/api.hxx:16,
                 from /nix/store/92drrlpjdj8g9fb63vg95wq5knrc04y5-zenoh-cpp/include/zenoh.hxx:19,
                 from /build/rmw_zenoh_cpp/src/detail/attachment_helpers.cpp:22:
/nix/store/spzviai36pxbfq8s6i1pm2nnvgbdysls-zenoh-c/include/zenoh_commons.h:339:3: error: 'z_owned_shm_mut_t' does not name a type; did you mean 'z_owned_sample_t'?
  339 |   z_owned_shm_mut_t buf;
      |   ^~~~~~~~~~~~~~~~~
      |   z_owned_sample_t
/nix/store/spzviai36pxbfq8s6i1pm2nnvgbdysls-zenoh-c/include/zenoh_commons.h:455:3: error: 'zc_owned_concurrent_close_handle_t' does not name a type
  455 |   zc_owned_concurrent_close_handle_t *internal_out_concurrent;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/nix/store/spzviai36pxbfq8s6i1pm2nnvgbdysls-zenoh-c/include/zenoh_commons.h:811:3: error: 'z_moved_source_info_t' does not name a type; did you mean 'z_moved_string_t'?
  811 |   z_moved_source_info_t *source_info;
      |   ^~~~~~~~~~~~~~~~~~~~~
      |   z_moved_string_t
/nix/store/spzviai36pxbfq8s6i1pm2nnvgbdysls-zenoh-c/include/zenoh_commons.h:895:3: error: 'z_moved_source_info_t' does not name a type; did you mean 'z_moved_string_t'?
  895 |   z_moved_source_info_t *source_info;
      |   ^~~~~~~~~~~~~~~~~~~~~
      |   z_moved_string_t
/nix/store/spzviai36pxbfq8s6i1pm2nnvgbdysls-zenoh-c/include/zenoh_commons.h:948:3: error: 'z_moved_source_info_t' does not name a type; did you mean 'z_moved_string_t'?
  948 |   z_moved_source_info_t *source_info;
      |   ^~~~~~~~~~~~~~~~~~~~~
      |   z_moved_string_t
/nix/store/spzviai36pxbfq8s6i1pm2nnvgbdysls-zenoh-c/include/zenoh_commons.h:975:3: error: 'z_moved_source_info_t' does not name a type; did you mean 'z_moved_string_t'?
```

Pleune, 2025-02-27:
Hmm more debugging, it actually seems I was having a kind-of cache invalidation issue. Possibly related to https://github.com/NixOS/nixpkgs/issues/183344? I'm not sure if that makes sense in my usecase. Anyways, while the flags were being passed in my nix file, nix was still using a version of zenoh where the cargo flags were not passed from a previous build.

Anyways thank you for the help, it did keep be pointed in the right direction!

closed 2025-02-27 by Pleune